dariuszseweryn/RxAndroidBle #451: Question regarding iBeacon

priyakar, 2018-06-26:
I have created a beacon through the Beacon Simulator app on android and I am trying to scan this by filtering through the UUID I get for the beacon. Ideally it should show up in the scan but its not! Any idea why? `new ScanSettings.Builder().setCallbackType(ScanSettings.CALLBACK_TYPE_ALL_MATCHES)
            .setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
            .build()` this is my Settings and `new ScanFilter.Builder().setServiceUuid(uuid).build())` is my filter

dariuszseweryn, 2018-06-27:
I have just tested locally and it works as expected here. Maybe your device is not advertising the UUID you expect?

priyakar, 2018-06-27:
I can see it if I use the android BluetoothScanner though! Which app did you use to simulate the iBeacon, if you dont mind me asking?

dariuszseweryn, 2018-06-27:
I did not used any beacon app. I just scanned a known peripheral that advertises a particular service UUID.

priyakar, 2018-06-27:
Oh that works. I have it working for everything BUT the ibeacon.

dariuszseweryn, 2018-06-27:
I do not know the iBeacon specification and cannot support you in this matter.

You may try some other library designed for specifically for beacons [RxBeacon](https://github.com/florent37/RxBeacon) or may go to [stackoverflow](https://stackoverflow.com/questions/tagged/rxandroidble) for help as this is more of a general usage question.

I am closing this topic as it is not an issue with the library.

Best Regards

closed 2018-06-27 by dariuszseweryn