dariuszseweryn/RxAndroidBle #458: scanBleDevices, Android 6+, invisible after another app scan

JeanChristophePal, 2018-07-11:
When I use  "scanBleDevices" I can see my BLE devices. Then un Use another app that scan BLE (from the ble device constructor, Microship). 
Then, when I go back to my app a new "scanBleDevices" can't find my BLE devices. 

If I disable the localization (and/or Bluetooth) manually, the system ask me to re-enable the localization. After pushing the "OK" button to re-enable the localization my BLE devices appear in my app.

Do you have an idea of the cause? 

version:
implementation "com.polidea.rxandroidble:rxandroidble:1.4.1"

I also use TedPermission : 
```text
new TedPermission(getContext())
                            .setPermissionListener(((AddOfiActivity)getActivity()))
                            .setPermissions(Manifest.permission.ACCESS_FINE_LOCATION, Manifest.permission.ACCESS_COARSE_LOCATION)
                            .setDeniedMessage(R.string.permissions_refused)
                            .check();
```

then 

```text
    @Override
    public void onPermissionGranted()
    {
OFIDeviceConnection.getInstance(getBaseContext()).scanBleDevices(state -> BleUtils.processErrorStates(TAG, state, this, this));
   }
```

Thanks you,

dariuszseweryn, 2018-07-11:
Hello!

Thank you for using the library.

This is a general usage question and as such should be posted on [stackoverflow](https://stackoverflow.com/questions/tagged/rxandroidble) with `rxandroidble` tag—it will be easier to find by other users.

Best Regards

P.S. If the other app has connected to your device it may no longer be advertising and thus not scannable

closed 2018-07-11 by dariuszseweryn

JeanChristophePal, 2018-07-12:
Hello, 

Sorry for posting this at the wrong place. On stackoverflow :
https://stackoverflow.com/questions/51299418/rxandroidble-scanbledevices-android-6-device-are-invisible-after-another-app