dariuszseweryn/RxAndroidBle #459: Help needed for Characteristic Write Error

keithle888, 2018-07-12:
Hello, 

I've been using the native android code for some BLE operations, but am now trying to migrate to this library for better stability.

I've been trying to write a payload (broken up into 20 byte packets) to a peripheral but I'm getting an error when the following code is called. (I know LongWriteBuilder is available but its not a function I can use atm.)

````
private fun RxBleConnection.batchWrites(payload: ByteArray, mtu: Int, characteristicUUID: UUID): Completable {
        val cPayload = payload.clone()
        return this.writeCharacteristic(characteristicUUID, cPayload.getNextPacket(mtu))
                .doOnSuccess { Timber.d("Successfully wrote packet: ${payload.toPlainHex()}") }
                .repeatUntil { cPayload.isEmpty() }
                .ignoreElements()

    }
````

I've been getting error 133 from multiple phones (Pixel 2, Oneplus 3) when it tries to write to the peripheral. Any advice would be appreciated.

dariuszseweryn, 2018-07-12:
Hello

Thanks for using the library!

It looks that something is off with your peripheral if BLE stacks of Pixel 2 and OnePlus 3 are not handling normal writes

Btw. The above code will repeat indefinitely as `cPayload.getNextPacket(mtu)` is called only once.

keithle888, 2018-07-12:
Hello!

Thanks for the prompt reply. cPayload.getNextPacket() will drop the bytes it returns, so the array will eventually empty out.

I've implemented the same behaviour with the Native Android BLE library and it works fine with both phones. So the difference in behaviour is somewhat confusing. I'll put up the logs and more code from my implementation in a few hours.

dariuszseweryn, 2018-07-12:
try to change this:
```
this.writeCharacteristic(characteristicUUID, cPayload.getNextPacket(mtu))
```
into this:
```
Single.defer { this.writeCharacteristic(characteristicUUID, cPayload.getNextPacket(mtu)) }
```
Then each `.repeatUntil { cPayload.isEmpty() }` will actually be called with `cPayload.getNextPacket(mtu)`.

keithle888, 2018-07-13:
Thanks alot for the heads up!

Here is the logs for the error I'm receiving.

````
14307-16162 D: QUEUED   CharacteristicWriteOperation(129362019)
14307-16161 D: STARTED  CharacteristicWriteOperation(129362019)
14307-14327 D: onCharacteristicWrite characteristic=5c3a659f-897e-45e1-b016-007107c96df6 status=133
14307-14327 D: onClientConnectionState() - status=19 clientIf=6 device=ED:11:D4:AD:25:69
14307-16161 D: FINISHED CharacteristicWriteOperation(129362019) in 311 ms
14307-16162 E: Write failed.
    com.polidea.rxandroidble2.exceptions.BleGattCharacteristicException: GATT exception from MAC address ED:11:D4:AD:25:69, status 133, type BleGattOperation{description='CHARACTERISTIC_WRITE'}. (Look up status 0x85 here https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h)
        at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback.propagateErrorIfOccurred(RxBleGattCallback.java:228)
        at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback.access$800(RxBleGattCallback.java:33)
        at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onCharacteristicWrite(RxBleGattCallback.java:122)
        at android.bluetooth.BluetoothGatt$1$7.run(BluetoothGatt.java:407)
        at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:711)
        at android.bluetooth.BluetoothGatt.-wrap0(Unknown Source:0)
        at android.bluetooth.BluetoothGatt$1.onCharacteristicWrite(BluetoothGatt.java:403)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:136)
        at android.os.Binder.execTransact(Binder.java:692)
14307-14327 D: onConnectionStateChange newState=0 status=19
14307-14327 I: Connection operations queue to be terminated (ED:11:D4:AD:25:69)
14307-16161 D: Terminated.
14307-16162 D: QUEUED   DisconnectOperation(266990873)
````

And this is what I'm using to connect:
````
fun connect(connectionType: ConnectionType): Observable<ConnectionState> {
        val startTime = System.currentTimeMillis()
        this.connectionType = connectionType
        return rxBleDevice.establishConnection(false)
                .doOnSubscribe { Timber.d("Attempting to connect to: ${this.name}") }
                .doOnNext {
                    this.bleConnection = it
                    Timber.d("BLE Connection to ${name} took: ${System.currentTimeMillis() - startTime}ms")
                }.flatMap {connection -> connection.discoverServices().toObservable().map { Pair(connection, it) } }
                .doOnNext {
                    Timber.d("Successfully discovered services. Took: ${System.currentTimeMillis() - startTime}ms")
                    if (disposableList.find { it.purpose == DisposableHolder.PURPOSE_NOTIFICATION } == null) {
                        disposableList.add(DisposableHolder(
                                DisposableHolder.PURPOSE_NOTIFICATION,
                                it.first.setupNotification(getCurrentCharacteristicUUID(), NotificationSetupMode.COMPAT)
                                        .mergeWith(it.first.writeDescriptor(serviceUUID.uuid, getCurrentCharacteristicUUID(), getCurrentDescriptorUUID(), BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE))
                                        .doOnNext { Timber.d("Successfully setup notification. Took: ${System.currentTimeMillis() - startTime}ms") }
                                        .flatMap { it }
                                        .subscribe({
                                            handleBleNotification(it)
                                        }, {
                                            Timber.e(it, "Error setting up notification.")
                                            teardownConnection()
                                        })
                        ))}
                }
                .flatMapCompletable { establishSecureConnection(connectionType) }
                .andThen(Observable.just(ConnectionState.Connected))
    }
````

dariuszseweryn, 2018-07-13:
Your peripheral is not properly responding for the write and then it disconnects ([`14307-14327 D: onConnectionStateChange newState=0 status=19`](https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h#109))

This seems to be a business logic issue — not library

keithle888, 2018-07-16:
Hello Darius, 

I found the issue. Was send the peripheral incorrect data, forcing the peripheral to disconnect. Thanks alot for your time and advice!

closed 2018-07-16 by keithle888