dariuszseweryn/RxAndroidBle #462: ScanSettings.CALLBACK_TYPE_ALL_MATCHES no results

balakzde, 2018-07-19:
If I set ScanSettings.CALLBACK_TYPE_ALL_MATCHES, I am getting no results on Sony xperia devices.

`rxBleClient.scanBleDevices(
                ScanSettings.Builder().setScanMode(scanMode).setCallbackType(ScanSettings.CALLBACK_TYPE_ALL_MATCHES).build(),
                ScanFilter.Builder().setServiceUuid(ParcelUuid.fromString(Constants.HELIOS_SERVICE_UUID)).build()
)`

 I am only getting results with settings ScanSettings.CALLBACK_TYPE_FIRST_MATCH or ScanSettings.CALLBACK_TYPE_MATCH_LOST which is not good enough for me.

`rxBleClient.scanBleDevices(
                ScanSettings.Builder().setScanMode(scanMode).setCallbackType(ScanSettings.CALLBACK_TYPE_FIRST_MATCH or ScanSettings.CALLBACK_TYPE_MATCH_LOST).build(),
                ScanFilter.Builder().setServiceUuid(ParcelUuid.fromString(Constants.HELIOS_SERVICE_UUID)).build()
        )`

I know this is not the problem of RxAndroidBle library, becouse I have tried the same with native android api with same result, but I assume you have much more experience with bluetooth api and may have found solution for this.

dariuszseweryn, 2018-07-19:
Could you add what exact model and OS version is causing the issues? This information could be valuable for future reference.

Unfortunately I have no solution apart of not passing a `ScanFilter` to the `scanBleDevices` and filter the output via `.filter()` operator. This will unfortunately make the scan to be paused on newer OSes when the app goes background.

balakzde, 2018-07-19:
Sony Xperia XZ1 Compact, model: g8441, build number: 47.1.A.12.235
OS: 8.0.0

I believe that what you are suggesting would not solve my problem either, there is simply problem with ScanSettings.CALLBACK_TYPE_ALL_MATCHES which is not problem of your library but different (probably incorect) implementation of android api on sonys side.

After a lot of research on this topic I found out there are tons of problems with ble scanning on android devices and lot of articles suggested robust restart and reconnect strategy.

I managed to resolve my issue by using ScanSettings.CALLBACK_TYPE_FIRST_MATCH and on every scan result I plan reset of scanning, which will act similar as ScanSettings.CALLBACK_TYPE_ALL_MATCHES, with only difference that the period can not be shorter that 6 seconds becouse of: https://blog.classycode.com/undocumented-android-7-ble-behavior-changes-d1a9bd87d983

Since some devices can have problem with ScanSettings.CALLBACK_TYPE_ALL_MATCHES and some with ScanSettings.CALLBACK_TYPE_FIRST_MATCH the only option is to use both or let the user decide in settings according to his experience with app.

dariuszseweryn, 2018-07-19:
The `ScanSettings.CALLBACK_TYPE_ALL_MATCHES` is the default for using a BLE scan. I am almost sure that it will work if you will not provide any `ScanFilter`s.

> After a lot of research on this topic I found out there are tons of problems with ble scanning on android devices and lot of articles suggested robust restart and reconnect strategy.

If you have the articles still opened could you also reference them here? I would also like to go through them as I do not recall any of such

balakzde, 2018-07-19:
> The ScanSettings.CALLBACK_TYPE_ALL_MATCHES is the default for using a BLE scan. I am almost sure that it will work if you will not provide any ScanFilters.

You were right, if no filters is supplied it works. I may try this solution as well.

> If you have the articles still opened could you also reference them here? I would also like to go through them as I do not recall any of such

I think I got the idea of restarting scann from davidgyoung's answer https://stackoverflow.com/questions/29731176/ble-scanning-callback-only-get-called-several-times-then-stopped

dariuszseweryn, 2018-09-28:
Not much can be done on the library side. Closing

closed 2018-09-28 by dariuszseweryn