dariuszseweryn/RxAndroidBle #463: connection.writeCharacteristic failed when onCharacteristicChanged has callback

Kilnn, 2018-07-28:
I have test with two diffrent command:
command A : writeCharacteristic and device response nothing
command B : writeCharacteristic and device response something with onCharacteristicChanged

Command A can work properly, every writeCharacteristic can succeed
Command B does not work properly, only the first writeCharacteristic can succeed, and often fails afterwards.

And the strange thing is that when I added a Thread.sleep(100) before the writeCharacteristic, everything is ok. I don't know the cause of this problem.

code with setupNotification:
```
@Override
protected Observable<?> rxPrepare(final RxBleConnection connection) {
    return connection
            .discoverServices()
            .flatMap(new Function<RxBleDeviceServices, SingleSource<BluetoothGattService>>() {
                @Override
                public SingleSource<BluetoothGattService> apply(RxBleDeviceServices rxBleDeviceServices) throws Exception {
                    return rxBleDeviceServices.getService(SERVICE_UUID);
                }
            })
            .toObservable()
            .flatMap(new Function<BluetoothGattService, ObservableSource<?>>() {
                @Override
                public ObservableSource<?> apply(BluetoothGattService bluetoothGattService) throws Exception {
                    mWriteCharacteristic = bluetoothGattService.getCharacteristic(WRITE_CHARACTERISTIC_UUID);
                    if (mWriteCharacteristic == null) {
                        throw new BleCharacteristicNotFoundException(WRITE_CHARACTERISTIC_UUID);
                    }
                    final BluetoothGattCharacteristic notifyCharacteristic = bluetoothGattService.getCharacteristic(NOTIFY_CHARACTERISTIC_UUID);
                    if (notifyCharacteristic == null) {
                        throw new BleCharacteristicNotFoundException(NOTIFY_CHARACTERISTIC_UUID);
                    }
                    return connection.setupNotification(notifyCharacteristic, NotificationSetupMode.DEFAULT)
                            .doOnNext(new Consumer<Observable<byte[]>>() {
                                @Override
                                public void accept(Observable<byte[]> observable) throws Exception {
                                    doListenCharacteristicChanged(observable);
                                }
                            });
                }
            })
            .retryWhen(new RetryWithDelay(1000, 3, new Predicate<Throwable>() {
                @Override
                public boolean test(Throwable throwable) throws Exception {
                    return mBluetoothAdapter != null && mBluetoothAdapter.isEnabled();
                }
            }));
}

@SuppressLint("CheckResult")
private void doListenCharacteristicChanged(Observable<byte[]> observable) {
    observable
            .subscribe(new Consumer<byte[]>() {
                @Override
                public void accept(byte[] bytes) throws Exception {
                    onReceiveData(bytes);
                }
            }, new Consumer<Throwable>() {
                @Override
                public void accept(Throwable throwable) throws Exception {
                    Log.e(TAG, "throwable:" + throwable + " Thread:" + Thread.currentThread().getName());
                }
            });
}
```

code with send command:
```
public boolean sendDataSync(@NonNull byte[] data) {
    RxBleConnection connection = getRxBleConnection();
    if (!isConnected() || connection == null || mWriteCharacteristic == null) {
        return false;
    } else {
        if (WristbandApplication.isDebugEnable())
            Log.e(TAG, "Send Data :" + BytesUtil.byte2HexStr(data));
        boolean success = true;
        try {
            //Thread.sleep(100); //Add a sleep ,at then everything is ok.???
            if (data.length <= MTU_SIZE) {
                connection.writeCharacteristic(mWriteCharacteristic, data).blockingGet();
            } else {
                RxBleConnection.LongWriteOperationBuilder builder = connection.createNewLongWriteBuilder();
                builder.setBytes(data);
                builder.setCharacteristic(mWriteCharacteristic);
                builder.setMaxBatchSize(MTU_SIZE);
                builder.build().blockingSubscribe();
            }
        } catch (Exception e) {
            e.printStackTrace();
            success = false;
        }
        return success;
    }
}
```

test code:
```
public void set_user_info() {
    boolean success = mWristManager.sendDataSync(datas);
    Log.e(TAG, "sendDataSync result:" + success);

    success = mWristManager.sendDataSync(datas);
    Log.e(TAG, "sendDataSync result:" + success);

    success = mWristManager.sendDataSync(datas);
    Log.e(TAG, "sendDataSync result:" + success);

    success = mWristManager.sendDataSync(datas);
    Log.e(TAG, "sendDataSync result:" + success);

    success = mWristManager.sendDataSync(datas);
    Log.e(TAG, "sendDataSync result:" + success);
}
```

error:
```
com.polidea.rxandroidble2.exceptions.BleGattCannotStartException: GATT exception from MAC address 10:00:A1:12:09:AC, with type BleGattOperation{description='CHARACTERISTIC_LONG_WRITE'}
    at com.polidea.rxandroidble2.internal.operations.CharacteristicLongWriteOperation.writeData(CharacteristicLongWriteOperation.java:177)
    at com.polidea.rxandroidble2.internal.operations.CharacteristicLongWriteOperation.access$200(CharacteristicLongWriteOperation.java:44)
    at com.polidea.rxandroidble2.internal.operations.CharacteristicLongWriteOperation$2.subscribe(CharacteristicLongWriteOperation.java:155)
    at io.reactivex.internal.operators.observable.ObservableCreate.subscribeActual(ObservableCreate.java:40)
    at io.reactivex.Observable.subscribe(Observable.java:12051)
    at io.reactivex.internal.operators.observable.ObservableSubscribeOn$SubscribeTask.run(ObservableSubscribeOn.java:96)
    at io.reactivex.internal.schedulers.ScheduledDirectTask.call(ScheduledDirectTask.java:38)
    at io.reactivex.internal.schedulers.ScheduledDirectTask.call(ScheduledDirectTask.java:26)
    at java.util.concurrent.FutureTask.run(FutureTask.java:237)
    at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
    at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
    at java.lang.Thread.run(Thread.java:762)
```

dariuszseweryn, 2018-07-29:
Hello!

Thank you for using the library.

This is a general usage question and as such should be posted on [stackoverflow](https://stackoverflow.com/questions/tagged/rxandroidble) with `rxandroidble` tag—it will be easier to find by other users.

Best Regards

Kilnn, 2018-07-30:
After many tests, I found that it was not caused by the onCharacteristicChanged callback, but by LongWriteOperation. I think it might be that LongWriteOperation didn't end properly, and the Operation queue started the next operation. Using writeCharacteristic instead of LongWriteOperation will work fine. So I think this might be a bug

dariuszseweryn, 2018-07-30:
In this case — could you add logs from your application with `RxBleLog.setLogLevel(RxBleLog.VERBOSE)` and as many other data that is described in the [issue template](https://github.com/Polidea/RxAndroidBle/blob/master/.github/ISSUE_TEMPLATE.md)?

This will help understand what exactly is going on

Kilnn, 2018-07-30:
Sorry, I read the issue template, but I missed some details. Below are all the logs, but it looks like there are no process errors.

```
07-30 22:33:45.838 6039-6039/com.htsmart.wristband2sample E/GattLayer: Send Data :00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
07-30 22:33:46.347 6039-6039/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: QUEUED   CharacteristicLongWriteOperation(103365458)
07-30 22:33:46.351 6039-6074/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: STARTED  CharacteristicLongWriteOperation(103365458)
07-30 22:33:46.446 6039-6052/com.htsmart.wristband2sample D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=0000ff02-0000-1000-8000-00805f9b34fb status=0
07-30 22:33:46.543 6039-6060/com.htsmart.wristband2sample D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=0000ff02-0000-1000-8000-00805f9b34fb status=0
07-30 22:33:46.550 6039-6039/com.htsmart.wristband2sample E/GattLayer: Send Data :00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
07-30 22:33:46.552 6039-6074/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: FINISHED CharacteristicLongWriteOperation(103365458) in 200 ms
07-30 22:33:47.055 6039-6039/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: QUEUED   CharacteristicLongWriteOperation(64662860)
07-30 22:33:47.060 6039-6074/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: STARTED  CharacteristicLongWriteOperation(64662860)
07-30 22:33:47.178 6039-6051/com.htsmart.wristband2sample D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=0000ff02-0000-1000-8000-00805f9b34fb status=0
07-30 22:33:47.274 6039-6052/com.htsmart.wristband2sample D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=0000ff02-0000-1000-8000-00805f9b34fb status=0
07-30 22:33:47.281 6039-6039/com.htsmart.wristband2sample E/GattLayer: Send Data :00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
07-30 22:33:47.284 6039-6074/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: FINISHED CharacteristicLongWriteOperation(64662860) in 225 ms
07-30 22:33:47.788 6039-6039/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: QUEUED   CharacteristicLongWriteOperation(67534710)
07-30 22:33:47.792 6039-6074/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: STARTED  CharacteristicLongWriteOperation(67534710)
07-30 22:33:47.908 6039-6060/com.htsmart.wristband2sample D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=0000ff02-0000-1000-8000-00805f9b34fb status=0
07-30 22:33:48.004 6039-6051/com.htsmart.wristband2sample D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=0000ff02-0000-1000-8000-00805f9b34fb status=0
07-30 22:33:48.016 6039-6074/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: FINISHED CharacteristicLongWriteOperation(67534710) in 223 ms
07-30 22:33:48.019 6039-6039/com.htsmart.wristband2sample E/GattLayer: Send Data :00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
07-30 22:33:48.103 6039-6052/com.htsmart.wristband2sample D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=0000ff02-0000-1000-8000-00805f9b34fb status=0
07-30 22:33:48.116 6039-6044/com.htsmart.wristband2sample I/art: Do partial code cache collection, code=25KB, data=30KB
07-30 22:33:48.117 6039-6044/com.htsmart.wristband2sample I/art: After code cache collection, code=25KB, data=30KB
    Increasing code cache capacity to 128KB
07-30 22:33:48.525 6039-6039/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: QUEUED   CharacteristicLongWriteOperation(220868688)
07-30 22:33:48.531 6039-6074/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: STARTED  CharacteristicLongWriteOperation(220868688)
07-30 22:33:48.639 6039-6060/com.htsmart.wristband2sample D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=0000ff02-0000-1000-8000-00805f9b34fb status=0
07-30 22:33:48.740 6039-6051/com.htsmart.wristband2sample D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=0000ff02-0000-1000-8000-00805f9b34fb status=0
07-30 22:33:48.753 6039-6074/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: FINISHED CharacteristicLongWriteOperation(220868688) in 221 ms
07-30 22:33:48.757 6039-6039/com.htsmart.wristband2sample E/GattLayer: Send Data :00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
07-30 22:33:48.882 6039-6052/com.htsmart.wristband2sample D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=0000ff02-0000-1000-8000-00805f9b34fb status=0
07-30 22:33:49.131 6039-6060/com.htsmart.wristband2sample D/BluetoothGatt: onClientConnParamsChanged() - Device=10:00:A1:12:09:XX interval=240 status=0
07-30 22:33:49.263 6039-6039/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: QUEUED   CharacteristicLongWriteOperation(190007898)
07-30 22:33:49.267 6039-6074/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: STARTED  CharacteristicLongWriteOperation(190007898)
07-30 22:33:50.029 6039-6051/com.htsmart.wristband2sample D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=0000ff02-0000-1000-8000-00805f9b34fb status=0
07-30 22:33:50.631 6039-6052/com.htsmart.wristband2sample D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=0000ff02-0000-1000-8000-00805f9b34fb status=0
[... cut ...]
        at java.lang.Thread.run(Thread.java:762)
07-30 22:34:04.979 6039-6074/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: FINISHED CharacteristicLongWriteOperation(27215022) in 24 ms
07-30 22:34:04.980 6039-6039/com.htsmart.wristband2sample I/Choreographer: Skipped 376 frames!  The application may be doing too much work on its main thread.
07-30 22:34:05.031 6039-6051/com.htsmart.wristband2sample D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=0000ff02-0000-1000-8000-00805f9b34fb status=0
```

dariuszseweryn, 2018-07-30:
This looks like a threading issue. The `BluetoothGatt` seems to be still busy even when the `onCharacteristicWrite()` callback was already called. Or there is an "off-by-one" mistake in the code... I'll try to prepare a snapshot release Tomorrow with more logs which would clarify the situation.

In the first situation the fix will not be made quickly, in the second — I should be able to address it quite shortly.

If you could also add as the issue template suggest the version of the library it would be nice. The info about on what phone model/OS version you have observed the issue may also be valuable.

Kilnn, 2018-07-31:
I used two mobile phones to test, and this problem will appear on both phones.
Samsung SM-G9300,Android 7.0
Smartisan OS105,Android 7.1.1

Kilnn, 2018-07-31:
I have test many times with the release 1.7.0-SNAPSHOT on Smartisan OS105(Android 7.1.1). My test code send data with 30 bytes, and the LongWriteOperation maxBatchSize is 20. And i found the log has three batch write operation when error occur, maybe it is the problem caused by this place. When the log `Long write batch #2` appear, the error occur.

```
08-01 05:56:16.093 8375-8728/com.htsmart.wristband2sample D/RxBle#ClientOperationQueue: QUEUED   ConnectOperation(156172013)
08-01 05:56:16.094 8375-8394/com.htsmart.wristband2sample D/RxBle#ClientOperationQueue: STARTED  ConnectOperation(156172013)
08-01 05:56:16.098 8375-8396/com.htsmart.wristband2sample V/RxBle#BleConnectionCompat: Connecting without reflection
08-01 05:56:16.102 8375-8396/com.htsmart.wristband2sample D/BluetoothGatt: connect() - device: 00:E0:12:34:56:87, auto: false
    registerApp()
    registerApp() - UUID=7ba0e6b7-0c0a-4940-9083-042c61b0ce19
08-01 05:56:16.104 8375-8390/com.htsmart.wristband2sample D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
08-01 05:56:17.541 8375-8389/com.htsmart.wristband2sample D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=00:E0:12:34:56:87
08-01 05:56:17.543 8375-8389/com.htsmart.wristband2sample D/RxBle#BluetoothGatt: onConnectionStateChange newState=2 status=0
08-01 05:56:17.545 8375-8394/com.htsmart.wristband2sample D/RxBle#ClientOperationQueue: FINISHED ConnectOperation(156172013) in 1451 ms
08-01 05:56:18.049 8375-8730/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: QUEUED   ServiceDiscoveryOperation(78224617)
08-01 05:56:18.050 8375-8727/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: STARTED  ServiceDiscoveryOperation(78224617)
08-01 05:56:18.052 8375-8396/com.htsmart.wristband2sample D/BluetoothGatt: discoverServices() - device: 00:E0:12:34:56:87
08-01 05:56:18.385 8375-8390/com.htsmart.wristband2sample D/BluetoothGatt: onSearchComplete() = Device=00:E0:12:34:56:87 Status=0
08-01 05:56:18.385 8375-8390/com.htsmart.wristband2sample D/RxBle#BluetoothGatt: onServicesDiscovered status=0
08-01 05:56:18.386 8375-8728/com.htsmart.wristband2sample V/RxBle#ServiceDiscoveryOperation: Preparing services description
08-01 05:56:18.392 8375-8728/com.htsmart.wristband2sample V/RxBle#ServiceDiscoveryOperation: --------------- ====== Printing peripheral content ====== ---------------
    PERIPHERAL ADDRESS: 00:E0:12:34:56:87
    PERIPHERAL NAME: HBracelet-687
    -------------------------------------------------------------------------
    
    Primary Service - Generic Access (00001800-0000-1000-8000-00805f9b34fb)
    Instance ID: 1
    -> Characteristics:
        * Device Name (00002a00-0000-1000-8000-00805f9b34fb)
          Properties: [ READ ]
        * Appearance (00002a01-0000-1000-8000-00805f9b34fb)
          Properties: [ READ ]
        * Peripheral Preferred Connection Parameters (00002a04-0000-1000-8000-00805f9b34fb)
          Properties: [ READ ]
    
    Primary Service - Unknown service (000001ff-3c17-d293-8e48-14fe2e4da212)
    Instance ID: 8
    -> Characteristics:
        * Unknown characteristic (0000ff02-0000-1000-8000-00805f9b34fb)
          Properties: [ WRITE ]
        * Unknown characteristic (0000ff03-0000-1000-8000-00805f9b34fb)
          Properties: [ READ WRITE_NO_RESPONSE NOTIFY ]
          -> Descriptors: 
            * Client Characteristic Configuration (00002902-0000-1000-8000-00805f9b34fb)
        * Unknown characteristic (0000ff04-0000-1000-8000-00805f9b34fb)
          Properties: [ READ WRITE ]
    
    Primary Service - Unknown service (0000fee7-0000-1000-8000-00805f9b34fb)
    Instance ID: 16
    -> Characteristics:
        * Unknown characteristic (0000fec9-0000-1000-8000-00805f9b34fb)
          Properties: [ READ ]
        * Unknown characteristic (0000fea1-0000-1000-8000-00805f9b34fb)
          Properties: [ READ NOTIFY ]
          -> Descriptors: 
            * Client Characteristic Configuration (00002902-0000-1000-8000-00805f9b34fb)
    
    Primary Service - Unknown service (0000d0ff-3c17-d293-8e48-14fe2e4da212)
    Instance ID: 22
    -> Characteristics:
        * Unknown characteristic (0000ffd1-0000-1000-8000-00805f9b34fb)
          Properties: [ WRITE_NO_RESPONSE ]
        * Unknown characteristic (0000ffd2-0000-1000-8000-00805f9b34fb)
          Properties: [ READ ]
        * Unknown characteristic (0000ffd3-0000-1000-8000-00805f9b34fb)
          Properties: [ READ ]
        * Unknown characteristic (0000ffd4-0000-1000-8000-00805f9b34fb)
          Properties: [ READ ]
        * Unknown characteristic (0000ffd5-0000-1000-8000-00805f9b34fb)
          Properties: [ READ ]
        * Unknown characteristic (0000fff1-0000-1000-8000-00805f9b34fb)
[... 71 lines cut ...]
08-01 05:56:37.801 8375-8375/com.htsmart.wristband2sample D/GattLayer: Send Data :00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
08-01 05:56:37.802 8375-8727/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: FINISHED CharacteristicLongWriteOperation(118313718) in 180 ms
08-01 05:56:37.802 8375-8396/com.htsmart.wristband2sample D/RxBle#CharacteristicLongWriteOperation: Writing next batch
08-01 05:56:37.802 8375-8375/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: QUEUED   CharacteristicLongWriteOperation(44984272)
08-01 05:56:37.803 8375-8727/com.htsmart.wristband2sample D/RxBle#ConnectionOperationQueue: STARTED  CharacteristicLongWriteOperation(44984272)
[... cut ...]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
        at java.lang.Thread.run(Thread.java:761)
08-01 05:56:37.807 8375-8375/com.htsmart.wristband2sample I/Choreographer: Skipped 47 frames!  The application may be doing too much work on its main thread.
08-01 05:56:37.931 8375-8390/com.htsmart.wristband2sample D/RxBle#BluetoothGatt: onCharacteristicWrite characteristic=0000ff02-0000-1000-8000-00805f9b34fb status=0
```

closed 2018-08-03 by dariuszseweryn

dariuszseweryn, 2018-08-03:
I think I found the culprit. I have just released a `1.7.0` library version. It should be available shortly and have the bug fixed.

Thank you very much for spotting and helping with tests

Kilnn, 2018-08-04:
Great, thanks for you help.

PR #465 (2018-08-03): Fixed CharacteristicLongWriteOperation repetition logic
It was susceptible to a race condition due to a repetition observable using `.takeUntil()` which emits a value before checking the complete condition. This could lead to an excess characteristic write being executed that was not secured on a connection queue leading to next operation being rejected.
Fixes #463
commit 2fa21d68 (2018-08-03): Fixed CharacteristicLongWriteOperation repetition logic (#465)

It was susceptible to a race condition due to a repetition observable using `.takeUntil()` which emits a value before checking the complete condition. This could lead to an excess characteristic write being executed was not secured on a connection queue.
Fixes #463