dariuszseweryn/RxAndroidBle #466: BleAlreadyConnectedException throws when I try to connect a device

JoeSteven, 2018-08-07:
I disposed the connection disposable after exit Activity,but still got BleAlreadyConnectedException when try to connect a device. here is the log, you can see unregisterApp() invoke after exit Activity。
Also got this exception when connect to a device that never connect before
```
08-07 18:02:10.297 3804-4055/com.joey.rxble D/BluetoothGatt: connect() - device: DC:2C:26:00:09:90, auto: false
08-07 18:02:10.298 3804-4055/com.joey.rxble D/BluetoothGatt: registerApp()
    registerApp() - UUID=208be5ca-e68e-4527-b7eb-04c4cc4fc480
08-07 18:02:10.301 3804-3830/com.joey.rxble D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
08-07 18:02:13.658 3804-3830/com.joey.rxble D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=DC:2C:26:00:09:90
08-07 18:02:13.666 3804-4055/com.joey.rxble D/BluetoothGatt: discoverServices() - device: DC:2C:26:00:09:90
08-07 18:02:14.285 3804-3830/com.joey.rxble D/BluetoothGatt: onConnectionUpdated() - Device=DC:2C:26:00:09:90 interval=6 latency=0 timeout=500 status=0
08-07 18:02:14.590 3804-3830/com.joey.rxble D/BluetoothGatt: onSearchComplete() = Device=DC:2C:26:00:09:90 Status=0
08-07 18:02:14.671 3804-3821/com.joey.rxble I/zygote64: Do full code cache collection, code=502KB, data=349KB
    After code cache collection, code=502KB, data=299KB
08-07 18:02:14.690 3804-3830/com.joey.rxble D/BluetoothGatt: onConnectionUpdated() - Device=DC:2C:26:00:09:90 interval=9 latency=0 timeout=300 status=0
08-07 18:02:21.863 3804-4055/com.joey.rxble D/BluetoothManager: getConnectionState()
    getConnectedDevices
08-07 18:02:21.866 3804-4055/com.joey.rxble D/BluetoothGatt: cancelOpen() - device: DC:2C:26:00:09:90
08-07 18:02:21.867 3804-3830/com.joey.rxble D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=DC:2C:26:00:09:90
08-07 18:02:21.869 3804-4055/com.joey.rxble D/BluetoothGatt: close()
    unregisterApp() - mClientIf=5

08-07 18:02:43.469 3804-4055/com.joey.rxble D/BluetoothGatt: connect() - device: DC:2C:26:00:09:90, auto: false
    registerApp()
08-07 18:02:43.470 3804-4055/com.joey.rxble D/BluetoothGatt: registerApp() - UUID=2f04d40e-528b-469f-b03c-f37c1943405b
08-07 18:02:43.471 3804-3830/com.joey.rxble D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
08-07 18:02:44.124 3804-3830/com.joey.rxble D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=5 device=DC:2C:26:00:09:90
08-07 18:02:44.134 3804-3830/com.joey.rxble D/RxBleDemo: retry connection
08-07 18:02:44.135 3804-3830/com.joey.rxble D/RxBleDemo: retry connection
08-07 18:02:44.136 3804-3804/com.joey.rxble W/System.err: com.polidea.rxandroidble2.exceptions.BleAlreadyConnectedException: Already connected to device with MAC address DC:2C:26:00:09:90
        at com.polidea.rxandroidble2.internal.RxBleDeviceImpl$1.call(RxBleDeviceImpl.java:83)
08-07 18:02:44.137 3804-3804/com.joey.rxble W/System.err:     at com.polidea.rxandroidble2.internal.RxBleDeviceImpl$1.call(RxBleDeviceImpl.java:71)
08-07 18:02:44.137 3804-4055/com.joey.rxble D/BluetoothManager: getConnectionState()
    getConnectedDevices
08-07 18:02:44.137 3804-3804/com.joey.rxble W/System.err:     at io.reactivex.internal.operators.observable.ObservableDefer.subscribeActual(ObservableDefer.java:32)
        at io.reactivex.Observable.subscribe(Observable.java:12051)
        at io.reactivex.internal.operators.observable.ObservableOnErrorNext.subscribeActual(ObservableOnErrorNext.java:38)
        at io.reactivex.Observable.subscribe(Observable.java:12051)
        at io.reactivex.internal.operators.observable.ObservableDoOnLifecycle.subscribeActual(ObservableDoOnLifecycle.java:33)
        at io.reactivex.Observable.subscribe(Observable.java:12051)
08-07 18:02:44.138 3804-3804/com.joey.rxble W/System.err:     at io.reactivex.internal.operators.observable.ObservableDoOnLifecycle.subscribeActual(ObservableDoOnLifecycle.java:33)
        at io.reactivex.Observable.subscribe(Observable.java:12051)
        at io.reactivex.internal.operators.observable.ObservableDoOnEach.subscribeActual(ObservableDoOnEach.java:42)
        at io.reactivex.Observable.subscribe(Observable.java:12051)
        at io.reactivex.internal.operators.observable.ObservableRetryPredicate$RepeatObserver.subscribeNext(ObservableRetryPredicate.java:111)
        at io.reactivex.internal.operators.observable.ObservableRetryPredicate$RepeatObserver.onError(ObservableRetryPredicate.java:92)
        at io.reactivex.internal.operators.observable.ObservableDoOnEach$DoOnEachObserver.onError(ObservableDoOnEach.java:119)
        at io.reactivex.internal.observers.DisposableLambdaObserver.onError(DisposableLambdaObserver.java:64)
        at io.reactivex.internal.observers.DisposableLambdaObserver.onError(DisposableLambdaObserver.java:64)
08-07 18:02:44.139 3804-3804/com.joey.rxble W/System.err:     at io.reactivex.internal.operators.observable.ObservableOnErrorNext$OnErrorNextObserver.onError(ObservableOnErrorNext.java:78)
        at io.reactivex.internal.disposables.EmptyDisposable.error(EmptyDisposable.java:63)
        at io.reactivex.internal.operators.observable.ObservableError.subscribeActual(ObservableError.java:37)
        at io.reactivex.Observable.subscribe(Observable.java:12051)
        at io.reactivex.internal.operators.observable.ObservableOnErrorNext$OnErrorNextObserver.onError(ObservableOnErrorNext.java:105)
        at io.reactivex.internal.operators.observable.ObservableDoFinally$DoFinallyObserver.onError(ObservableDoFinally.java:87)
        at io.reactivex.internal.operators.observable.ObservableUnsubscribeOn$UnsubscribeObserver.onError(ObservableUnsubscribeOn.java:70)
        at io.reactivex.internal.operators.observable.ObservableSubscribeOn$SubscribeOnObserver.onError(ObservableSubscribeOn.java:63)
        at io.reactivex.internal.operators.observable.ObservableDoFinally$DoFinallyObserver.onError(ObservableDoFinally.java:87)
08-07 18:02:44.140 3804-3804/com.joey.rxble W/System.err:     at io.reactivex.internal.observers.DisposableLambdaObserver.onError(DisposableLambdaObserver.java:64)
        at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.checkTerminate(ObservableFlatMap.java:506)
08-07 18:02:44.140 3804-4055/com.joey.rxble D/BluetoothGatt: close()
08-07 18:02:44.140 3804-3804/com.joey.rxble W/System.err:     at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.drainLoop(ObservableFlatMap.java:335)
08-07 18:02:44.140 3804-4055/com.joey.rxble D/BluetoothGatt: unregisterApp() - mClientIf=5
08-07 18:02:44.140 3804-3804/com.joey.rxble W/System.err:     at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.drain(ObservableFlatMap.java:327)
        at io.reactivex.internal.operators.observable.ObservableFlatMap$InnerObserver.onError(ObservableFlatMap.java:582)
        at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.checkTerminate(ObservableFlatMap.java:506)
        at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.drainLoop(ObservableFlatMap.java:335)
        at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.drain(ObservableFlatMap.java:327)
        at io.reactivex.internal.operators.observable.ObservableFlatMap$InnerObserver.onError(ObservableFlatMap.java:582)
08-07 18:02:44.141 3804-3804/com.joey.rxble W/System.err:     at io.reactivex.internal.disposables.EmptyDisposable.error(EmptyDisposable.java:63)
        at io.reactivex.internal.operators.observable.ObservableError.subscribeActual(ObservableError.java:37)
        at io.reactivex.Observable.subscribe(Observable.java:12051)
        at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.subscribeInner(ObservableFlatMap.java:165)
        at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.onNext(ObservableFlatMap.java:139)
        at io.reactivex.internal.operators.observable.ObservableCreate$CreateEmitter.onNext(ObservableCreate.java:67)
        at com.polidea.rxandroidble2.internal.connection.DisconnectionRouter.notifySubscribersAboutException(DisconnectionRouter.java:125)
        at com.polidea.rxandroidble2.internal.connection.DisconnectionRouter.onExceptionOccurred(DisconnectionRouter.java:114)
        at com.polidea.rxandroidble2.internal.connection.DisconnectionRouter.onDisconnectedException(DisconnectionRouter.java:100)
[... cut ...]
        at android.bluetooth.BluetoothGatt.-wrap0(Unknown Source:0)
        at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:244)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:70)
        at android.os.Binder.execTransact(Binder.java:714)
08-07 18:02:44.154 3804-3804/com.joey.rxble D/RxBleDemo: com.polidea.rxandroidble2.exceptions.BleAlreadyConnectedException: Already connected to device with MAC address DC:2C:26:00:09:90
```

dariuszseweryn, 2018-08-07:
Hello,

Thank you for using the library.
Your post seems to be a bug report but it lacks some crucial information. Could you edit to contain information according to the [issue template](https://github.com/Polidea/RxAndroidBle/blob/master/.github/ISSUE_TEMPLATE.md)?

Best Regards

MarcinDe, 2018-08-09:
The problem lies here

**You try to connect to device**

```
08-07 18:02:43.469 3804-4055/com.joey.rxble D/BluetoothGatt: connect() - device: DC:2C:26:00:09:90, auto: false
    registerApp()
08-07 18:02:43.470 3804-4055/com.joey.rxble D/BluetoothGatt: registerApp() - UUID=2f04d40e-528b-469f-b03c-f37c1943405b
08-07 18:02:43.471 3804-3830/com.joey.rxble D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
```

**The connection is unsuccessful with error code 133**

```
08-07 18:02:44.124 3804-3830/com.joey.rxble D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=5 device=DC:2C:26:00:09:90
```

**You retry connection twice, first time im assuming it connects and the second will throw the error**

```
08-07 18:02:44.134 3804-3830/com.joey.rxble D/RxBleDemo: retry connection
08-07 18:02:44.135 3804-3830/com.joey.rxble D/RxBleDemo: retry connection
```

**Here is the error**

```
08-07 18:02:44.136 3804-3804/com.joey.rxble W/System.err: com.polidea.rxandroidble2.exceptions.BleAlreadyConnectedException: Already connected to device with MAC address DC:2C:26:00:09:90
```

dariuszseweryn, 2018-08-10:
Judging from [the message of this commit](https://github.com/JoeSteven/RxBle/commit/f97171cbd262484bca9544d17e04a8ddb0a73b15) and the original author silence I assume that the issue is already solved.

Closing

closed 2018-08-10 by dariuszseweryn

JoeSteven, 2018-08-13:
@MarcinDe  thanks man ! I also found this problem and fixed after I changed the reconnect strategy.
@dariuszseweryn sorry for I didn't reply immediately