dariuszseweryn/RxAndroidBle #471: BleGattCharacteristicException

SoNep, 2018-08-22:
### Summary
Can not write characterictic
### Library version
`1.7.0`

#### Preconditions
establishConnection

#### Steps to reproduce actual result
<br getBleDevice />1.
<br establishConnection />2.
<br writeCharacteristic />3.
#### Minimum code snippet reproducing the issue
`device.establishConnection(false)
                            .flatMapSingle { rxBleConnection ->
                                rxBleConnection.writeCharacteristic(UUID.fromString("00010003-1000-11e5-8000-70b3d5ab9052"), byteArray)
                            }
                            .doOnError { it.printStackTrace() }
`
#### Logs from the application running with setting `RxBleLog.setLogLevel(RxBleLog.VERBOSE)`
* Unknown characteristic (00010003-1000-11e5-8000-70b3d5ab9052)
    	  Properties: [ WRITE ]

08-22 16:21:11.602 19333-19423/com.D/RxBle#ConnectionOperationQueue: QUEUED   CharacteristicWriteOperation(99183906)
08-22 16:21:11.603 19333-19422/com.D/RxBle#ConnectionOperationQueue: FINISHED ServiceDiscoveryOperation(233563396) in 1391 ms
    STARTED  CharacteristicWriteOperation(99183906)
08-22 16:21:11.611 19333-19370/com.onConnectionUpdated() - Device=1C:A0:D3:5F:3D:B2 interval=36 latency=0 timeout=500 status=0
08-22 16:21:11.705 19333-19370/com. D/RxBle#BluetoothGatt$1: onCharacteristicWrite characteristic=00010003-1000-11e5-8000-70b3d5ab9052 status=6
08-22 16:21:11.714 19333-19423/com. W/System.err: com.polidea.rxandroidble2.exceptions.BleGattCharacteristicException: GATT exception from MAC address 1C:A0:D3:5F:3D:B2, status 6 (GATT_REQ_NOT_SUPPORTED), type BleGattOperation{description='CHARACTERISTIC_WRITE'}. (Look up status 0x06 here https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h)
        at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback.propagateErrorIfOccurred(RxBleGattCallback.java:228)
        at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback.access$800(RxBleGattCallback.java:33)
08-22 16:21:11.715 19333-19423/com. W/System.err:     at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onCharacteristicWrite(RxBleGattCallback.java:122)
        at android.bluetooth.BluetoothGatt$1$7.run(BluetoothGatt.java:438)
        at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:770)
08-22 16:21:11.715 19333-19422/com.dynamiccontrols.poc.steve D/RxBle#ConnectionOperationQueue: FINISHED CharacteristicWriteOperation(99183906) in 110 ms
08-22 16:21:11.715 19333-19423/com W/System.err:     at android.bluetooth.BluetoothGatt.access$200(BluetoothGatt.java:39)
        at android.bluetooth.BluetoothGatt$1.onCharacteristicWrite(BluetoothGatt.java:433)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:137)
        at android.os.Binder.execTransact(Binder.java:731)
08-22 16:21:11.722 19333-19423/com D/RxBle#ClientOperationQueue: QUEUED   DisconnectOperation(181225705)
08-22 16:21:11.724 19333-19410/com. D/RxBle#ClientOperationQueue: STARTED  DisconnectOperation(181225705)
08-22 16:21:11.725 19333-19424/com.D/BluetoothManager: getConnectionState()
08-22 16:21:11.726 19333-19424/com. D/BluetoothManager: getConnectedDevices
08-22 16:21:11.729 19333-19423/com. I/RxBle#ConnectionOperationQueue: Connection operations queue to be terminated (1C:A0:D3:5F:3D:B2)
08-22 16:21:11.731 19333-19424/com. D/BluetoothGatt: cancelOpen() - device: 1C:A0:D3:5F:3D:B2
08-22 16:21:11.732 19333-19422/com. D/RxBle#Executors$RunnableAdapter: Terminated.
08-22 16:21:11.735 19333-19370/com. D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=1C:A0:D3:5F:3D:B2
08-22 16:21:11.736 19333-19370/com. D/RxBle#BluetoothGatt$1: onConnectionStateChange newState=0 status=0
08-22 16:21:11.739 19333-19424/com. D/BluetoothGatt: close()
 
#### Actual result
status 6 (GATT_REQ_NOT_SUPPORTED), type BleGattOperation{description='CHARACTERISTIC_WRITE'}.

#### Expected result
should write characteristic

dariuszseweryn, 2018-09-03:
Hello,

Why do you think it is an issue with the library? From the logs it seems that the Android OS is returning an error code. Judging from the timings the request was sent via radio and the peripheral responded with an error code.

SoNep, 2018-09-04:
Thank you for your response. 
I found problem with ble device need to set up MTU.

closed 2018-09-04 by dariuszseweryn