dariuszseweryn/RxAndroidBle #475: Table of Observable behaviors in ReadMe needs updating

RobLewis, 2018-09-02:
For example: in v.2 `discoverServices()` returns a `Single`, not an `Observable`. This can be confused with the "Number of values" column in the table. 

Maybe the easiest solution would be to add a column listing the kind of object returned by each function. And perhaps change "Single" to "One" in the "Number of values" column.

dariuszseweryn, 2018-09-03:
I have pushed `fix/readme_observable_table` branch which has an updated paragraph `Observable behaviour`. Feel free to check the readme on this branch and give feedback if it is now clear enough

RobLewis, 2018-09-09:
Comments: 
 

1. Can you explain why `scanBleDevices()` returns a cold `Observable`? Does this mean that a new subscriber receives the entire list of devices discovered since the scan began? 

2. I assume that all the items removed from the table for Rx2 have been converted into `Single`s and there is no need to further describe them? 

3. I'm having trouble parsing this where it says "i.e.2": 

> ** this Observable may complete i.e. observeStateChanges() does emit only a single value and finishes in exactly one situation — when Bluetooth Adapter is not available on the device. There is no reason to monitor other states as the adapter does not appear during runtime. i.e.2 Observables emitted from setupNotification / setupIndication may complete when the parent Observable is disposed.

Could you clarify a bit? Are you meaning to say "two Observables emitted…"

RobLewis, 2018-09-11:
1. My understanding is that a cold `Observable` emits the same set of items to every `Subscriber`. By contrast, a hot `Observable` "joins the program in progress" when a `Subscriber` subscribes. Is this wrong? It seems to me that subscribing to a scan would only deliver new `ScanResult`s detected after the subscription. Does `scanBleDevices()` cache previous `ScanResult`s and deliver them to new `Subscriber`s. 

3. "i.e." is used to clarify something by restating it in a different way. Examples are introduced with "e.g.". Your rewrite seems fine. [Interesting: in "Write" mode this item is numbered "3", but when I switch to "Preview" it is converted to "2".]

dariuszseweryn, 2018-09-12:
1. As for the cold `Observable` it is to me that the producer inside the `Observable` in question is activated at the time of subscription. Which is also what can be read in a [referenced site](https://medium.com/@benlesh/hot-vs-cold-observables-f8094ed53339)

> Does scanBleDevices() cache previous ScanResults and deliver them to new Subscribers.

No, it does not cache previous results.

RobLewis, 2018-09-12:
Thanks. I read the article, which seems to say that cold `Observable`s are unicast, while hot ones may be multicast. This has been a source of some confusion to me: which `Observable`s support multiple `Subscriber`s? How can we identify them? Does `scanBleDevices()`, for example? Clearly its producer is activated at subscription; does this mean that multiple subscriptions create multiple instances of the scanner? Is this even possible? Is an error generated if you try to have multiple `Subscriber`s? Could you use `.share()`? [sorry for too many questions]

dariuszseweryn, 2018-09-13:
> which `Observable`s support multiple Subscribers?

All of them. But they may behave differently i.e. cold will usually start something for each subscription, hot will just connect to something that is happening.
> How can we identify them?

Hopefully `hot/cold` differentiation in the readme will do the trick for this library
> does this mean that multiple subscriptions create multiple instances of the scanner? Is this even possible?

It's depending on the implementation but I would expect that yes — it may be possible
> Is an error generated if you try to have multiple `Subscribers`?

Only when the user would try to subscribe to `RxBleDevice.establishConnection()` more than once
> Could you use `.share()`?

`.share()` may be used at will even for `RxBleDevice.establishConnection()` as it will get then subscribed only once

RobLewis, 2018-09-21:
Are you awaiting feedback from others? I'm OK with the update.

dariuszseweryn, 2018-09-21:
Unfortunately I do not have time on every day to work on the library. It is mostly Fridays. I am currently working on issues and actually wanted to merge the changes Today.