dariuszseweryn/RxAndroidBle #477: RxAndroidBle on Vive Focus

Montross7, 2018-09-04:
We are attempting to port our Android app onto the Vive Focus, but when starting the Bluetooth scan the app returns a LOCATION_SERVICES_DISABLED error. This does not seem to be a permissions issue, as I can use the native Android Bluetooth library and scan and detect devices without any issue. 
After some research, I found the following issue #142, which seems to be similar to what is happening on the Vive Focus, as it also does not have GPS or any location settings.

Is there a way of solving this issue for the Vive Focus device?

dariuszseweryn, 2018-09-04:
What version of the library do you use?

Montross7, 2018-09-04:
We are using version 1.4.3.

dariuszseweryn, 2018-09-04:
1.5.0 has an improved check for `Location Services`. If it still would not work then either one needs to enable `Location Services` (do not confuse with `Location Permissions`) on your Vive or make a fork that will suppress the `Location Services` check.

Montross7, 2018-09-05:
Thank you! Updated the library and now everything is working.

closed 2018-09-05 by Montross7