dariuszseweryn/RxAndroidBle #481: NPE when trying to stop scan in background using BackgroundScanner when Bluetooth is disabled

gryffindor14, 2018-09-17:
### Summary
When Bluetooth is not enabled, calling client.getBackgroundScanner().stopScan(Pending) causes NPE

I assume this is due to BluetoothAdapter#getBluetoothLEScanner() returns null if the BluetoothAdapter is off. 
### Library version
`1.7.0`

#### Preconditions
Bluetooth is off
#### Steps to reproduce actual result
<br />1. startScanningWith background scanner (client.getBackgroundScanner().scanBleDeviceInBackground(settings, filter, PendingIntent)
<br />2. turn off bluetooth
<br />3. stop scanning in background client.getBackgroundScanner().stopScanningInBackground(PendingIntent)
#### Minimum code snippet reproducing the issue

#### Logs from the application running with setting `RxBleLog.setLogLevel(RxBleLog.VERBOSE)`
 Attempt to invoke virtual method 'void android.bluetooth.le.BluetoothLeScanner.stopScan(android.app.PendingIntent)' on a null object reference
        at com.polidea.rxandroidble2.internal.util.RxBleAdapterWrapper.stopLeScan(RxBleAdapterWrapper.java:62)
        at com.polidea.rxandroidble2.internal.scan.BackgroundScannerImpl.stopBackgroundBleScan(BackgroundScannerImpl.java:74)
        at com.validic.mobile.ble.BackgroundPeripheralScanner.stopScanning(BackgroundPeripheralScanner.java:49)
        at com.validic.mobile.ble.PassiveBluetoothManager.stopScanningInBackground(PassiveBluetoothManager.java:178)
#### Actual result
Uncatchable NPE

#### Expected result
No Exception or BluetoothScanException with Code for Bluetooth being off

gryffindor14, 2018-09-25:
Somewhat related, there is a different internal exception when trying to start scanning in the background while bluetooth is disabled. 

java.lang.SecurityException: Need BLUETOOTH_PRIVILEGED permission: Neither user 10188 nor current process has android.permission.BLUETOOTH_PRIVILEGED.
        at android.os.Parcel.createException(Parcel.java:1942)
        at android.os.Parcel.readException(Parcel.java:1910)
        at android.os.Parcel.readException(Parcel.java:1860)
        at android.bluetooth.IBluetoothGatt$Stub$Proxy.startScanForIntent(IBluetoothGatt.java:977)
        at android.bluetooth.le.BluetoothLeScanner.startScan(BluetoothLeScanner.java:249)
        at android.bluetooth.le.BluetoothLeScanner.startScan(BluetoothLeScanner.java:166)
        at com.polidea.rxandroidble2.internal.util.RxBleAdapterWrapper.startLeScan(RxBleAdapterWrapper.java:57)
        at com.polidea.rxandroidble2.internal.scan.BackgroundScannerImpl.scanBleDeviceInBackground(BackgroundScannerImpl.java:56)

dariuszseweryn, 2018-09-26:
This case should also be handled by the fix on `fix/background_scanner_npe` branch as it checks if adapter is on. I have created a different issue as potentially a `SecurityException` may also be thrown when specific settings will be passed to the scan.

PR #487 (2018-09-28): Fixed NPE when calling BackgroundScanner while BT is OFF
Fixes #481
closed 2018-09-28 by dariuszseweryn