dariuszseweryn/RxAndroidBle #490: requestMTU() throws BleGattCallbackTimeoutException

amharms795, 2018-09-25:
### Summary
I seem to have a problem reconnecting to a device that has already been bonded to my phone. It connects fine if I manually unpair/unbond in my phones OS and try to connect again in the app. I have found the issue to be a problem with requestMTU(). If the device is bonded and tries to connect in my app, RxBleConnectionState will tell me the device is connected (not data is being received though); however after about 15 seconds I will get an exception from requestMTU() which is this: 

```
com.polidea.rxandroidble.exceptions.BleGattCallbackTimeoutException: GATT exception from MAC address B0:B4:48:DD:8A:60, with type BleGattOperation{description='ON_MTU_CHANGED'}
        at com.polidea.rxandroidble.internal.SingleResponseOperation.timeoutFallbackProcedure(SingleResponseOperation.java:87)
        at com.polidea.rxandroidble.internal.SingleResponseOperation.protectedRun(SingleResponseOperation.java:52)
        at com.polidea.rxandroidble.internal.QueueOperation$1.call(QueueOperation.java:42)
        at com.polidea.rxandroidble.internal.QueueOperation$1.call(QueueOperation.java:38)
        at rx.internal.operators.OnSubscribeCreate.call(OnSubscribeCreate.java:72)
        at rx.internal.operators.OnSubscribeCreate.call(OnSubscribeCreate.java:32)
        at rx.Observable.unsafeSubscribe(Observable.java:10256)
        at rx.internal.operators.OperatorSubscribeOn$SubscribeOnSubscriber.call(OperatorSubscribeOn.java:100)
        at rx.internal.schedulers.ScheduledAction.run(ScheduledAction.java:55)
        at rx.internal.schedulers.ExecutorScheduler$ExecutorSchedulerWorker.run(ExecutorScheduler.java:107)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
        at java.lang.Thread.run(Thread.java:762)
```

### Library version
`1.5.0`

#### Minimum code snippet reproducing the issue
```
private RxBleClient bleClient;
private SubscriptionList subList;
private RxBleDevice bleDevice;
private PublishSubject<byte[]> sendSubject = PublishSubject.create();
private ResponseSubject responseSubject = ResponseSubject.create();

private void connect(String macAddress, boolean autoConnect, final int mtu, final UUID notificationUuid, final boolean withAck, final UUID writeUuid, final Runnable afterNotif)
{
        disconnect();

        subList = new SubscriptionList();
        subList.add(setMacAddress(macAddress)
            .establishConnection(autoConnect)
            .flatMap(connection ->
            {
                if(Build.VERSION.SDK_INT >= Build.VERSION_CODES.LOLLIPOP)
                    return connection.requestMtu(mtu).flatMap(request -> Observable.just(connection));
                else
                    return Observable.just(connection);
            })
            .flatMap(connection ->
            {
                subList.add(sendSubject.flatMap(bytes -> connection.writeCharacteristic(writeUuid, bytes)).subscribe());
                return connection.setupNotification(notificationUuid);
            })
            .doOnNext(observable -> afterNotif.run())
            .flatMap(observable -> observable)changes.
            .subscribe(response -> responseSubject.onNext(response),
            throwable ->
            {
                Log.e(TAG,"Connection Error",throwable);
            }));
}

public void disconnect()
{
        if (subList != null)
            subList.unsubscribe();
}
```

amharms795, 2018-09-26:
Hmm okay, so what do you think a good fix would be? Ask for a slightly different MTU on each connect? The MTU on the device I am connecting to gets reset on connection loss so I have to change it on each new connect. If you need me to provide logs I can.

dariuszseweryn, 2018-09-26:
A good fix should be provided by Google as the contract here is not fulfilled. There is no callback in `BluetoothGattCallback.onMtuChange()` after calling `BluetoothGatt.requestMtu()`.

A workaround of asking for a slightly different MTU might work.

Yes, logs may be useful.

amharms795, 2018-09-26:
Here is all the relevant information from the log.

```
I/BtR100ConnectHandler: RxBleConnectionState{CONNECTING}
D/TextView: setTypeface with style : 0
D/ViewRootImpl@89dbd25[Toast]: ThreadedRenderer.create() translucent=true
D/InputTransport: Input channel constructed: fd=94
D/ViewRootImpl@89dbd25[Toast]: setView = android.widget.LinearLayout{a68fefa V.E...... ......I. 0,0-0,0} touchMode=true
D/ViewRootImpl@89dbd25[Toast]: dispatchAttachedToWindow
D/ViewRootImpl@89dbd25[Toast]: Relayout returned: oldFrame=[0,0][0,0] newFrame=[451,2128][989,2304] result=0x27 surface={isValid=true 547087768576} surfaceGenerationChanged=true
    mHardwareRenderer.initialize() mSurface={isValid=true 547087768576} hwInitialized=true
D/mali_winsys: EGLint new_window_surface(egl_winsys_display*, void*, EGLSurface, EGLConfig, egl_winsys_surface**, egl_color_buffer_format*, EGLBoolean) returns 0x3000,  [538x176]-format:1
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=7
D/ViewRootImpl@89dbd25[Toast]: MSG_RESIZED_REPORT: ci=Rect(0, 0 - 0, 0) vi=Rect(0, 0 - 0, 0) or=1
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=B0:B4:48:DD:8A:60
D/RxBle#BluetoothGatt: onConnectionStateChange newState=2 status=0
D/RxBle#ClientOperationQueue: FINISHED ConnectOperation(189967255) in 381 ms
I/BleManager: MTU VALUE: 23
I/BtR100ConnectHandler: RxBleConnectionState{CONNECTED}
D/RxBle#ConnectionOperationQueue: QUEUED   MtuRequestOperation(62394566)
D/RxBle#ConnectionOperationQueue: STARTED  MtuRequestOperation(62394566)
D/BluetoothGatt: configureMTU() - device: B0:B4:48:DD:8A:60 mtu: 247
I/art: Do partial code cache collection, code=28KB, data=29KB
I/art: After code cache collection, code=28KB, data=29KB
    Increasing code cache capacity to 128KB
D/BluetoothGatt: onClientConnParamsChanged() - Device=B0:B4:48:DD:8A:60 interval=6 status=0
D/BluetoothGatt: onClientConnParamsChanged() - Device=B0:B4:48:DD:8A:60 interval=39 status=0
D/ViewRootImpl@89dbd25[Toast]: mHardwareRenderer.destroy()#4
    dispatchDetachedFromWindow
D/InputTransport: Input channel destroyed: fd=94
D/RxBle#ConnectionOperationQueue: FINISHED MtuRequestOperation(62394566) in 30008 ms
E/BleManager: Connection Error
    com.polidea.rxandroidble.exceptions.BleGattCallbackTimeoutException: GATT exception from MAC address B0:B4:48:DD:8A:60, with type BleGattOperation{description='ON_MTU_CHANGED'}
        at com.polidea.rxandroidble.internal.SingleResponseOperation.timeoutFallbackProcedure(SingleResponseOperation.java:87)
        at com.polidea.rxandroidble.internal.SingleResponseOperation.protectedRun(SingleResponseOperation.java:52)
        at com.polidea.rxandroidble.internal.QueueOperation$1.call(QueueOperation.java:42)
        at com.polidea.rxandroidble.internal.QueueOperation$1.call(QueueOperation.java:38)
        at rx.internal.operators.OnSubscribeCreate.call(OnSubscribeCreate.java:72)
        at rx.internal.operators.OnSubscribeCreate.call(OnSubscribeCreate.java:32)
        at rx.Observable.unsafeSubscribe(Observable.java:10256)
        at rx.internal.operators.OperatorSubscribeOn$SubscribeOnSubscriber.call(OperatorSubscribeOn.java:100)
        at rx.internal.schedulers.ScheduledAction.run(ScheduledAction.java:55)
        at rx.internal.schedulers.ExecutorScheduler$ExecutorSchedulerWorker.run(ExecutorScheduler.java:107)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
        at java.lang.Thread.run(Thread.java:762)
I/RxBle#ConnectionOperationQueue: Connection operations queue to be terminated (B0:B4:48:DD:8A:60)
D/RxBle#ClientOperationQueue: QUEUED   DisconnectOperation(250689941)
D/RxBle#Executors$RunnableAdapter: Terminated.
D/RxBle#ClientOperationQueue: STARTED  DisconnectOperation(250689941)
D/BluetoothManager: getConnectionState()
    getConnectedDevices
I/BtR100ConnectHandler: RxBleConnectionState{DISCONNECTING}
D/BluetoothGatt: cancelOpen() - device: B0:B4:48:DD:8A:60
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=B0:B4:48:DD:8A:60
D/RxBle#BluetoothGatt: onConnectionStateChange newState=0 status=0
D/BluetoothGatt: close()
D/BluetoothGatt: unregisterApp() - mClientIf=7
I/BtR100ConnectHandler: RxBleConnectionState{DISCONNECTED}
D/RxBle#ClientOperationQueue: FINISHED DisconnectOperation(250689941) in 49 ms
```

dariuszseweryn, 2018-09-26:
And it only happens after the device is bonded? Please write your phone model and OS version.

amharms795, 2018-09-26:
Correct, this happens 100% of the time if the device is already bonded and works 100% of the time when it is not.

Model: Galaxy S6
OS: Nougat 7.0

dariuszseweryn, 2018-09-27:
I have moved the logs from a comment to the original post to match the issue template and for clarity. This is an Android bug and I do not see what can be done inside the library to mitigate it. I have not found any bug related to this case on the [google issue tracker](https://issuetracker.google.com/issues?q=android%20mtu%20(%22ble%22%20OR%20%22low%20energy%22)) but it may be worth to do so.

amharms795, 2018-09-27:
I have a "Hack Fix" that will work, which is never keep the device bonded on the android OS. However, then the user will get the pair device message on every connect. I also have some more information I would like to show. I will show the logs in order of the following:

```
1. Log of connecting while the device in unbonded/unpaired (which works).
2. Log of when I disconnect.
3. Log of when I reconnect (says its connected, but after 15 seconds throws the 'MTU Change' Exception).
4. Log of when I disconnect before the exception happens (before the 15 seconds)
5. Finally Log of when I reconnect. (which works)???
```

As you can see there is a pattern. While the device is bonded, I have to disconnect and reconnect twice before it works. Maybe something is getting queued or started that shouldn't be when I first disconnect? Then when I abruptly disconnect the second time, some exception gets thrown putting the device in good state for reconnect?

##### `Log 1`
```
I/BtR100ConnectHandler: RxBleConnectionState{CONNECTING}
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=7
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=B0:B4:48:DD:8C:EC
D/RxBle#BluetoothGatt: onConnectionStateChange newState=2 status=0
D/RxBle#ClientOperationQueue: FINISHED ConnectOperation(53828653) in 379 ms
D/RxBle#ConnectionOperationQueue: QUEUED   MtuRequestOperation(212342132)
D/RxBle#ConnectionOperationQueue: STARTED  MtuRequestOperation(212342132)
D/BluetoothGatt: configureMTU() - device: B0:B4:48:DD:8C:EC mtu: 245
I/BtR100ConnectHandler: RxBleConnectionState{CONNECTED}
D/ViewRootImpl@ed17da5[BleDeviceActivity]: ThreadedRenderer.create() translucent=false
D/BluetoothGatt: onClientConnParamsChanged() - Device=B0:B4:48:DD:8C:EC interval=6 status=0
D/BluetoothGatt: onConfigureMTU() - Device=B0:B4:48:DD:8C:EC mtu=245 status=0
D/RxBle#BluetoothGatt: onMtuChanged mtu=245 status=0
D/RxBle#ConnectionOperationQueue: QUEUED   ServiceDiscoveryOperation(21897875)
D/RxBle#ConnectionOperationQueue: FINISHED MtuRequestOperation(212342132) in 610 ms
D/RxBle#ConnectionOperationQueue: STARTED  ServiceDiscoveryOperation(21897875)
D/BluetoothGatt: discoverServices() - device: B0:B4:48:DD:8C:EC
D/BluetoothGatt: onSearchComplete() = Device=B0:B4:48:DD:8C:EC Status=0
D/RxBle#BluetoothGatt: onServicesDiscovered status=0
V/RxBle#ServiceDiscoveryOperation: Preparing services description
V/RxBle#ServiceDiscoveryOperation: --------------- ====== Printing peripheral content ====== ---------------
-- SNIP --
--------------- ====== Finished peripheral content ====== ---------------
D/BluetoothGatt: onClientConnParamsChanged() - Device=B0:B4:48:DD:8C:EC interval=39 status=0
D/BluetoothGatt: setCharacteristicNotification() - uuid: 09b0d0a6-f374-4051-a819-67c235f31438 enable: true
D/RxBle#ConnectionOperationQueue: QUEUED   DescriptorWriteOperation(253023875)
D/RxBle#ConnectionOperationQueue: FINISHED ServiceDiscoveryOperation(21897875) in 30 ms
D/RxBle#ConnectionOperationQueue: STARTED  DescriptorWriteOperation(253023875)
D/BluetoothAdapter: cancelDiscovery
D/RxBle#BluetoothGatt: onDescriptorWrite descriptor=00002902-0000-1000-8000-00805f9b34fb status=0
D/RxBle#ConnectionOperationQueue: FINISHED DescriptorWriteOperation(253023875) in 240 ms
```

##### `Log 2`
```
I/RxBle#ConnectionOperationQueue: Connection operations queue to be terminated (B0:B4:48:DD:8C:EC)
D/RxBle#ClientOperationQueue: QUEUED   DisconnectOperation(136959590)
D/RxBle#Executors$RunnableAdapter: Terminated.
D/BluetoothGatt: setCharacteristicNotification() - uuid: 09b0d0a6-f374-4051-a819-67c235f31438 enable: false
D/RxBle#ClientOperationQueue: STARTED  DisconnectOperation(136959590)
D/BluetoothManager: getConnectionState()
D/BluetoothManager: getConnectedDevices
D/RxBle#ConnectionOperationQueue: QUEUED   DescriptorWriteOperation(238790234)
I/BtR100ConnectHandler: RxBleConnectionState{DISCONNECTING}
D/BluetoothGatt: cancelOpen() - device: B0:B4:48:DD:8C:EC
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=B0:B4:48:DD:8C:EC
D/RxBle#BluetoothGatt: onConnectionStateChange newState=0 status=0
D/BluetoothGatt: close()
D/BluetoothGatt: unregisterApp() - mClientIf=7
I/BtR100ConnectHandler: RxBleConnectionState{DISCONNECTED}
D/RxBle#ClientOperationQueue: FINISHED DisconnectOperation(136959590) in 36 ms
```

##### `Log 3` (Exception would be thrown here after about 15 seconds)
```
I/MAC-ADDRESS: B0:B4:48:DD:8C:EC
[... 56 lines cut ...]
I/BtR100ConnectHandler: RxBleConnectionState{CONNECTING}
V/RxBle#BleConnectionCompat: Connecting without reflection
D/BluetoothGatt: connect() - device: B0:B4:48:DD:8C:EC, auto: false
D/BluetoothAdapter: isSecureModeEnabled
D/BluetoothGatt: registerApp()
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=7
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=B0:B4:48:DD:8C:EC
D/RxBle#BluetoothGatt: onConnectionStateChange newState=2 status=0
I/BtR100ConnectHandler: RxBleConnectionState{CONNECTED}
D/RxBle#ClientOperationQueue: FINISHED ConnectOperation(186231241) in 132 ms
D/RxBle#ConnectionOperationQueue: QUEUED   MtuRequestOperation(116007399)
D/RxBle#ConnectionOperationQueue: STARTED  MtuRequestOperation(116007399)
D/BluetoothGatt: configureMTU() - device: B0:B4:48:DD:8C:EC mtu: 215
D/BluetoothGatt: onClientConnParamsChanged() - Device=B0:B4:48:DD:8C:EC interval=6 status=0
D/BluetoothGatt: onConfigureMTU() - Device=B0:B4:48:DD:8C:EC mtu=215 status=0
[... cut ...]
D/RxBle#ConnectionOperationQueue: FINISHED MtuRequestOperation(116007399) in 599 ms
D/RxBle#ConnectionOperationQueue: STARTED  ServiceDiscoveryOperation(138848138)
D/BluetoothGatt: discoverServices() - device: B0:B4:48:DD:8C:EC
D/BluetoothGatt: onSearchComplete() = Device=B0:B4:48:DD:8C:EC Status=0
D/RxBle#BluetoothGatt: onServicesDiscovered status=0
V/RxBle#ServiceDiscoveryOperation: Preparing services description
V/RxBle#ServiceDiscoveryOperation: --------------- ====== Printing peripheral content ====== ---------------
```

dariuszseweryn, 2018-09-27:
Out of curiosity — have you checked what would happen if you would just ignore the timeout error on the MTU change? I mean I know that it would make the connection unusable for 30 seconds but I wonder if it works for other things later.

> Maybe something is getting queued or started that shouldn't be when I first disconnect? Then when I abruptly disconnect the second time, some exception gets thrown putting the device in good state for reconnect?

There is a good chance that something is off inside the Android API that get's fixed. Or your device is getting silently unpaired in the process. You could observe broadcasts with filter `BluetoothDevice.ACTION_BOND_STATE_CHANGED)` to find out.

amharms795, 2018-09-27:
Yes it works if I reconnect after the timeout error happens or if I disconnect before and reconnect.

amharms795, 2018-09-27:
With your suggestion, I am now getting a BleCharacteristicNotFoundException after the MTU_Changed exception gets caught. Which makes sense because the errors will just propagate down since nothing was fixed. Didn't know BLE was this bugged.

amharms795, 2018-09-28:
Here is all the relevant info

```
D/BleManager: Constructing BleManager
I/MAC-ADDRESS: B0:B4:48:DD:8A:60
D/RxBle#ClientOperationQueue: QUEUED   ConnectOperation(212596467)
D/RxBle#ClientOperationQueue: STARTED  ConnectOperation(212596467)
V/RxBle#BleConnectionCompat: Connecting without reflection
D/BluetoothGatt: connect() - device: B0:B4:48:DD:8A:60, auto: false
D/BluetoothAdapter: isSecureModeEnabled
D/BluetoothGatt: registerApp()
    registerApp() - UUID=e19b44b6-daf6-483d-a733-fe56f3f9ba68
I/BtR100ConnectHandler: RxBleConnectionState{CONNECTING}
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=7
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=B0:B4:48:DD:8A:60
D/RxBle#BluetoothGatt: onConnectionStateChange newState=2 status=0
D/RxBle#ClientOperationQueue: FINISHED ConnectOperation(212596467) in 431 ms
D/RxBle#ConnectionOperationQueue: QUEUED   MtuRequestOperation(219982610)
D/RxBle#ConnectionOperationQueue: STARTED  MtuRequestOperation(219982610)
I/BtR100ConnectHandler: RxBleConnectionState{CONNECTED}
D/BluetoothGatt: configureMTU() - device: B0:B4:48:DD:8A:60 mtu: 202
D/BluetoothGatt: onClientConnParamsChanged() - Device=B0:B4:48:DD:8A:60 interval=6 status=0
D/BluetoothGatt: onClientConnParamsChanged() - Device=B0:B4:48:DD:8A:60 interval=39 status=0
D/BluetoothAdapter: cancelDiscovery
D/RxBle#ConnectionOperationQueue: FINISHED MtuRequestOperation(219982610) in 30007 ms
D/Error happened: whoops!
    com.polidea.rxandroidble.exceptions.BleGattCallbackTimeoutException: GATT exception from MAC address B0:B4:48:DD:8A:60, with type BleGattOperation{description='ON_MTU_CHANGED'}
        at com.polidea.rxandroidble.internal.SingleResponseOperation.timeoutFallbackProcedure(SingleResponseOperation.java:87)
        at com.polidea.rxandroidble.internal.SingleResponseOperation.protectedRun(SingleResponseOperation.java:52)
        at com.polidea.rxandroidble.internal.QueueOperation$1.call(QueueOperation.java:42)
        at com.polidea.rxandroidble.internal.QueueOperation$1.call(QueueOperation.java:38)
        at rx.internal.operators.OnSubscribeCreate.call(OnSubscribeCreate.java:72)
        at rx.internal.operators.OnSubscribeCreate.call(OnSubscribeCreate.java:32)
        at rx.Observable.unsafeSubscribe(Observable.java:10256)
        at rx.internal.operators.OperatorSubscribeOn$SubscribeOnSubscriber.call(OperatorSubscribeOn.java:100)
        at rx.internal.schedulers.ScheduledAction.run(ScheduledAction.java:55)
        at rx.internal.schedulers.ExecutorScheduler$ExecutorSchedulerWorker.run(ExecutorScheduler.java:107)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
        at java.lang.Thread.run(Thread.java:762)

D/RxBle#ConnectionOperationQueue: QUEUED   ServiceDiscoveryOperation(226472656)
D/RxBle#ConnectionOperationQueue: STARTED  ServiceDiscoveryOperation(226472656)
D/BluetoothGatt: discoverServices() - device: B0:B4:48:DD:8A:60
D/BluetoothGatt: onSearchComplete() = Device=B0:B4:48:DD:8A:60 Status=0
D/RxBle#BluetoothGatt: onServicesDiscovered status=0
V/RxBle#ServiceDiscoveryOperation: Preparing services description
V/RxBle#ServiceDiscoveryOperation: --------------- ====== Printing peripheral content ====== ---------------
    PERIPHERAL ADDRESS: B0:B4:48:DD:8A:60
    PERIPHERAL NAME: R100 L8A60
    -------------------------------------------------------------------------
    
    Primary Service - Generic Access (00001800-0000-1000-8000-00805f9b34fb)
    Instance ID: 1
    -> Characteristics:
    --------------- ====== Finished peripheral content ====== ---------------
E/BleManager: Connection Error
    com.polidea.rxandroidble.exceptions.BleCharacteristicNotFoundException: Characteristic not found with UUID 09b0d0a6-f374-4051-a819-67c235f31438
        at com.polidea.rxandroidble.RxBleDeviceServices$2.call(RxBleDeviceServices.java:76)
        at com.polidea.rxandroidble.RxBleDeviceServices$2.call(RxBleDeviceServices.java:67)
        at rx.internal.operators.OnSubscribeFromCallable.call(OnSubscribeFromCallable.java:48)
        at rx.internal.operators.OnSubscribeFromCallable.call(OnSubscribeFromCallable.java:33)
        at rx.Observable.unsafeSubscribe(Observable.java:10256)
        at rx.internal.operators.OperatorMerge$MergeSubscriber.onNext(OperatorMerge.java:248)
        at rx.internal.operators.OperatorMerge$MergeSubscriber.onNext(OperatorMerge.java:148)
        at rx.internal.operators.OnSubscribeMap$MapSubscriber.onNext(OnSubscribeMap.java:77)
        at rx.observers.Subscribers$5.onNext(Subscribers.java:235)
        at rx.internal.operators.NotificationLite.accept(NotificationLite.java:135)
        at rx.internal.operators.CachedObservable$ReplayProducer.replay(CachedObservable.java:403)
        at rx.internal.operators.CachedObservable$CacheState.dispatch(CachedObservable.java:220)
[... 24 lines cut ...]
        at rx.internal.operators.OperatorTake$1.onNext(OperatorTake.java:80)
        at rx.internal.operators.OnSubscribeDoOnEach$DoOnEachSubscriber.onNext(OnSubscribeDoOnEach.java:101)
        at rx.internal.operators.OperatorObserveOn$ObserveOnSubscriber.call(OperatorObserveOn.java:224)
        at rx.internal.schedulers.ScheduledAction.run(ScheduledAction.java:55)
        at rx.internal.schedulers.ExecutorScheduler$ExecutorSchedulerWorker.run(ExecutorScheduler.java:107)
[... cut ...]
    getConnectedDevices
I/BtR100ConnectHandler: RxBleConnectionState{DISCONNECTING}
D/BluetoothGatt: cancelOpen() - device: B0:B4:48:DD:8A:60
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=B0:B4:48:DD:8A:60
D/RxBle#BluetoothGatt: onConnectionStateChange newState=0 status=0
D/BluetoothGatt: close()
D/BluetoothGatt: unregisterApp() - mClientIf=7
I/BtR100ConnectHandler: RxBleConnectionState{DISCONNECTED}
D/RxBle#ClientOperationQueue: FINISHED DisconnectOperation(256215346) in 36 ms
```

dariuszseweryn, 2018-10-12:
Unfortunately there is nothing known that the library can do about this issue. If anyone has an idea — feel free to post/reopen.

closed 2018-10-12 by dariuszseweryn

gumbelmj, 2022-06-17:
I ran into this same issue on a Pixel 5, Android 12.

As a work around the `MTURequestOperation` can be adjusted as:
```
    protected boolean startOperation(BluetoothGatt bluetoothGatt) {
        bluetoothGatt.requestMtu(mtu);
        return true;
    }
```

`bluetoothGatt.requestMtu()` always returns false, but the MTU is set anyway.