dariuszseweryn/RxAndroidBle #492: stuck on connecting on Huawei device trying to connect to BLE device

amirziarati, 2018-10-02:
im trying to connect to different BLE devices from a `Huawei P9 lite` but it gets stuck in connecting state in most of the times. if i clear `bluetooth share` system application's cache, connection will be successful for the first time sometimes but it will again get stuck to connecting state after trying to reconnect.

my code snippet is:

                      rxBleClient

                        .scanBleDevices(getSettings(), getFilter())

                        .timeout(SCAN_TIMEOUT, TimeUnit.MILLISECONDS, Observable.empty())

                        .observeOn(AndroidSchedulers.mainThread())

                        .subscribeBy(
                                onNext = { scanResult ->
                                    device = scanResult.bleDevice
                                    Timber.i("Found $device")
                                },
                                onError = { throwable ->
                                    Timber.e(throwable)
                                },
                                onComplete = {
                                    device?.let {
                                        startConnecting(it)
                                    }
                                            ?: Toast.makeText(context, "No device found!", Toast.LENGTH_LONG).show()
                                })

the log is:

I/RxBle#QueueOperation: Scan operation is requested to start.
I/MainActivity$startScan(MainActivity.kt:152): Found RxBleDeviceImpl{bluetoothDevice=Things(76:50:5F:AF:34:4D)
I/RxBle#CancellableDisposable: Scan operation is requested to stop.
I/MainActivity(MainActivity.kt:183): connecting to RxBleDeviceImpl{bluetoothDevice=Things(76:50:5F:AF:34:4D)}

dariuszseweryn, 2018-10-02:
Hello,

Thank you for using the library.
Your post seems to be a bug report but it lacks some crucial information. Could you edit to contain information according to the [issue template](https://github.com/Polidea/RxAndroidBle/blob/master/.github/ISSUE_TEMPLATE.md)?

Best Regards

amirziarati, 2018-10-02:
tried to edit the post to be according the template.

dariuszseweryn, 2018-10-02:
Please attach logs when `RxBleLog.setLogLevel(RxBleLog.VERBOSE)` is set so the library logs will be present

amirziarati, 2018-10-02:
its already set to verbose and the whole log from the library side is added to post. since the library doesnt log when it send me a connecting state i put a log by myself printing when connecting state is emitted.

ill also add client and GAT source repo to my post.

dariuszseweryn, 2018-10-02:
1. The log you have posted is not full
2. Have you read about `autoConnect` flag? If not — please do

dariuszseweryn, 2018-10-02:
Try using `autoConnect=false`

amirziarati, 2018-10-02:
I read about auto connect. obviously i dont want a exception after 30 seconds if the device was not found. i prefer it keeps searching and inform me whenever it finds the device. am i missing something ?

dariuszseweryn, 2018-10-02:
```
Be careful not to overuse the autoConnect flag. On the other side it has negative impact on the connection initialization speed. Scanning window and interval is lowered as it is optimized for background use and depending on Bluetooth parameters it may (and usually do) take more time to establish the connection.
```

amirziarati, 2018-10-02:
the code i sent you is just an example. i am sing this code in a background service and i need it to wait for te device to advertise. so i need to call establish with autoConnect=true. what should i do to make it work ?

dariuszseweryn, 2018-10-02:
If you want to do it with `autoConnect=true` then you need wait. Or you can create a logic that would retry with `autoConnect=false` but that could potentially eat up battery pretty quickly. Or you can scan on some low priority and start connecting only when scanned. Either way this does not seem to be a problem with the library

amirziarati, 2018-10-04:
anyway changing it to false works on android things but i have a problem with another hardware. it keeps giving me 133 GatError while trying to reconnect to a previously bonded device. i searched a lot about this problem and i found [this](https://github.com/NordicSemiconductor/Android-DFU-Library/issues/1#issuecomment-361663614): 

cGatt.connectGatt(this, false, gattCallback, 2);

tat if i add a 2 as last parameter of connectGatt this will be fixed while i dont use android api directly and im using your library. so is there a way to do that by your library ?

amirziarati, 2018-10-04:
anyway changing it to false works on android things but i have a problem with another hardware. it keeps giving me 133 GatError while trying to reconnect to a previously bonded device. it only happens on android versions lower than 8.

amirziarati, 2018-10-04:
i just read the library code and i saw you are doing the same for android versions higher than M. but anyway this 133 error is happening to me. i dont know what to do.

amirziarati, 2018-10-04:
this is the error:

``io.reactivex.exceptions.UndeliverableException: com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from 88:D2:BF:01:05:B4 with status 133 (GATT_ERROR)
        at io.reactivex.plugins.RxJavaPlugins.onError(RxJavaPlugins.java:367)
        at io.reactivex.internal.operators.observable.ObservableUnsubscribeOn$UnsubscribeObserver.onError(ObservableUnsubscribeOn.java:67)
        at io.reactivex.internal.operators.observable.ObservableSubscribeOn$SubscribeOnObserver.onError(ObservableSubscribeOn.java:63)
        at io.reactivex.internal.operators.observable.ObservableCreate$CreateEmitter.tryOnError(ObservableCreate.java:85)
        at com.polidea.rxandroidble2.internal.util.DisposableUtil$3.onError(DisposableUtil.java:60)
        at io.reactivex.internal.operators.single.SingleDoFinally$DoFinallyObserver.onError(SingleDoFinally.java:79)
        at io.reactivex.internal.operators.single.SingleTimeout$TimeoutMainObserver.onError(SingleTimeout.java:142)
        at io.reactivex.internal.operators.single.SingleCreate$Emitter.tryOnError(SingleCreate.java:95)
        at com.polidea.rxandroidble2.internal.util.DisposableUtil$1.onError(DisposableUtil.java:24)
        at io.reactivex.internal.operators.flowable.FlowableElementAtSingle$ElementAtSubscriber.onError(FlowableElementAtSingle.java:101)
        at io.reactivex.internal.operators.flowable.FlowableFlatMap$MergeSubscriber.checkTerminate(FlowableFlatMap.java:566)
        at io.reactivex.internal.operators.flowable.FlowableFlatMap$MergeSubscriber.drainLoop(FlowableFlatMap.java:374)
        at io.reactivex.internal.operators.flowable.FlowableFlatMap$MergeSubscriber.drain(FlowableFlatMap.java:366)
        at io.reactivex.internal.operators.flowable.FlowableFlatMap$MergeSubscriber.innerError(FlowableFlatMap.java:605)
        at io.reactivex.internal.operators.flowable.FlowableFlatMap$InnerSubscriber.onError(FlowableFlatMap.java:668)
        at io.reactivex.internal.operators.single.SingleToFlowable$SingleToFlowableObserver.onError(SingleToFlowable.java:68)
        at io.reactivex.internal.operators.observable.ObservableElementAtSingle$ElementAtObserver.onError(ObservableElementAtSingle.java:104)
        at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.checkTerminate(ObservableFlatMap.java:506)
        at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.drainLoop(ObservableFlatMap.java:335)
        at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.drain(ObservableFlatMap.java:327)
        at io.reactivex.internal.operators.observable.ObservableFlatMap$InnerObserver.onError(ObservableFlatMap.java:582)
        at io.reactivex.internal.disposables.EmptyDisposable.error(EmptyDisposable.java:63)
        at io.reactivex.internal.operators.observable.ObservableError.subscribeActual(ObservableError.java:37)
        at io.reactivex.Observable.subscribe(Observable.java:12030)
        at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.subscribeInner(ObservableFlatMap.java:165)
        at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.onNext(ObservableFlatMap.java:139)
        at io.reactivex.internal.operators.observable.ObservableCreate$CreateEmitter.onNext(ObservableCreate.java:67)
        at com.polidea.rxandroidble2.internal.connection.DisconnectionRouter.notifySubscribersAboutException(DisconnectionRouter.java:124)
        at com.polidea.rxandroidble2.internal.connection.DisconnectionRouter.onExceptionOccurred(DisconnectionRouter.java:113)
        at com.polidea.rxandroidble2.internal.connection.DisconnectionRouter.onDisconnectedException(DisconnectionRouter.java:99)
        at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:77)
        at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:191)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:70)
        at android.os.Binder.execTransact(Binder.java:565)
    	Suppressed: java.lang.Throwable: HERE SDK Version: 3.7.0.118
        at com.nokia.maps.MapsEngine$l.uncaughtException(MapsEngine.java:378)
        at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1068)
        at java.lang.ThreadGroup.uncaughtException(ThreadGroup.java:1063)
        at io.reactivex.plugins.RxJavaPlugins.uncaught(RxJavaPlugins.java:429)
        at io.reactivex.plugins.RxJavaPlugins.onError(RxJavaPlugins.java:383)``

it happens only on Huawei p9 android 7.

dariuszseweryn, 2018-10-05:
The `status=133` is usually emitted when Android BLE stack does not handle something well. Unfortunately this is usually the case with bonded devices. [Other libraries even encourage to clear bonding after each connection](https://github.com/iDevicesInc/SweetBlue/wiki/Android-BLE-Issues#bondingpairing-issues). There are [quite a lot of BLE issues on the Google bugtracker](https://issuetracker.google.com/issues?q=android%20bond%20ble) about connecting to bonded devices.

My own experience with bonded devices is that it is not stable across devices to use unless one wants to maintain a whitelist of tested devices/OS versions pairs.

I am afraid that the library cannot do anything on it's own but if you think otherwise please add more information.

closed 2018-10-12 by dariuszseweryn

g-balas, 2020-12-17:
Hey @amirziarati, did you find a solution since then ?
Having some BLE problems with Huawei smartphones...