dariuszseweryn/RxAndroidBle #493: BleGattCallbackTimeoutException: GATT exception from MAC address ***, with type BleGattOperation{description='CONNECTION_STATE'}

saramotiv, 2018-10-03:
### Summary
Receiving BleGattCallbackTimeoutException when calling  `RxBleDevice.establishConnection(boolean autoConnect)`

### Library version
`1.5.0`

#### Steps to reproduce actual result
<br />1. call RxBleDevice.establishConnection(boolean autoConnect) 

#### Logs from the application running with setting `RxBleLog.setLogLevel(RxBleLog.VERBOSE)`
D/RxBle#ClientOperationQueue: QUEUED   ConnectOperation(15489449)
D/RxBle#ClientOperationQueue: STARTED  ConnectOperation(15489449)
V/RxBle#BleConnectionCompat: Connecting without reflection
D/TimberLogger: device.paired - Q00941676B42
D/BluetoothGatt: connect() - device: CA:16:7E:C6:6B:42, auto: false
D/BluetoothGatt: registerApp() - UUID=7849ef51-9eb4-47c3-a509-26c0fab373af
 D/BluetoothGatt: onClientRegistered() - status=0 clientIf=7
 I/DefaultDeliveryClient: Event: 'devic...' recorded to local filestore
 D/DefaultDeliveryClient: Time of enqueueEventForDelivery: 2
D/RxBle#ClientOperationQueue: FINISHED ConnectOperation(15489449) in 35009 ms
E/RxBlePeripheral: com.polidea.rxandroidble2.exceptions.BleGattCallbackTimeoutException: GATT exception from MAC address CA:16:7E:C6:6B:42, with type BleGattOperation{description='CONNECTION_STATE'}
        at com.polidea.rxandroidble2.internal.operations.ConnectOperation$3.call(ConnectOperation.java:106)
        at com.polidea.rxandroidble2.internal.operations.ConnectOperation$3.call(ConnectOperation.java:103)
        at io.reactivex.internal.operators.single.SingleFromCallable.subscribeActual(SingleFromCallable.java:44)
        at io.reactivex.Single.subscribe(Single.java:3224)
        at io.reactivex.internal.operators.single.SingleTimeout$TimeoutMainObserver.run(SingleTimeout.java:118)
        at io.reactivex.internal.schedulers.ScheduledDirectTask.call(ScheduledDirectTask.java:38)
        at io.reactivex.internal.schedulers.ScheduledDirectTask.call(ScheduledDirectTask.java:26)
        at java.util.concurrent.FutureTask.run(FutureTask.java:237)
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:272)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
        at java.lang.Thread.run(Thread.java:761)

D/RxBle#ClientOperationQueue: QUEUED   DisconnectOperation(123334216)
D/RxBle#ClientOperationQueue: STARTED  DisconnectOperation(123334216)
D/BluetoothManager: getConnectionState()
I/RxBle#ConnectionOperationQueue: Connection operations queue to be terminated (CA:16:7E:C6:6B:42)
D/BluetoothManager: getConnectedDevices
I/DefaultDeliveryClient: Event: 'perip...' recorded to local filestore
D/DefaultDeliveryClient: Time of enqueueEventForDelivery: 7
D/RxBle#Executors$RunnableAdapter: Terminated.
D/BluetoothGatt: close()
    unregisterApp() - mClientIf=7
D/RxBle#ClientOperationQueue: FINISHED DisconnectOperation(123334216) in 20 ms

#### Actual result
Receiving com.polidea.rxandroidble2.exceptions.BleGattCallbackTimeoutException: GATT exception from MAC address CA:16:7E:C6:6B:42, with type BleGattOperation{description='CONNECTION_STATE'}

#### Expected result
Device connects

dariuszseweryn, 2018-10-03:
Hello,

What's the phone model and OS version you are testing? Have you read the [Wiki](https://github.com/Polidea/RxAndroidBle/wiki/FAQ:-Cannot-connect)? Have you tried with the newest version of the library?

saramotiv, 2018-10-03:
Erasing app data and resetting the phone actually helped. I had the problem in 2 different devices: Google Pixel Android 7.1.2 API 25 and Samsung Galaxy S9 SM-G960U. I will update the library.

closed 2018-10-03 by saramotiv

purrsong-km, 2020-03-07:
Is anyone experiences same issue till now?

tworkool, 2021-01-04:
Hey, I also experience the same issue on two devices (both IPhones)