dariuszseweryn/RxAndroidBle #495: How to setWriteType for a characteristic

mohsenoid, 2018-10-09:
Hi, after greeting and appreciation for your nice job,

My question is, our BLE peripheral writable characteristic supports `PROPERTY_WRITE` plus `PROPERTY_WRITE_NO_RESPONSE`, and on App side when I call `bleConnection.writeCharacteristic` it only works with no response cause I get `onSuccesss` so fast even when something is wrong on the device side.

Based on Android BLE API doc it is possible to set how to write:
https://developer.android.com/reference/android/bluetooth/BluetoothGattCharacteristic.html#setWriteType(int)

and I noticed if I want to set it using the library I have to use `getCharacteristic(uuid)` method which has been already deprecated from your side!

is there any solution to set write with a response when Characteristic support both?

dariuszseweryn, 2018-10-09:
Hello

You may use `RxBleConnection.discoverServices()` and then search the `RxBleDeviceServices` for the `BluetoothGattCharacteristic` you are interested in. Then you can set the `BluetoothGattCharacteristic.setWriteType(int)` method and start to use `RxBleConnection.writeCharacteristic(BluetoothGattCharacteristic, byte[])`.

Best Regards

dariuszseweryn, 2018-10-15:
Since the OP has reacted to the answer I assume thing is sorted out. Closing

If mistaken feel free to add more info or reopen.

closed 2018-10-15 by dariuszseweryn

hjh2881, 2018-12-10:
can you share the basic code ?i don't know how to send data with write type no_response