dariuszseweryn/RxAndroidBle #496: Android Pie - Google Pixel XL - not connectable with autoConnect

skywall, 2018-10-09:
### Summary
Hi guys, 

I'm facing quite tricky problem with combination Android Pie (9.0) & Google Pixel XL & `autoConnect == true`. The problem is I'm not able to connect at all. The same code works flawlessly on OnePlus 6 (Android Pie too). I'm not sure, if it happens with every BLE peripheral, but in our case, I have selected random characteristic and I tried read from it.

### Library version
1.7.1

#### Preconditions

No specific preconditions.

#### Steps to reproduce actual result
<br />1. Clone code from https://github.com/skywall/ble_pie_issue
<br />2. Change `DEVICE_MAC` & `TEST_CHARACTERISTIC` to match your peripheral
<br />2. Run app on Google Pixel XL with Android Pie
<br />2. Click button Read Vin
#### Minimum code snippet reproducing the issue
```kotlin
companion object {
    const val DEVICE_MAC = "C9:A4:18:52:C1:11"
    val TEST_CHARACTERISTIC = UUID.fromString("6B880002-7A7E-678A-CE40-8A0AB472E2B0")!!
}

val bleClient = RxBleClient.create(this)
val bleDevice = bleClient.getBleDevice(DEVICE_MAC)
val connection = bleDevice.establishConnection(true).compose(ConnectionSharingAdapter())

connection
        .subscribeOn(Schedulers.io())
        .timeout(5, TimeUnit.SECONDS)
        .flatMapSingle { rxBleConnection ->
            rxBleConnection.readCharacteristic(TEST_CHARACTERISTIC)
                    .timeout(5, TimeUnit.SECONDS)
        }
        .observeOn(AndroidSchedulers.mainThread())
        .subscribe({ bytes ->
            Toast.makeText(this, "Read Success: $bytes", Toast.LENGTH_SHORT).show()
        }, { throwable ->
            Toast.makeText(this, "Read Failed: $throwable", Toast.LENGTH_SHORT).show()
        })
```
#### Logs from the application running with setting `RxBleLog.setLogLevel(RxBleLog.VERBOSE)`

```
D/RxBle#ClientOperationQueue: QUEUED   ConnectOperation(114728808)
D/RxBle#ClientOperationQueue: STARTED  ConnectOperation(114728808)
V/RxBle#BleConnectionCompat: Connecting without reflection
D/BluetoothGatt: connect() - device: C9:A4:18:52:C1:11, auto: true
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=6d254619-7f5e-437e-b53b-41ed10094d88
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=8
// At this moment, `TimeoutException` finishes the connection chain
D/RxBle#ClientOperationQueue: FINISHED ConnectOperation(114728808) in 54 ms
I/RxBle#ConnectionOperationQueue: Connection operations queue to be terminated (C9:A4:18:52:C1:11)
D/RxBle#Executors$RunnableAdapter: Terminated.
D/RxBle#ClientOperationQueue: QUEUED   DisconnectOperation(67883709)
D/RxBle#ClientOperationQueue: STARTED  DisconnectOperation(67883709)
D/BluetoothManager: getConnectionState()
D/BluetoothGatt: close()
D/BluetoothGatt: unregisterApp() - mClientIf=8
D/RxBle#ClientOperationQueue: FINISHED DisconnectOperation(67883709) in 40 ms
```

#### Actual result
Device wait infinitely for connection, but never connects. There are some `timeout` operators in code snippet. Five seconds limit to connect to the peripheral is imho big enough, but feel free to comment it out.

#### Expected result
Device connects to BLE peripheral.

#### Before you ask

- With `autoConnect == false` this code snippet works quite good. I haven't experienced any problem, but I really need this feature, because I get `BleDisconnectedException` every couple of seconds otherwise.
- In case I connect to device with app [nRF Mobile for Mobile](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp) (option `autoConnect` checked to true), snippet above starts to work. I expect there is some internal Android BLE Stack cache.

Thanks for every help.

dariuszseweryn, 2018-10-09:
1. 5 seconds is usually not enough -> check description in readme
2. There is a bug with mac address resolution which you may accidentally mitigate by using `nRF Connect` -> check Wiki `Cannot Connect`

dariuszseweryn, 2018-10-15:
Since the OP has reacted to the answer and no more info is available I am closing this issue. Feel free to reopen or add more info if you think otherwise.

closed 2018-10-15 by dariuszseweryn

akshayoyo16, 2019-04-26:
can you elaborate on that @dariuszseweryn . I am trying :
device.connectGatt(getApplicationContext(), true, mGattCallback, BluetoothDevice.TRANSPORT_LE);

It is working in Android version 8 and below. Not working in android 9. Please Help.

dariuszseweryn, 2019-04-26:
[Help yourself — second point. Check Wiki](https://github.com/Polidea/RxAndroidBle/issues/496#issuecomment-428322494)

akshayoyo16, 2019-04-26:
Can you share the link please, unable to find