dariuszseweryn/RxAndroidBle #499: hello,what means ble disconnected status = -1

Lloyd0577, 2018-10-19:
BleDisconnectedException: Disconnected from D6:57:6B:F0:72:C1 with status -1 (UNKNOWN)

io.reactivex.exceptions.UndeliverableException:com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from D6:57:6B:F0:72:C1 with status -1 (UNKNOWN)
--
2 io.reactivex.plugins.RxJavaPlugins.onError(RxJavaPlugins.java:367)
3 ......
4 Caused by:
5 com.polidea.rxandroidble2.exceptions.BleAdapterDisabledException:
6 com.polidea.rxandroidble2.internal.connection.DisconnectionRouter$3.apply(DisconnectionRouter.java:57)
7 com.polidea.rxandroidble2.internal.connection.DisconnectionRouter$3.apply(DisconnectionRouter.java:54)
8 io.reactivex.internal.operators.observable.ObservableMap$MapObserver.onNext(ObservableMap.java:59)
9 io.reactivex.internal.operators.observable.ObservableFilter$FilterObserver.onNext(ObservableFilter.java:52)
10 io.reactivex.internal.operators.observable.ObservableConcatMap$ConcatMapDelayErrorObserver$DelayErrorInnerObserver.onNext(ObservableConcatMap.java:500)
11 io.reactivex.internal.operators.observable.ObservableMap$MapObserver.onNext(ObservableMap.java:64)
12 com.polidea.rxandroidble2.RxBleAdapterStateObservable$1.onReceive(RxBleAdapterStateObservable.java:62)
13 android.app.LoadedApk$ReceiverDispatcher$Args.lambda$-android_app_LoadedApk$ReceiverDispatcher$Args_56045(LoadedApk.java:1403)
14 android.app.-$Lambda$aS31cHIhRx41653CMnd4gZqshIQ.$m$7(Unknown Source:4)
15 android.app.-$Lambda$aS31cHIhRx41653CMnd4gZqshIQ.run(Unknown Source:39)
16 android.os.Handler.handleCallback(Handler.java:790)
17 android.os.Handler.dispatchMessage(Handler.java:99)
18 android.os.Looper.loop(Looper.java:200)
19 android.app.ActivityThread.main(ActivityThread.java:7038)
20 java.lang.reflect.Method.invoke(Native Method)
21 com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:519)
22 com.android.internal.os.ZygoteInit.main(ZygoteInit.java:837)

Lloyd0577, 2018-10-21:
but why the exception throw out ,and how can I try catch it  ,my code is:
        rxBleDevice = rxBleClient.getBleDevice(mac)

        subscription = rxBleDevice!!.establishConnection(false)
                .subscribe({ rxBleConnection ->
                    this.rxBleConnection = rxBleConnection
                    Logger.d("conn succ")

                    shakeHand(userId, mac, success, failure)
                }, { throwable ->
                    disConnectListeners.forEach {
                        it.invoke("conn error")
                    }
                    failure.invoke("conn error")
                })
@dariuszseweryn

Lloyd0577, 2018-10-25:
hello , thank you for help, but I wonder that is there any reasons that there was no callback to BluetoothGattCallback.onConnectionStateChange() when the disconnection happened?

dariuszseweryn, 2018-10-25:
Reason: the BT Adapter was turned off — this is the way Android BLE API works

closed 2018-11-05 by dariuszseweryn