dariuszseweryn/RxAndroidBle #500: check out from version control doesn't have GitHub

Rogue899, 2018-10-24:
[image]
[image]

Before I used to have GitHub and Git but now after deleting android studio and reinstalling it, it seems that I can't find GitHub

dariuszseweryn, 2018-10-24:
Hello,

I do not see how this is a library issue

Rogue899, 2018-10-24:
> Hello,
> 
> I do not see how this is a library issue

i do not know what is the issue but after googling the problem many people suggested that maybe it is a plugin issue so i screenshot it for clarification, the question is why isn't github showing in the VCS...

dariuszseweryn, 2018-10-24:
I think that you should ask this question to the authors of `Android Studio` this is a project site of Android Bluetooth Low Energy library and we are not developing `Android Studio`.

There was however [an issue with checking out the library and building it later](https://github.com/Polidea/RxAndroidBle/issues/500) but it is also an issue in `Android Studio` itself.

Anyway I cannot help you with the lack of `Github` option when trying to checkout code in `Android Studio`.

You probably can simply use `Git` option as `Github` is using `Git`

Best Regards

closed 2018-10-24 by dariuszseweryn