dariuszseweryn/RxAndroidBle #507: io.reactivex.exceptions.UndeliverableException when Bluetooth Connect fail

rouddy, 2018-11-26:
### Summary

io.reactivex.exceptions.UndeliverableException is thrown sometimes when bluetooth connection failed.

here's the log of sample application.

11-26 17:25:11.050 4798-4848/com.polidea.rxandroidble.sample D/RxBle#ClientOperationQueue: QUEUED   ConnectOperation(181760501)
11-26 17:25:11.050 4798-4816/com.polidea.rxandroidble.sample D/RxBle#ClientOperationQueue: STARTED  ConnectOperation(181760501)
11-26 17:25:11.050 4798-4827/com.polidea.rxandroidble.sample V/RxBle#BleConnectionCompat: Connecting without reflection
11-26 17:25:11.050 4798-4827/com.polidea.rxandroidble.sample D/BluetoothGatt: connect() - device: DD:E9:B8:3B:58:57, auto: false
11-26 17:25:11.050 4798-4827/com.polidea.rxandroidble.sample D/BluetoothGatt: registerApp()
11-26 17:25:11.050 4798-4827/com.polidea.rxandroidble.sample D/BluetoothGatt: registerApp() - UUID=94699d9e-d5f2-4e13-a689-9dd1611ec14c
11-26 17:25:11.090 4798-4812/com.polidea.rxandroidble.sample D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
11-26 17:25:16.110 4798-4811/com.polidea.rxandroidble.sample D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=6 device=DD:E9:B8:3B:58:57
11-26 17:25:16.110 4798-4811/com.polidea.rxandroidble.sample D/RxBle#BluetoothGatt: onConnectionStateChange newState=0 status=133
11-26 17:25:16.110 4798-4811/com.polidea.rxandroidble.sample I/RxBle#ConnectionOperationQueue: Connection operations queue to be terminated (DD:E9:B8:3B:58:57)
11-26 17:25:16.110 4798-5517/com.polidea.rxandroidble.sample D/RxBle#Executors$RunnableAdapter: Terminated.
11-26 17:25:16.120 4798-4816/com.polidea.rxandroidble.sample D/RxBle#ClientOperationQueue: FINISHED ConnectOperation(181760501) in 5068 ms
11-26 17:25:16.120 4798-4848/com.polidea.rxandroidble.sample D/RxBle#ClientOperationQueue: QUEUED   DisconnectOperation(50279931)
11-26 17:25:16.120 4798-4816/com.polidea.rxandroidble.sample D/RxBle#ClientOperationQueue: STARTED  DisconnectOperation(50279931)
11-26 17:25:16.120 4798-4827/com.polidea.rxandroidble.sample D/BluetoothManager: getConnectionState()
11-26 17:25:16.120 4798-4827/com.polidea.rxandroidble.sample D/BluetoothManager: getConnectedDevices
11-26 17:25:16.120 4798-4811/com.polidea.rxandroidble.sample W/System.err: io.reactivex.exceptions.UndeliverableException: The exception could not be delivered to the consumer because it has already canceled/disposed the flow or the exception has nowhere to go to begin with. Further reading: https://github.com/ReactiveX/RxJava/wiki/What's-different-in-2.0#error-handling | Disconnected from DD:E9:B8:3B:58:57 with status 133 (GATT_ERROR)
11-26 17:25:16.120 4798-4798/com.polidea.rxandroidble.sample D/TextView: setTypeface with style : 0
11-26 17:25:16.120 4798-4827/com.polidea.rxandroidble.sample D/BluetoothGatt: close()
11-26 17:25:16.130 4798-4827/com.polidea.rxandroidble.sample D/BluetoothGatt: unregisterApp() - mClientIf=6
11-26 17:25:16.130 4798-4816/com.polidea.rxandroidble.sample D/RxBle#ClientOperationQueue: FINISHED DisconnectOperation(50279931) in 11 ms
11-26 17:25:16.130 4798-4811/com.polidea.rxandroidble.sample W/System.err:     at io.reactivex.plugins.RxJavaPlugins.onError(RxJavaPlugins.java:367)
11-26 17:25:16.130 4798-4811/com.polidea.rxandroidble.sample W/System.err:     at io.reactivex.internal.operators.flowable.FlowableFlatMap$MergeSubscriber.innerError(FlowableFlatMap.java:608)
11-26 17:25:16.130 4798-4811/com.polidea.rxandroidble.sample W/System.err:     at io.reactivex.internal.operators.flowable.FlowableFlatMap$InnerSubscriber.onError(FlowableFlatMap.java:672)
11-26 17:25:16.130 4798-4811/com.polidea.rxandroidble.sample W/System.err:     at io.reactivex.internal.operators.single.SingleToFlowable$SingleToFlowableObserver.onError(SingleToFlowable.java:67)
11-26 17:25:16.130 4798-4811/com.polidea.rxandroidble.sample W/System.err:     at io.reactivex.internal.operators.observable.ObservableElementAtSingle$ElementAtObserver.onError(ObservableElementAtSingle.java:102)
11-26 17:25:16.130 4798-4811/com.polidea.rxandroidble.sample W/System.err:     at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.checkTerminate(ObservableFlatMap.java:506)
11-26 17:25:16.130 4798-4811/com.polidea.rxandroidble.sample W/System.err:     at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.drainLoop(ObservableFlatMap.java:334)
11-26 17:25:16.130 4798-4811/com.polidea.rxandroidble.sample W/System.err:     at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.drain(ObservableFlatMap.java:326)
11-26 17:25:16.130 4798-4811/com.polidea.rxandroidble.sample W/System.err:     at io.reactivex.internal.operators.observable.ObservableFlatMap$InnerObserver.onError(ObservableFlatMap.java:585)
11-26 17:25:16.130 4798-4811/com.polidea.rxandroidble.sample W/System.err:     at io.reactivex.internal.disposables.EmptyDisposable.error(EmptyDisposable.java:63)
11-26 17:25:16.130 4798-4811/com.polidea.rxandroidble.sample W/System.err:     at io.reactivex.internal.operators.observable.ObservableError.subscribeActual(ObservableError.java:38)
11-26 17:25:16.130 4798-4811/com.polidea.rxandroidble.sample W/System.err:     at io.reactivex.Observable.subscribe(Observable.java:12090)
11-26 17:25:16.130 4798-4811/com.polidea.rxandroidble.sample W/System.err:     at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.subscribeInner(ObservableFlatMap.java:165)
11-26 17:25:16.130 4798-4811/com.polidea.rxandroidble.sample W/System.err:     at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.onNext(ObservableFlatMap.java:139)
11-26 17:25:16.130 4798-4811/com.polidea.rxandroidble.sample W/System.err:     at io.reactivex.internal.operators.observable.ObservableCreate$CreateEmitter.onNext(ObservableCreate.java:66)
11-26 17:25:16.130 4798-4811/com.polidea.rxandroidble.sample W/System.err:     at com.polidea.rxandroidble2.internal.connection.DisconnectionRouter.notifySubscribersAboutException(DisconnectionRouter.java:124)
11-26 17:25:16.130 4798-4811/com.polidea.rxandroidble.sample W/System.err:     at com.polidea.rxandroidble2.internal.connection.DisconnectionRouter.onExceptionOccurred(DisconnectionRouter.java:113)
11-26 17:25:16.140 4798-4811/com.polidea.rxandroidble.sample W/System.err:     at com.polidea.rxandroidble2.internal.connection.DisconnectionRouter.onDisconnectedException(DisconnectionRouter.java:99)
11-26 17:25:16.140 4798-4811/com.polidea.rxandroidble.sample W/System.err:     at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:77)
11-26 17:25:16.140 4798-4811/com.polidea.rxandroidble.sample W/System.err:     at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:228)
11-26 17:25:16.140 4798-4811/com.polidea.rxandroidble.sample W/System.err:     at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:70)
11-26 17:25:16.140 4798-4811/com.polidea.rxandroidble.sample W/System.err:     at android.os.Binder.execTransact(Binder.java:453)
11-26 17:25:16.140 4798-4811/com.polidea.rxandroidble.sample W/System.err: Caused by: com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from DD:E9:B8:3B:58:57 with status 133 (GATT_ERROR)
11-26 17:25:16.140 4798-4811/com.polidea.rxandroidble.sample W/System.err: 	... 4 more
11-26 17:25:16.140 4798-4811/com.polidea.rxandroidble.sample E/AndroidRuntime: FATAL EXCEPTION: Binder_1
    Process: com.polidea.rxandroidble.sample, PID: 4798
    io.reactivex.exceptions.UndeliverableException: The exception could not be delivered to the consumer because it has already canceled/disposed the flow or the exception has nowhere to go to begin with. Further reading: https://github.com/ReactiveX/RxJava/wiki/What's-different-in-2.0#error-handling | Disconnected from DD:E9:B8:3B:58:57 with status 133 (GATT_ERROR)
        at io.reactivex.plugins.RxJavaPlugins.onError(RxJavaPlugins.java:367)
        at io.reactivex.internal.operators.flowable.FlowableFlatMap$MergeSubscriber.innerError(FlowableFlatMap.java:608)
        at io.reactivex.internal.operators.flowable.FlowableFlatMap$InnerSubscriber.onError(FlowableFlatMap.java:672)
        at io.reactivex.internal.operators.single.SingleToFlowable$SingleToFlowableObserver.onError(SingleToFlowable.java:67)
[... cut ...]
     Caused by: com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from DD:E9:B8:3B:58:57 with status 133 (GATT_ERROR)
        at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:77) 
        at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:228) 
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:70) 
        at android.os.Binder.execTransact(Binder.java:453) 

### Library version
`1.7.1`

dariuszseweryn, 2018-12-03:
Have you checked the [FAQ](https://github.com/Polidea/RxAndroidBle/wiki/FAQ:-UndeliverableException)?

rouddy, 2018-12-07:
Thanks for your reply. I have not checked yet. I will try the solution.

closed 2018-12-07 by rouddy