dariuszseweryn/RxAndroidBle #508: Request connection params with callback from Android Oreo or higher

zetokz, 2018-11-27:
Android Oreo or higher allows receiving gatt callback about connection params update. It would be good to rely on this callback when request connection parameters instead of just waiting for a timeout to finish subscription.

dariuszseweryn, 2019-04-19:
I am looking into this request right now. I see a potential issue.

Right after connecting the OS changes connection parameters in order to speed up service discovery process. This behaviour can interfere with the proposed change. Any ideas?

dariuszseweryn, 2019-04-26:
I have created a PR that adds possibility of using the new callback and creating a custom request connection priority operation. I have not wired it up with the current API due to no possibility of distinguishing if a callback is related to a change in parameters due to the user's request, the system changing it or the peripheral.

dariuszseweryn, 2019-05-17:
Access to the updated parameters is available in `1.10.0` which should be available shortly. Unfortunately due to this not being a public API and an uncertain behaviour of the OS regarding params updates it is up to the user to implement it. E.g.
```java
@RequiresApi(26)
private Observable<ConnectionParameters> requestConnectionPriority(RxBleConnection rxBleConnection, int priority) {
    return Observable.merge(
            rxBleConnection.requestConnectionPriority(BluetoothGatt.CONNECTION_PRIORITY_HIGH, 1, TimeUnit.MILLISECONDS).toObservable(),
            rxBleConnection.observeConnectionParametersUpdates().take(1)
    );
}
```

PR #565 (2019-05-17): Added Connection Parameters Update Callback
The access to the callback is available through `RxBleConnection.observeConnectionParametersUpdates()`

Additionally it is possible to get it in `RxBleCustomOperation`:
- `RxBleGattCallback.getConnectionParametersUpdates()`
- `RxBleGattCallback.setHiddenGattCallback(HiddenBluetoothGattCallback)`

The callback is not used for completing `RxBleConnection.requestConnectionPriority()`. It is not possible to universally distinguish which callback is the one that should complete the request. Why? Android OS does change connection parameters on its own at t [... cut]
closed 2019-05-17 by dariuszseweryn