dariuszseweryn/RxAndroidBle #509: BLEScan is very Slow when already linked devices are in disconnected state.

padhur, 2018-11-29:
### Disclaimer
Before you will post a new issue to the RxAndroidBle library consider if you are filling a bug/feature request or a general question 
about:
* Usage of RxJava
* Usage of Bluetooth Low Energy
* Usage of RxAndroidBle
* Weird behaviour of the peripheral you are using
* Other not directly related to a new feature request or a bug

If any of the above points seems to match your use case — consider using Google or creating a question on www.stackoverflow.com with 
a tag `rxandroidble` where it will be easier to access by other people with similar questions. Issues that are not bugs or feature requests
will be closed.

### Summary
BLE Scan becomes slow and it takes longer time to scan the devices even its near by.
Even the scan is successful the connection takes longer.

### Library version
1.6.0

#### Preconditions
1.Connect 3 BLE devices to Android phone
2.Disconnect from 2  BLE devices
3.Now add a new BLE device and now the scan takes longer than before to get a successful scan-result. Even if the scan is successful, the connection takes more than time.
#### Steps to reproduce actual result
1.Connect 3 BLE devices to Android phone
2.Disconnect from 2  BLE devices
3.Now add a new BLE device and now the scan takes longer than before to get a successful scan-result. Even if the scan is successful, the connection takes more than time.
#### Minimum code snippet reproducing the issue

#### Logs from the application running with setting `RxBleLog.setLogLevel(RxBleLog.VERBOSE)`

#### Actual result
Scanning takes more time.
#### Expected result
Scan should be normal as when all the devices are connected.

P.S: We don't see any issue in IOS, issue exists  only in the Android platform

dariuszseweryn, 2018-12-04:
Hello,

Thanks for using the library. iOS has a totally different implementation of BLE stack so it is working differently there. 

Some questions:
1. Have you compared to the vanilla Android API usage? 
2. Have you checked the [FAQ](https://github.com/Polidea/RxAndroidBle/wiki/FAQ:-Cannot-connect)? It is related to scanning
3. Have you checked against the newest version of the library?

padhur, 2018-12-10:
Hi Dariuszeweryn,

Thanks for the quick response and input.

Please find the answers inline

Have you compared to the vanilla Android API usage?
No.I believe the Vanilla and RxBle API interaction logically(w.r.t to scan, connection) remains the same except Rx way.Could you please give me an brief on the major difference if any

Have you checked the FAQ? It is related to scanning
yes.Its not exactly related the scanning.

Have you checked against the newest version of the library?
No.I am using 1.6.0-RxJava2, i think the issue may present in the latest version as well, i can confirm it after testing with same.

Issue:
====

As per my observation, on the below devices
1.Pixel 2 - Android 8.1.0
2. Pixel XL - Android 8.0
3.Samsung S8,S9 - Android 8.1.0
4. Samsung J7 Prime - Android 7.1.1
5. Xiaomi Mi - Android 8.1.0

1.Link 3 or more BLE devices to Android based Handsets
2.Forcibly disconnect all the three Linked devices
3.Now, try to link a new BLE device to same Android handset
4.The scan takes very long time(more than 2 mins some times never) to get the successful scan result of the new BLE device

In the above case, when there is a disconnect we are immediately trying to reconnect to the disconnected devices( rxBleDevice.establishConnection(false) ) on the state-change callback.
So the request is queued and assume there are three rxBleDevice.establishConnection(false) request at a time and now a scan request has been issued. Until unless the above connection requests completes  either success or fail, the scan request never served.The issue is not observed when all the three devices are in connected state.

So, i have fixed it by, do not issue reconnect to the devices immediately on the state-change callback instead add a new scan thread with SCAN_MODE_LOW_POWER and if the bonded device is present in the scan-result go for the connection.
Now, i don not see the scan delay issue.

Please advice if there is any best solution.

Thanks,
Paddy

dariuszseweryn, 2018-12-10:
Establishing connection and starting a scan are sharing the same queue. `.establishConnection(false)` is blocking so no scan will start until the connection is either a success or a failure. First starting the scan before establishing the connection is subject to potential issues as described in the FAQ.

padhur, 2018-12-10:
okay got it, thanks Dariusz.

dariuszseweryn, 2018-12-10:
Closing if that is all

closed 2018-12-10 by dariuszseweryn