dariuszseweryn/RxAndroidBle #510: Can't write characteristic, status=1

cisekluke, 2018-11-29:
### Summary

Hi,

I get crash with status=1 when app trying to write the characteristic on two phones:
* LG G5,  Android 8
* Motorola G6 Android 7

I've checked is there any missing onError handling in my code but all places contain error handling. What's more, this error wasn't presented before (or I didn't encounter him) I updated to the library version `1.7.1`, previously I used `1.6.0`.
Unfortunately, I'm not able to reproduce this crash by normal using the app. I have only exceptions from Fabric.

In logs I also see BleGattCharacteristicException but this exception wasn't populated down to the stream.

```
Fatal Exception: io.reactivex.exceptions.OnErrorNotImplementedException: The exception was not handled due to missing onError handler in the subscribe() method call. Further reading: https://github.com/ReactiveX/RxJava/wiki/Error-Handling | GATT exception from MAC address 65:8C:31:6C:2F:8E, status 1 (GATT_INVALID_HANDLE), type BleGattOperation{description='CHARACTERISTIC_WRITE'}. (Look up status 0x01 here https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h)
       at io.reactivex.internal.observers.CallbackCompletableObserver.accept(CallbackCompletableObserver.java:47)
       at io.reactivex.internal.observers.CallbackCompletableObserver.accept(CallbackCompletableObserver.java:26)
       at io.reactivex.internal.observers.CallbackCompletableObserver.onError(CallbackCompletableObserver.java:64)
       at io.reactivex.internal.operators.completable.CompletableDoFinally$DoFinallyObserver.onError(CompletableDoFinally.java:72)
       at io.reactivex.internal.operators.completable.CompletableMergeArray$InnerCompletableObserver.onError(CompletableMergeArray.java:80)
       at io.reactivex.internal.operators.single.SingleFlatMapCompletable$FlatMapCompletableObserver.onError(SingleFlatMapCompletable.java:97)
       at io.reactivex.internal.operators.completable.CompletableFromSingle$CompletableFromSingleObserver.onError(CompletableFromSingle.java:41)
       at io.reactivex.internal.operators.observable.ObservableElementAtSingle$ElementAtObserver.onError(ObservableElementAtSingle.java:102)
       at io.reactivex.internal.operators.observable.ObservableFlatMapSingle$FlatMapSingleObserver.drainLoop(ObservableFlatMapSingle.java:239)
       at io.reactivex.internal.operators.observable.ObservableFlatMapSingle$FlatMapSingleObserver.drain(ObservableFlatMapSingle.java:210)
       at io.reactivex.internal.operators.observable.ObservableFlatMapSingle$FlatMapSingleObserver.innerError(ObservableFlatMapSingle.java:202)
       at io.reactivex.internal.operators.observable.ObservableFlatMapSingle$FlatMapSingleObserver$InnerObserver.onError(ObservableFlatMapSingle.java:289)
       at io.reactivex.internal.operators.single.SingleFlatMap$SingleFlatMapCallback$FlatMapSingleObserver.onError(SingleFlatMap.java:116)
       at io.reactivex.internal.operators.observable.ObservableElementAtSingle$ElementAtObserver.onError(ObservableElementAtSingle.java:102)
       at io.reactivex.internal.operators.mixed.CompletableAndThenObservable$AndThenObservableObserver.onError(CompletableAndThenObservable.java:70)
       at io.reactivex.internal.operators.observable.ObservableCreate$CreateEmitter.tryOnError(ObservableCreate.java:84)
       at com.polidea.rxandroidble2.internal.serialization.FIFORunnableEntry$1.onError(FIFORunnableEntry.java:71)
       at io.reactivex.internal.operators.observable.ObservableUnsubscribeOn$UnsubscribeObserver.onError(ObservableUnsubscribeOn.java:70)
       at io.reactivex.internal.operators.observable.ObservableSubscribeOn$SubscribeOnObserver.onError(ObservableSubscribeOn.java:63)
       at io.reactivex.internal.operators.observable.ObservableCreate$CreateEmitter.tryOnError(ObservableCreate.java:84)
       at com.polidea.rxandroidble2.internal.util.QueueReleasingEmitterWrapper.onError(QueueReleasingEmitterWrapper.java:45)
       at io.reactivex.internal.observers.DeferredScalarDisposable.error(DeferredScalarDisposable.java:100)
       at io.reactivex.internal.operators.single.SingleToObservable$SingleToObservableObserver.onError(SingleToObservable.java:77)
       at io.reactivex.internal.operators.single.SingleTimeout$TimeoutMainObserver.onError(SingleTimeout.java:142)
       at io.reactivex.internal.operators.single.SingleMap$MapSingleObserver.onError(SingleMap.java:69)
       at io.reactivex.internal.operators.observable.ObservableElementAtSingle$ElementAtObserver.onError(ObservableElementAtSingle.java:102)
       at io.reactivex.internal.observers.BasicFuseableObserver.onError(BasicFuseableObserver.java:100)
       at io.reactivex.internal.operators.observable.ObservableObserveOn$ObserveOnObserver.checkTerminated(ObservableObserveOn.java:281)
       at io.reactivex.internal.operators.observable.ObservableObserveOn$ObserveOnObserver.drainNormal(ObservableObserveOn.java:172)
       at io.reactivex.internal.operators.observable.ObservableObserveOn$ObserveOnObserver.run(ObservableObserveOn.java:255)
       at io.reactivex.internal.schedulers.ExecutorScheduler$ExecutorWorker$BooleanRunnable.run(ExecutorScheduler.java:260)
       at io.reactivex.internal.schedulers.ExecutorScheduler$ExecutorWorker.run(ExecutorScheduler.java:225)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
       at java.lang.Thread.run(Thread.java:764)

Caused by com.polidea.rxandroidble2.exceptions.BleGattCharacteristicException: GATT exception from MAC address 65:8C:31:6C:2F:8E, status 1 (GATT_INVALID_HANDLE), type BleGattOperation{description='CHARACTERISTIC_WRITE'}. (Look up status 0x01 here https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/android-5.1.0_r1/stack/include/gatt_api.h)
       at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback.propagateErrorIfOccurred(RxBleGattCallback.java:228)
       at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback.access$800(RxBleGattCallback.java:33)
       at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onCharacteristicWrite(RxBleGattCallback.java:122)
       at android.bluetooth.BluetoothGatt$1$7.run(BluetoothGatt.java:407)
       at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:711)
       at android.bluetooth.BluetoothGatt.-wrap0(Unknown Source)
       at android.bluetooth.BluetoothGatt$1.onCharacteristicWrite(BluetoothGatt.java:403)
       at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:136)
       at android.os.Binder.execTransact(Binder.java:674)
```

BTW. Thanks for the library it's really great.

cisekluke, 2019-01-07:
Yes, I'm using this method.

dariuszseweryn, 2019-01-07:
Then most probably you are obtaining the `BluetoothGattCharacteristic` during one connection and you try to write it on another connection. `BluetoothGattCharacteristic` is valid only for a single/same connection.

A safer method (but a bit slower) is to use `RxBleConnection.writeCharacteristic(UUID, byte[])` which will internally use the `BluetoothGattCharacteristic` from the given connection.

closed 2019-01-18 by dariuszseweryn