dariuszseweryn/RxAndroidBle #513: can't write characteristic to communicate with ESP32

EverSinceWWI, 2018-12-03:
I can't find a way to make my android device write characteristic with RxAndroidBle to send strings to an ESP32 microcontroller (Arduino code). The two devices can establish a connection but no values shows up in the Arduino serial monitor.
Regarding the Arduino part, everything seems to work as expected since i tested to send variables manually with my phone through nRF Connect.
Here's the code:
[image]
The code is running inside a listener from the Mapbox SDK.

I'm by no means an expert in java or BLE communication, but since the compiler isn't giving me any error i can't understand where the problem can reside.

Thanks in advance

jreehuis, 2018-12-03:
Hi, I guess the issue is not in the lib.
You are subscribing to a long running RxJava chain and disposing it immediately after it. So the whole call gets canceled before it could be executed completely.

EverSinceWWI, 2018-12-03:
> 
> 
> Hi, I guess the issue is not in the lib.
> You are subscribing to a long running RxJava chain and disposing it immediately after it. So the whole call gets canceled before it could be executed completely.

as soon as i moved the disposable.dispose() line everything worked perfectly! I feel so stupid for not having though about it earlier :P
Thank you so much!

closed 2018-12-03 by dariuszseweryn