dariuszseweryn/RxAndroidBle #514: There is no response from writeDescriptor() when used in a chain

akhbulatov, 2018-12-07:
Hey. I have encountered such a problem when no response comes from method `writeDescriptor` when used in a chain. Here is my code:
```
override fun connectDevice() {
        rxBleClient.scanBleDevices(scanSettings, scanFilter)
            .flatMap { scanResult -> scanResult.bleDevice.establishConnection(false) }
            .flatMapSingle {
                rxBleConnection = it
                it.discoverServices()
            }
            .flatMapSingle { obtainCharacteristics(it) }
            .flatMapCompletable {
                readCharacteristic = it.first
                writeCharacteristic = it.second

                writeDescriptor(it.first)
            }
            .andThen(Single.fromCallable { startConnection() })
            .flatMap { it }
            .subscribeBy(
                onSuccess = { setupAndObserveNotifications(readCharacteristic!!) },
                onError = { Timber.d("$it") }
            )
    }
```

My method `writeDescriptor`:
```
private fun writeDescriptor(readCharacteristic: BluetoothGattCharacteristic): Completable {
        val descriptor = readCharacteristic.getDescriptor(READ_CHARACTERISTIC_CONFIG)
        return rxBleConnection!!.writeDescriptor(descriptor, BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE)
            .doOnComplete { Timber.d("Descriptor wrote.") }
            .doOnError { Timber.e("Error while descriptor is writing: $it") }
    }
```

The chain comes to `writeDescriptor(it.first)`,  also called `doOnComplete { Timber.d("Descriptor wrote.")`,  but it does not go further down the chain in `andThen(Single.fromCallable { startConnection() })` and next in `subscribe`. What could be the problem?

P.S. I reproduced the same example with the chain separately (without the library) with`flatMapCompletable`, `andThen` and etc., and everything was done correctly.

```
fun main() {
    Single.just(1)
        .delay(1, TimeUnit.SECONDS)
        .flatMapCompletable { getCompletable() }
        .andThen(Single.fromCallable { getSingle() })
        .flatMap { it }
        .subscribe { i -> println(i) }

    Thread.sleep(5000)
}

private fun getCompletable(): Completable {
    return Completable.fromRunnable {
        Thread.sleep(1000)
    }
}

private fun getSingle(): Single<Int> {
    return Single.just(5)
}
```
In the console comes the correct answer - 5

dariuszseweryn, 2018-12-07:
`.establishConnection()` does not complete on its own. You can read about it in the [`Observable behaviour` part of Readme](https://github.com/Polidea/RxAndroidBle#observable-behaviour)

akhbulatov, 2018-12-07:
@dariuszseweryn thank you, I realized what was the problem and fixed it. Here's a solution if anyone else needs it:

```
override fun connectDevice() {
        rxBleClient.scanBleDevices(scanSettings, scanFilter)
            .flatMap { scanResult -> scanResult.bleDevice.establishConnection(false) }
            .flatMapSingle {
                rxBleConnection = it
                it.discoverServices()
            }
            .flatMapSingle { obtainCharacteristics(it) }
            .subscribeBy(
                onNext = {
                    writeCharacteristic = it.second
                    proceedConnectDevice(it.first)
                },
                onComplete = {},
                onError = { Timber.d("$it") }
            )
    }

private fun proceedConnectDevice(readCharacteristic: BluetoothGattCharacteristic) {
        writeDescriptor(readCharacteristic)
            .andThen(Single.fromCallable { startConnection() })
            .flatMap { it }
            .flatMapObservable { setupAndObserveNotifications(readCharacteristic) }
            .subscribeBy(
                onNext = { keepConnection() },
                onComplete = {},
                onError = {}
            )
    }

akhbulatov, 2018-12-07:
@dariuszseweryn I would like to clarify one point: I did not notice the use of `subscribeOn` in sample. Requests in the library by default are not executed on the current thread?

dariuszseweryn, 2018-12-07:
> Requests in the library by default are not executed on the current thread?

Yes. Android BLE API is very sensitive (read poorly implemented) to threading so the library handles it on its own. The Android API is asynchronous anyway

akhbulatov, 2018-12-07:
Ok, thanks.

closed 2018-12-07 by akhbulatov