dariuszseweryn/RxAndroidBle #525: State.BLUETOOTH_NOT_ENABLED in observeStateChanges is called twice

akhbulatov, 2018-12-29:
Hi. I have a problem with the fact that `State.BLUETOOTH_NOT_ENABLED` in `observeStateChanges` is called twice. After the first call I connect Bluetooth via Intent, and after connecting, this state is called again first, and then `State.LOCATION_SERVICES_NOT_ENABLED`. Perhaps this behavior is not related to the library, but to my lack of knowledge of RxJava, but I would like to know the answer. Here is the log:

dariuszseweryn, 2019-01-02:
This is an implementation detail of the library. When the user turns the BT adapter ON the predicted adapter changes are `BluetoothAdapter.STATE_OFF -> BluetoothAdapter.STATE_TURNING_ON` and `BluetoothAdapter.STATE_TURNING_ON -> BluetoothAdapter.STATE_ON`.

Currently `rxBleClient.observeStateChanges()` does not track the state at the subscription time and only relays changes. The first change is `BluetoothAdapter.STATE_OFF -> BluetoothAdapter.STATE_TURNING_ON` which translates to `RxBleClient.State.BLUETOOTH_NOT_ENABLED` (it is the same for `BluetoothAdapter.STATE_OFF` and `BluetoothAdapter.STATE_TURNING_OFF` as well)

If that is an issue for you — you can quickly workaround by using:
```
        rxBleClient.observeStateChanges()
            .startWith(rxBleClient.state)
            .distinctUntilChanged()
```

akhbulatov, 2019-01-02:
Okay, thanks for the answer)

closed 2019-01-02 by akhbulatov