dariuszseweryn/RxAndroidBle #532: BleAlreadyConnectedException: Already connected to device with MAC address

akhbulatov, 2019-01-17:
Hey. When I connect to the device with off states, I get an error `BleAlreadyConnectedException`. If after this, reopen the app (with Bluetooth enabled and location), the connection is successful. I looked [here](https://github.com/Polidea/RxAndroidBle/issues/466), but it didn't clear up my problem.

Initial state: Bluetooth and location are off, permission to get location has not been requested yet.

Connection code:
```
override fun connect() {
        subscriptions += rxBleClient.scanBleDevices(scanSettings, scanFilter)
            .flatMap { scanResult ->
                rxBleDevice = scanResult.bleDevice
                scanResult.bleDevice.establishConnection(false)
            }
            .flatMapSingle {
                rxBleConnection = it
                it.discoverServices()
            }
            .flatMapSingle { obtainCharacteristics(it) }
            .flatMapSingle {
                readCharacteristic = it.first
                writeCharacteristic = it.second

                writeDescriptor(it.first)
                    .andThen(Single.fromCallable { sendStartConnection() })
            }
            .flatMapSingle { it }
            .subscribeBy(
                onNext = {
                    Timber.d("Connect to a device is next: ${it?.contentToString()}")
                    setupAndObserveNotification(readCharacteristic!!)
                },
                onComplete = { Timber.d("Connect to a device is completed.") },
                onError = { Timber.e("Error while connecting to a device: $it") }
            )
}
```

Log error:
```
E/BluetoothLeRepositoryImpl$observeStateChanges: Location permission not granted.
D/ConnectDeviceFragment$enableLocation: ACCESS_FINE_LOCATION permission is granted.
E/BluetoothLeRepositoryImpl$observeStateChanges: Bluetooth not enabled.
D/BluetoothAdapter: onBluetoothServiceUp: android.bluetooth.IBluetooth$Stub$Proxy@e69a46f
E/BluetoothLeRepositoryImpl$observeStateChanges: Location services not enabled.
D/BluetoothLeRepositoryImpl$observeStateChanges: Bluetooth state is ready.
D/RxBle#ClientOperationQueue: QUEUED   ScanOperationApi21(139653596)
D/RxBle#ClientOperationQueue: STARTED  ScanOperationApi21(139653596)
I/RxBle#QueueOperation: Scan operation is requested to start.
D/BluetoothAdapter: isLeEnabled(): ON
D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=5 mScannerId=0
D/RxBle#ClientOperationQueue: FINISHED ScanOperationApi21(139653596) in 47 ms
D/RxBle#ClientOperationQueue: QUEUED   ConnectOperation(123558721)
D/RxBle#ClientOperationQueue: STARTED  ConnectOperation(123558721)
V/RxBle#BleConnectionCompat: Connecting without reflection
D/BluetoothGatt: connect() - device: 9C:1D:58:92:85:E3, auto: false
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=cb82dafd-ba5f-47e6-851c-5f8fc43eae1e
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=8
I/RxBle#CancellableDisposable: Scan operation is requested to stop.
D/BluetoothAdapter: isLeEnabled(): ON
E/BluetoothLeRepositoryImpl$connect: Error while connecting to a device: com.polidea.rxandroidble2.exceptions.BleAlreadyConnectedException: Already connected to device with MAC address 9C:1D:58:92:85:E3
I/RxBle#ConnectionOperationQueue: Connection operations queue to be terminated (9C:1D:58:92:85:E3)
D/RxBle#Executors$RunnableAdapter: Terminated.
D/RxBle#ClientOperationQueue: FINISHED ConnectOperation(123558721) in 179 ms
D/RxBle#ClientOperationQueue: QUEUED   DisconnectOperation(53321255)
D/RxBle#ClientOperationQueue: STARTED  DisconnectOperation(53321255)
D/BluetoothManager: getConnectionState()
D/BluetoothManager: getConnectedDevices
D/BluetoothGatt: close()
D/BluetoothGatt: unregisterApp() - mClientIf=8
D/RxBle#ClientOperationQueue: FINISHED DisconnectOperation(53321255) in 9 ms
```

Log successful (after a previous failed connection):
```
D/BluetoothLeRepositoryImpl$observeStateChanges: Bluetooth state is ready.
D/RxBle#ClientOperationQueue: QUEUED   ScanOperationApi21(72294046)
D/RxBle#ClientOperationQueue: STARTED  ScanOperationApi21(72294046)
I/RxBle#QueueOperation: Scan operation is requested to start.
D/BluetoothAdapter: isLeEnabled(): ON
D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=7 mScannerId=0
D/RxBle#ClientOperationQueue: FINISHED ScanOperationApi21(72294046) in 16 ms
D/RxBle#ClientOperationQueue: QUEUED   ConnectOperation(118003238)
D/RxBle#ClientOperationQueue: STARTED  ConnectOperation(118003238)
V/RxBle#BleConnectionCompat: Connecting without reflection
D/BluetoothGatt: connect() - device: 9C:1D:58:92:85:E3, auto: false
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=b5748661-d640-4d50-8e04-ebd2900983c7
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=8
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=8 device=9C:1D:58:92:85:E3
D/RxBle#BluetoothGatt$1: onConnectionStateChange newState=2 status=0
D/RxBle#ClientOperationQueue: FINISHED ConnectOperation(118003238) in 488 ms
D/RxBle#ConnectionOperationQueue: QUEUED   ServiceDiscoveryOperation(250099901)
D/RxBle#ConnectionOperationQueue: STARTED  ServiceDiscoveryOperation(250099901)
D/BluetoothGatt: discoverServices() - device: 9C:1D:58:92:85:E3
```

dariuszseweryn, 2019-01-17:
I believe this error is correct. You try to connect more than once to the same device.

akhbulatov, 2019-01-17:
@dariuszseweryn and why it tries to connect twice?

dariuszseweryn, 2019-01-17:
> The problem is my connection method I added on top?

Yes
> and why it tries to connect twice? 

I believe that you would learn more if you would figure it out for yourself. To make it easier the below code will exhibit the same behaviour:

```
override fun connect() {
        subscriptions += rxBleClient.scanBleDevices(scanSettings, scanFilter)
            .flatMap { scanResult ->
                rxBleDevice = scanResult.bleDevice
                scanResult.bleDevice.establishConnection(false)
            }
            .subscribeBy(
                onNext = { Timber.d("Connect to a device is next: ${it?.contentToString()}") },
                onComplete = { Timber.d("Connect to a device is completed.") },
                onError = { Timber.e("Error while connecting to a device: $it") }
            )
}
```

If you would need more assistance please head to stackoverflow with a tags like `rxjava` and `rxandroidble` as it was described in the disclaimer because this is more a general usage question than an issue with the library

Best Regards

closed 2019-01-17 by dariuszseweryn

lby1992, 2019-11-28:
Hi, @akhbulatov . One device may be returned more than once in your code above.So, are you looking for like this:
```
subscriptions += rxBleClient.scanBleDevices(scanSettings, scanFilter)
            .take(1) // just take the first one
            .flatMap { scanResult ->
                rxBleDevice = scanResult.bleDevice
                scanResult.bleDevice.establishConnection(false)
            }
            ...
```