dariuszseweryn/RxAndroidBle #534: Cannot write client characteristic config descriptor (code 3) with characteristic UUID if the device is already paired

mohank1288, 2019-01-20:
Not able to write client characteristic if the BLE peripheral is in paired device list.
It's working fine if the device is paired and connected for the first time, we are able read and write characteristic from the BLE peripheral.
After removing the BLE peripheral from the paired devices list and try to pair and connect with the BLE Peripheral device from the android application it's working fine.

Logs for the issue is mentioned below

 com.polidea.rxandroidble2.exceptions.BleCannotSetCharacteristicNotificationException: Cannot write client characteristic config descriptor (code 3) with characteristic UUID 00002a52-0000-1000-8000-00805f9b34fb
        at com.polidea.rxandroidble2.internal.connection.NotificationAndIndicationManager$6.apply(NotificationAndIndicationManager.java:194)
        at com.polidea.rxandroidble2.internal.connection.NotificationAndIndicationManager$6.apply(NotificationAndIndicationManager.java:191)
        at io.reactivex.internal.operators.completable.CompletableResumeNext$ResumeNextObserver.onError(CompletableResumeNext.java:82)
        at io.reactivex.internal.operators.observable.ObservableIgnoreElementsCompletable$IgnoreObservable.onError(ObservableIgnoreElementsCompletable.java:61)
        at io.reactivex.internal.operators.observable.ObservableCreate$CreateEmitter.tryOnError(ObservableCreate.java:84)
        at com.polidea.rxandroidble2.internal.serialization.FIFORunnableEntry$1.onError(FIFORunnableEntry.java:71)
        at io.reactivex.internal.operators.observable.ObservableUnsubscribeOn$UnsubscribeObserver.onError(ObservableUnsubscribeOn.java:70)
        at io.reactivex.internal.operators.observable.ObservableSubscribeOn$SubscribeOnObserver.onError(ObservableSubscribeOn.java:63)
        at io.reactivex.internal.operators.observable.ObservableCreate$CreateEmitter.tryOnError(ObservableCreate.java:84)
        at com.polidea.rxandroidble2.internal.util.QueueReleasingEmitterWrapper.onError(QueueReleasingEmitterWrapper.java:45)
        at io.reactivex.internal.observers.DeferredScalarDisposable.error(DeferredScalarDisposable.java:100)
        at io.reactivex.internal.operators.single.SingleToObservable$SingleToObservableObserver.onError(SingleToObservable.java:77)
        at io.reactivex.internal.operators.single.SingleTimeout$TimeoutMainObserver$TimeoutFallbackObserver.onError(SingleTimeout.java:91)
        at io.reactivex.internal.disposables.EmptyDisposable.error(EmptyDisposable.java:78)
        at io.reactivex.internal.operators.single.SingleError.subscribeActual(SingleError.java:42)
        at io.reactivex.Single.subscribe(Single.java:3438)
        at io.reactivex.internal.operators.single.SingleTimeout$TimeoutMainObserver.run(SingleTimeout.java:118)
        at io.reactivex.internal.schedulers.ScheduledDirectTask.call(ScheduledDirectTask.java:38)
        at io.reactivex.internal.schedulers.ScheduledDirectTask.call(ScheduledDirectTask.java:26)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:301)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:764)
     Caused by: com.polidea.rxandroidble2.exceptions.BleGattCallbackTimeoutException: GATT exception from MAC address E2:F9:C7:27:22:72, with type BleGattOperation{description='DESCRIPTOR_WRITE'}
        at com.polidea.rxandroidble2.internal.SingleResponseOperation.timeoutFallbackProcedure(SingleResponseOperation.java:91)
        at com.polidea.rxandroidble2.internal.SingleResponseOperation.protectedRun(SingleResponseOperation.java:54)
        at com.polidea.rxandroidble2.internal.QueueOperation$1.subscribe(QueueOperation.java:41)
        at io.reactivex.internal.operators.observable.ObservableCreate.subscribeActual(ObservableCreate.java:40)
        at io.reactivex.Observable.subscribe(Observable.java:12090)
        at io.reactivex.internal.operators.observable.ObservableSubscribeOn$SubscribeTask.run(ObservableSubscribeOn.java:96)
        at io.reactivex.internal.schedulers.ScheduledDirectTask.call(ScheduledDirectTask.java:38)
        at io.reactivex.internal.schedulers.ScheduledDirectTask.call(ScheduledDirectTask.java:26)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167) 
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641) 
        at java.lang.Thread.run(Thread.java:764) 

### Disclaimer
Before you will post a new issue to the RxAndroidBle library consider if you are filling a bug/feature request or a general question 
about:
* Usage of RxJava
* Usage of Bluetooth Low Energy
* Usage of RxAndroidBle
* Weird behaviour of the peripheral you are using
* Other not directly related to a new feature request or a bug

If any of the above points seems to match your use case — consider using Google or creating a question on www.stackoverflow.com with 
a tag `rxandroidble` where it will be easier to access by other people with similar questions. Issues that are not bugs or feature requests
will be closed.

### Summary

### Library version
`X.X.X`

#### Preconditions

#### Steps to reproduce actual result
<br />1.
<br />2.
<br />3.
#### Minimum code snippet reproducing the issue

#### Logs from the application running with setting `RxBleLog.setLogLevel(RxBleLog.VERBOSE)`

#### Actual result

#### Expected result

dariuszseweryn, 2019-01-23:
Hello,

It seems to be an Android bug — have you tried to recreate the issue with the vanilla Android API?

dariuszseweryn, 2019-02-01:
From my experience the Android BLE stack does not always work well with bound devices.
The log you have provided is only the stacktrace without any information on what happened before unfortunately. There is no information about what peripheral you're talking to nor on what device model / OS version your app runs — too small amount of actionable info. I am closing until more info will be added

Best Regards

closed 2019-02-01 by dariuszseweryn

BenelliFurtado, 2025-03-04:
If anybody stumbles upon this: I noticed this behvior on my Custom Characteristic after the device was bonded and then reconnected. It sometimes worked and sometimes not. For me the solution was to request a higher(?) MTU (I just did not request any MTU)

    peripheral.connect(requestMtu: 239);
    // or await peripheral.requestMtu(239);