dariuszseweryn/RxAndroidBle #536: Scanning stops when screen turned off

Steppschuh, 2019-01-22:
### Summary
If scanning without a filter on devices running Android 8.1 or newer, scanning will stop when the screen turns off (and resume when turned back on).

See:
- [Related issue on StackOverflow](https://stackoverflow.com/questions/48077690/ble-scan-is-not-working-when-screen-is-off-on-android-8-1-0)
- [Related commit in Android Source](https://android.googlesource.com/platform/packages/apps/Bluetooth/+/319aeae6f4ebd13678b4f77375d1804978c4a1e1)

#### Library version
`1.6.0` & `1.8.0` (should also happen in all other versions)

#### Preconditions
Deploy on a device running Android 8.1 or newer.

#### Steps to reproduce actual result

1. Start scanning without a filter
2. Turn screen off

#### Minimum code snippet reproducing the issue
```java
rxBleClient = RxBleClient.create(context);
scanSettings = new ScanSettings.Builder().build();
rxBleClient.scanBleDevices(scanSettings).subscribe(
    scanResult -> Log.v("BLE Scanner", scanResult.toString())
);
```

#### Actual result
No scan results received as long as screen is turned off.

#### Expected result
Same scan results received as if screen would be on.

#### Suggested solution
[Add an empty scan filter](https://stackoverflow.com/a/48079800/1188330).

dariuszseweryn, 2019-01-23:
Hello,

Thanks for using the library. I have some thoughts

> If scanning without a filter on devices running Android 8.1 or newer, scanning will stop when the screen turns off (and resume when turned back on).

It is a behaviour change for newer Android OS systems, not a bug. Why do you think this should be worked around by this library?

In the suggested solution it is written that:
> EDIT 2: On the Galaxy Note 9 with Android 8.1 and perhaps other Samsung devices with 8.1, scans are blocked with the screen off even with an empty scan filter. Scans are allowed with the screen off with a non-empty scan filter as described above.

So it is not a bulletproof solution and in my opinion it may confuse users when on some devices the scan stops if the screen is turned off and on other not. Additionally the user may simply add an empty `ScanFilter` by themselves if they see it necessary (unless the app is a generic BLE tool app it should probably have a non-empty filter list anyway).

Steppschuh, 2019-01-24:
After thinking more about it I agree that the library should not work around this.

However, I stumbled upon this issue because this behaviour change was not well documented / announced. It could save other developers some time if the library would print an informational message if it detects that someone is scanning without a filter on 8.1 or later. Maybe there's room for a section mentioning this change in the readme, too.

closed 2019-01-24 by Steppschuh