dariuszseweryn/RxAndroidBle #541: [Proposal] Explain usage of ReplayingShare by Jake

davordev, 2019-02-06:
I've been fighting with this framework for a while to properly set up sharing of the established connection. It seems that the real issue was with me not providing `ReplayingShare.instance()` inside `compose` operator on `Observable<RxBleConnection>`. You might want to add tips on how to properly share connection to avoid future miss understanding.

dariuszseweryn, 2019-02-06:
Thanks for the feedback. A question for my understanding: wasn't the description on [`RxReplayingShare`](https://github.com/JakeWharton/RxReplayingShare) enough or is it hard to find?

davordev, 2019-02-06:
RxReplayingShare isn't mentioned anywhere in the docs. I literally had to go through samples to find out on how to properly set up connection sharing. It was hard to find, but once I found it, I had no idea is it going to let me share the connection. I fixed my issue by trial and error method

RobLewis, 2019-02-11:
For those who may have dealt with this issue in the past and need catching up, Jake has updated RxReplayingShare so that any terminal event from upstream will cause the cached value to be nulled out. So, for example, if the `RxBleConnection` is lost, future subscribers won't be served a stale, invalid instance.

dariuszseweryn, 2019-02-14:
Hmmm... Indeed. I thought the `ConnectionSharingAdapter` was put in the `helpers` but it is in `utils` package which may make it harder to find.

Personally I am against using any kind of sharing adapters — one should strive for using a single `.subscribe()` and just model the observable/operator chain correctly. Unfortunately there is no "one size fits all" solution for that.

If you wish — you can prepare points with what you would like to see in the Wiki as a newcomer. If you have already learned all you want you could help us all and write down a tutorial which I could then put on a Wiki page

RobLewis, 2019-02-14:
Wait—`ConnectionSharingAdapter` is deprecated, right? (for reasons that were never clear to me). I thought you now recommend using `RxReplayingShare` instead.

dariuszseweryn, 2019-02-14:
That is true — it is deprecated. It has a message that points to `RxReplayingShare`. Thing is that it still may be hard to discover

RobLewis, 2019-02-14:
As for the wiki, a concrete example of "modeling the observable/operator chain correctly" vs. sharing a connection would help this not-exactly-a-newcomer.

dariuszseweryn, 2019-02-14:
I recommend this [talk by Jake Wharton](https://www.youtube.com/watch?v=0IKHxjkgop4) — I tend to post a link to it in many places on the web — it is really great in developing a good mindset for RxJava

davordev, 2019-02-15:
Personally, I am against using operator to preserve the connection or even putting the connection stream to Disposable (I had some real unlogic issues while applying this framework to Clean Architecture). The best, user-friendly way would be to have two functions on RxBleClient:
- getConnection(): Observable<RxBleConnection> throws RxBleDeviceNotConnected
- closeConnection(cleanUpCode: () -> Unit): Unit

There may be some syntax errors, but generally, you get the idea.
Cheers

dariuszseweryn, 2019-02-15:
This approach was evaluated at the very beginning of this library development. It has been deliberately rejected as it introduces state to something that is basically a signal processing communication.

In my experience having more state in the app may seem to be user-friendly at first but it does not model BLE connections well enough and tend to be the cause of many hard-to-debug issues. Additionally the code that has more state is harder to understand.

If you have overcome the obstacle and reintroduced state — you may share your results so others could benefit.

Best Regards

davordev, 2019-02-15:
I could possibly make a gist. Is it okay if I post it here?

dariuszseweryn, 2019-02-15:
Of course you can

Even if I may be against some solutions to particular class of problems does not mean that other people could find it helpful. Depending on my time I try to work on Wiki pages and your gist may be at some point referenced there — it would be easier to find by others

davordev, 2019-03-04:
Sorry for the delay. Here's an example:

BleSessionManager.kt

```
/**
 * Manages Bluetooth connection.
 * @property rxBleClient RxBleClient
 * @property mSchedulerProvider SchedulerProvider
 * @property eventStream CompositeDisposable
 * @property macAddress String?
 * @property disconnectTriggerSubject PublishSubject<Boolean>
 * @property _connection Observable<RxBleConnection>?
 * @property connection Observable<RxBleConnection>
 * @constructor
 */
class BleSessionManager(
    val rxBleClient: RxBleClient,
    val mSchedulerProvider: SchedulerProvider
) {

    private val eventStream: CompositeDisposable = CompositeDisposable()
    private var macAddress: String? = null
    private val disconnectTriggerSubject: PublishSubject<Boolean> = PublishSubject.create<Boolean>()
    private var _connection: Observable<RxBleConnection>? = null

    var connection: Observable<RxBleConnection>
        get() {
            return _connection ?: connectionFallback()
        }
        private set(value) {
            this._connection = value
        }

    /**
     * Fallback in case connection isn't valid.
     * @return Observable<RxBleConnection>
     */
    private fun connectionFallback(): Observable<RxBleConnection> {
        return Observable.error(NotConnectedException())
    }

    /**
     * Used to set mac address.
     * @param macAddress String
     */
    fun setMacAddress(macAddress: String) {
        this.macAddress = macAddress
    }

    /**
     * Configures connection. Called after `setMacAddress()`.
     * @throws NotFoundException May throw NotFoundException if mac address is not set.
     */
    fun setupConnection() {
        val address = this.macAddress ?: throw NotFoundException()

        val observable = rxBleClient.getBleDevice(address)
            .establishConnection(true, Timeout(BLE_CONNECT_OPERATION_TIMEOUT, TimeUnit.SECONDS))
            .takeUntil(disconnectTriggerSubject)
            .compose(ReplayingShare.instance())

        // Subscribe actions should be implemented even though they have no implementation.
        val disposable = observable
            .subscribeOn(mSchedulerProvider.io())
            .retry(10)
            .subscribe(
                {},
                {},
                {}
            )
        eventStream.add(disposable)
        connection = observable
    }

    /**
     * Shuts down the connection and clears event stream.
     */
    private fun closeConnection() {
        disconnectTriggerSubject.onNext(true)
        eventStream.clear()
    }

    /**
     * Nullifies connection. Triggered by End User.
     */
    fun disconnect() {
        closeConnection()
        _connection = null
    }

    /**
     * Nullifies connection, but executes the given code before. Triggered by End User.
     */
    fun disconnectButFirst(extensionCode: () -> Unit) {
        extensionCode()
        disconnect()
    }
}
```

BluetoothService.kt
```
class BluetoothService(
    val bluetoothClient: RxBleClient,
    val bluetoothScanSettings: ScanSettings,
    val mSchedulerProvider: SchedulerProvider,
    val bleSessionManager: BleSessionManager
) : IBluetoothService {

    private fun observeConnection(device: RxBleDevice): Observable<ConnectionState> {
        return device
            .observeConnectionStateChanges()
            .throttleLatest(500, TimeUnit.MILLISECONDS)
            .subscribeOn(mSchedulerProvider.io())
            .map { it.mapToConnectionState() }
    }

    override fun connectToDevice(macAddress: String): Observable<ConnectionState> {
        val device = bluetoothClient.getBleDevice(macAddress)
        bleSessionManager.setMacAddress(macAddress)
        bleSessionManager.setupConnection()
        return observeConnection(device)
    }

    override fun getDeviceShadow(): Single<HelixDeviceShadow> {

        return bleSessionManager.connection
            .subscribeOn(mSchedulerProvider.io())
            .firstOrError()
            .flatMap { it.readCharacteristic(MY_UUID) }
            .map {
                it.mapToMyResponse()
            }
            .doOnError { it.mapBluetoothGattOperationExceptions() }
    }

    override fun disconnect() {

        var disposable: Disposable? = null

        disposable = bleSessionManager.connection
            .subscribeOn(mSchedulerProvider.io())
            .firstOrError()
            .flatMap {
                it.writeCharacteristic(SLEEP_UUID, putHelixToSleep())
            }
            .ignoreElement()
            .onErrorComplete()
            .doFinally {

                disposable?.dispose()
                otaUpdateDisposables.clear()
                bleSessionManager.disconnect()
            }
            .subscribe()
    }
```

dariuszseweryn, 2019-03-11:
Thanks for sharing the code. Could you provide some more background on how it should be used? I could then create a Wiki page that could be easy to understand for other developers

davordev, 2019-03-26:
The whole idea behind this is to have a single source of truth for active bluetooth connection. So, in order to do anything with the ble device, you should call `bleSessionManager.connection` which will either return an active connection or throw an exception (preferably set by developer) notifying the user that the connection isn't established. In order to establish an connection, the following code should be called first: 
```bleSessionManager.apply {
    setMacAddress("MAC_ADDRESS_OF_DEVICE")
    setupConnection()
}```

Preferably, you can call `RxBleDevice.observeConnectionStateChanges()` to track the connection state.

If you want to terminate the connection, you can either: 
* call `bleSessionManager.disconnect()` to close down the connection instantly, or
* call `bleSessionManager.disconnectButFirst { [..code to execute..] }` to close down the connection after the code has been executed

dariuszseweryn, 2019-03-29:
Thank you! Added a new [wiki page](https://github.com/Polidea/RxAndroidBle/wiki/Tutorial:-Connection-Observable-sharing).

If anyone would like to add something, feel free to suggest changes or (better) create ready-to-add parts and post it here or in new issues. Closing now

closed 2019-03-29 by dariuszseweryn