dariuszseweryn/RxAndroidBle #542: Can't find any device during scan with ScanFilter.setServiceUuid()

hyyu, 2019-02-06:
Hi,

I'm working on an app that's linked to a BLE connected device, thus I'm using the RxAndroidBle library.

When I launch a scan, I want it to scan only for my devices. For this, I tried to use ScanFilter.setServiceUuid() and I've tried it with some service UUIDs belonging to the devices but the scan doesn't discover anything. However the scan detects, without this filter, the devices that should be detected.

I've already browsed the github issues, stackoverflow and searched on Google for an answer and didn't find anything that could help me. I didn't see anything that could help me on the readme.

Should I use other means?

Library version: 1.7.1

My code to launch the scan:

`		rxBleClient?.let {
			Log.i(TAG, "Launching Scan for Enovap devices")
			val scan = it.observeStateChanges()
				.subscribeOn(Schedulers.io())
				.observeOn(AndroidSchedulers.mainThread())
				.startWith(it.state).switchMap { state ->
					bleListener?.onBleStateChanged(state)
					when (state) {
						RxBleClient.State.READY -> {
							timeoutHandler.postDelayed(timeoutRunnable, timeout)
							it.scanBleDevices(
								ScanSettings.Builder()
									.setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY)
									.setCallbackType(ScanSettings.CALLBACK_TYPE_ALL_MATCHES)
									.build(),
								ScanFilter.Builder()
									.setServiceUuid(ParcelUuid.fromString(SERVICE_UUID)) // Scan only for Devices having an Enovap service UUID
								.build()
							)
						}
						else -> Observable.empty()
					}
				}.subscribeOn(Schedulers.io()).observeOn(AndroidSchedulers.mainThread())
			scanSubscriptionDisposable.add(
				scan.subscribe(
					{ scanResult ->
						scanResult.bleDevice?.let { device ->
							if (device.name != null) {
								Log.v(TAG, "device: $device | services: ${device.bluetoothDevice.uuids}")
								if (autoConnect) {
									stopScan()
									Thread.sleep(200)
								} else {
									bleListener?.onDeviceFound(device)
								}
							}
						}
					},
					{ t ->
						bleListener?.onScanError(t)
						stopScan()
					}
			)
			)
		}`

dariuszseweryn, 2019-02-06:
Make sure that the service UUID is available in the advertisement packet

hyyu, 2019-02-06:
How do I check that? Using RxBleDevice? If I'm not wrong it's the class that represents it no?

hyyu, 2019-02-06:
I've used the `filter` function from RxJava, and it just appears that the UUIDS that my devices are sending weren't the ones I was waiting for.

Sorry for the ticket, you can close it.