dariuszseweryn/RxAndroidBle #545: Possibility to disable Toast messages coming from RxAndroidBle

hyyu, 2019-02-12:
Hi,

Is there any mean to remove the toast when the connection is interrupted? I would like to have my own error handling, as I have my own styled components to fit my App Theme.

Version: 1.7.1

Behaviour:

When the connection is interrupted, a Toast appears to inform the connection couldn't be established
Screenshot:
[image]

dariuszseweryn, 2019-02-12:
I am afraid that what you are referencing has no connection to the library. You should ask the app developer.

Best Regards

closed 2019-02-12 by dariuszseweryn

jreehuis, 2019-02-12:
I already experienced the same issue. It is not connected to the library itself.
Some smartphone vendors include this toasts upon BT errors.
I think they can not be avoided without switching the phone/vendor.

hyyu, 2019-02-12:
Thanks for the answers, it seems that it's like jreehuis told, coming from the phone itself.

dariuszseweryn, 2019-02-12:
What phone / OS version you experience those toasts with? I have not seen them at all

hyyu, 2019-02-12:
I've had it on Google phones. My personnal phone is a Pixel 3 XL, and had it on Nexus phones.

jreehuis, 2019-02-12:
E.g. on a Huawei P10, Android 8 or some Samsung Phones but I can not recall which ones. I also have in mind to have them seen on Sony devices aswell.
It is already some time ago when I had a closer look. By mind the toast came up in the case that the phone has still a BT bonding but the device has lost or invalidated the bonding key.

dariuszseweryn, 2019-02-12:
Is the app you provided a screenshot from open sourced? If so could you add a link?

hyyu, 2019-02-12:
It's an app I'm currently developing, it is associated to a connected Ble device.