dariuszseweryn/RxAndroidBle #549: Rapid writes to different characteristics followed by quick disconnect causes BleDisconnectedException unhandled inside library

dziubekUK, 2019-02-26:
### Summary
During firmware update performed on peripheral (a lot of quick writes to certain characteristic) I also start rapidly writing to another characteristic via hardware button. After few seconds of doing so followed by quick disconnection (via on/off hardware button) app sometimes crashes (once in 4-5 times, possibly depends on how quickly the disconnection event occurs)

### Library version
`1.7.2`, `1.8.1`

#### Preconditions
One characteristic to which app writes rapidly and the other which value is controlled externally - in my case through hardware button.

#### Steps to reproduce actual result
<br />1. Start rapid writes to first characteristic.
<br />2. Start rapid writes to second characteristic.
<br />3. Quickly disconnect after few seconds of rapid value changing on both notifications.

#### Minimum code snippet reproducing the issue
Some method that would be helpful for you and to which I'm not quite sure:

    override fun startResponsesListener(connection: RxBleConnection): Single<Boolean> {
        val resultSubject = SingleSubject.create<Boolean>()
        setupNotificationDisposable = connection.setupNotification(PumpCharacteristicUuid.COMMAND.uuid)
                .subscribe(
                        {
                            commandNotificationObservable = it
                            resultSubject.onSuccess(true)
                        },
                        {
                            Timber.e("Error during notification setting: ${it.printStackTrace()}")
                        })
        return resultSubject
    }

    private inline fun RxBleConnection.responseFrom(request: RxBleConnection.() -> Single<ByteArray>): Single<ByteArray> {
        return commandNotificationObservable
                .firstOrNever()
                .zipWith(request(), BiFunction { response, _ -> response })
    }

```
fun <T> Observable<T>.firstOrNever(): Single<T> {
    return firstOrError()
            .onErrorResumeNext {
                // avoid propagation of NoSuchElementException when source stream is empty
                if (it is NoSuchElementException) Single.never<T>()
                else Single.error(it)
            }
}
```

Usage example:
```
        return connection
                .responseFrom {
                    writeCharacteristic(
                            PumpCharacteristicUuid.COMMAND.uuid,
                            JumpRequest.serialize())
                }
```

#### Logs from the application running with setting `RxBleLog.setLogLevel(RxBleLog.VERBOSE)`
```
   io.reactivex.exceptions.OnErrorNotImplementedException: The exception was not handled due to missing onError handler in the subscribe() method call. Further reading: https://github.com/ReactiveX/RxJava/wiki/Error-Handling | com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from C5:C1:C4:74:61:88 with status 0 (GATT_SUCCESS)
        at io.reactivex.internal.functions.Functions$OnErrorMissingConsumer.accept(Functions.java:704)
        at io.reactivex.internal.functions.Functions$OnErrorMissingConsumer.accept(Functions.java:701)
        at io.reactivex.internal.observers.ConsumerSingleObserver.onError(ConsumerSingleObserver.java:46)
        at io.reactivex.internal.operators.single.SingleFlatMap$SingleFlatMapCallback$FlatMapSingleObserver.onError(SingleFlatMap.java:116)
        at io.reactivex.internal.operators.observable.ObservableElementAtSingle$ElementAtObserver.onError(ObservableElementAtSingle.java:102)
        at io.reactivex.internal.operators.mixed.CompletableAndThenObservable$AndThenObservableObserver.onError(CompletableAndThenObservable.java:70)
        at io.reactivex.internal.operators.observable.ObservableCreate$CreateEmitter.tryOnError(ObservableCreate.java:84)
        at com.polidea.rxandroidble2.internal.serialization.FIFORunnableEntry$1.onError(FIFORunnableEntry.java:71)
        at io.reactivex.internal.operators.observable.ObservableUnsubscribeOn$UnsubscribeObserver.onError(ObservableUnsubscribeOn.java:70)
        at io.reactivex.internal.operators.observable.ObservableSubscribeOn$SubscribeOnObserver.onError(ObservableSubscribeOn.java:63)
        at io.reactivex.internal.operators.observable.ObservableCreate$CreateEmitter.tryOnError(ObservableCreate.java:84)
        at com.polidea.rxandroidble2.internal.util.QueueReleasingEmitterWrapper.onError(QueueReleasingEmitterWrapper.java:45)
        at io.reactivex.internal.observers.DeferredScalarDisposable.error(DeferredScalarDisposable.java:100)
        at io.reactivex.internal.operators.single.SingleToObservable$SingleToObservableObserver.onError(SingleToObservable.java:77)
        at io.reactivex.internal.operators.single.SingleTimeout$TimeoutMainObserver.onError(SingleTimeout.java:150)
        at io.reactivex.internal.operators.single.SingleMap$MapSingleObserver.onError(SingleMap.java:69)
        at io.reactivex.internal.operators.observable.ObservableElementAtSingle$ElementAtObserver.onError(ObservableElementAtSingle.java:102)
        at io.reactivex.internal.observers.BasicFuseableObserver.onError(BasicFuseableObserver.java:100)
        at io.reactivex.internal.operators.observable.ObservableObserveOn$ObserveOnObserver.checkTerminated(ObservableObserveOn.java:281)
        at io.reactivex.internal.operators.observable.ObservableObserveOn$ObserveOnObserver.drainNormal(ObservableObserveOn.java:172)
        at io.reactivex.internal.operators.observable.ObservableObserveOn$ObserveOnObserver.run(ObservableObserveOn.java:255)
        at io.reactivex.internal.schedulers.ExecutorScheduler$ExecutorWorker$BooleanRunnable.run(ExecutorScheduler.java:288)
        at io.reactivex.internal.schedulers.ExecutorScheduler$ExecutorWorker.run(ExecutorScheduler.java:253)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:764)
     Caused by: com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from C5:C1:C4:74:61:88 with status 0 (GATT_SUCCESS)
        at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:77)
        at android.bluetooth.BluetoothGatt$1$4.run(BluetoothGatt.java:262)
        at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:770)
        at android.bluetooth.BluetoothGatt.access$200(BluetoothGatt.java:39)
        at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:257)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:71)
2019-02-26 09:35:47.982 10588-10663/com.chiaro.elviepump.debug E/AndroidRuntime:     at android.os.Binder.execTransact(Binder.java:731)
```
No actual reference to my code ^

#### Actual result
Crash because of OnErrorNoImplementedException caused by BleDisconnectedException

#### Expected result
No crash

dariuszseweryn, 2019-02-26:
Hello,

Thanks for the report

You could add [this library](https://github.com/akaita/RxJava2Debug) to have a better stacktrace. Just because there is no code referenced in the plain `RxJava2` stacktrace does not yet mean that the error is in the library code.

Additionally the code provided is not purely functional/stateless and therefore is hard to debug. Having a code that uses a single `.subscribe()` would be easier to understand.

Best Regards

dziubekUK, 2019-02-26:
Great tip with the lib, thanks! The outcome after integrating RxJava2Debug still does not point to my code, but shows a bit more info:
```
2019-02-26 11:46:04.169 14490-14611/com.chiaro.elviepump.debug E/AndroidRuntime: FATAL EXCEPTION: pool-8-thread-1
    Process: com.chiaro.elviepump.debug, PID: 14490
    java.lang.Throwable: Disconnected from C5:C1:C4:74:61:88 with status 0 (GATT_SUCCESS)
        at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:77)
        at android.bluetooth.BluetoothGatt$1$4.run(BluetoothGatt.java:262)
        at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:770)
        at android.bluetooth.BluetoothGatt.access$200(BluetoothGatt.java:39)
        at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:257)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:71)
        at android.os.Binder.execTransact(Binder.java:731)
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable
     Caused by: java.lang.Throwable: caused by com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from C5:C1:C4:74:61:88 with status 0 (GATT_SUCCESS)
```

dariuszseweryn, 2019-02-26:
There is nothing unusual at this point. You can add more logs with `RxBleClient.setLogLevel(RxBleLog.VERBOSE)` from before the actual stacktrace. I have a hunch that your usage of subjects may cause problems

dziubekUK, 2019-02-26:
Still no useful info in stacktrace. How should I go about setting up on notification? I provided the snippets earlier. I know it's not optimal with subject, but it seems to be the only way. I want to setup notification once and pass info to caller when the notification is setup

dariuszseweryn, 2019-02-26:
Please add the full logs anyway — it may indicate whether the issue is on the library side. As for the `RxJava` state management there is [a good talk to watch](https://www.youtube.com/watch?v=0IKHxjkgop4) which I constantly recommend

dziubekUK, 2019-02-26:
Ok, below I'm pasting entire info I got with `RxBleClient.setLogLevel(RxBleLog.VERBOSE)` and `RxJava2Debug` integrated.
```
2019-02-26 14:10:05.651 22298-22389/com.chiaro.elviepump.debug D/RxBle#ConnectionOperationQueue: QUEUED   CharacteristicWriteOperation(50389996)
2019-02-26 14:10:05.655 1530-2214/? I/bt_btm_sec: btm_sec_disconnected clearing pending flag handle:9 reason:0
2019-02-26 14:10:05.656 2195-2208/? D/BluetoothGattServer: onServerConnectionState() - status=0 serverIf=5 device=C5:C1:C4:74:61:88
2019-02-26 14:10:05.657 1530-1603/? I/btif_av: btif_av_acl_disconnected: Peer c5:c1:c4:74:61:88 : ACL Disconnected
2019-02-26 14:10:05.657 22298-22380/com.chiaro.elviepump.debug D/RxBle#BluetoothGatt$1: onCharacteristicRead characteristic=ea0b2b14-aafc-2890-e611-a6f3de55935b status=133
2019-02-26 14:10:05.657 1530-2214/? I/btif_av: BtifAvPeer *BtifAvSource::FindOrCreatePeer(const RawAddress &, tBTA_AV_HNDL): Create peer: peer_address=c5:c1:c4:74:61:88 bta_handle=0x41 peer_id=0
2019-02-26 14:10:05.657 1530-2214/? W/bt_btif: btif_av_get_peer_sep: No active peer found
2019-02-26 14:10:05.657 1530-2214/? I/bt_btif_a2dp: btif_a2dp_on_idle: ## ON A2DP IDLE ## peer_sep = 1
2019-02-26 14:10:05.657 1530-2214/? W/bt_btif: btif_av_get_peer_sep: No active peer found
2019-02-26 14:10:05.657 1530-2214/? I/bt_btif_a2dp_source: btif_a2dp_source_on_idle: state=STATE_OFF
2019-02-26 14:10:05.658 1530-2214/? W/bt_btif: btif_av_get_peer_sep: No active peer found
2019-02-26 14:10:05.658 1530-2214/? I/bt_btif_a2dp: btif_a2dp_on_idle: ## ON A2DP IDLE ## peer_sep = 1
2019-02-26 14:10:05.658 1530-2214/? W/bt_btif: btif_av_get_peer_sep: No active peer found
2019-02-26 14:10:05.658 1530-2214/? I/bt_btif_a2dp_source: btif_a2dp_source_on_idle: state=STATE_OFF
2019-02-26 14:10:05.658 1530-2214/? I/btif_av: DeleteIdlePeers: Deleting idle peer: c5:c1:c4:74:61:88 bta_handle=0x41
2019-02-26 14:10:05.659 22298-22380/com.chiaro.elviepump.debug D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=C5:C1:C4:74:61:88
2019-02-26 14:10:05.659 22298-22389/com.chiaro.elviepump.debug D/RxBle#ConnectionOperationQueue: QUEUED   CharacteristicWriteOperation(195922506)
2019-02-26 14:10:05.659 22298-22380/com.chiaro.elviepump.debug D/RxBle#BluetoothGatt$1: onConnectionStateChange newState=0 status=0
2019-02-26 14:10:05.664 22298-22380/com.chiaro.elviepump.debug I/RxBle#ConnectionOperationQueue: Connection operations queue to be terminated (C5:C1:C4:74:61:88)
2019-02-26 14:10:05.666 1530-1530/? D/BluetoothMapService: onReceive
2019-02-26 14:10:05.666 1530-1530/? D/BluetoothMapService: onReceive: android.bluetooth.device.action.ACL_DISCONNECTED
2019-02-26 14:10:05.667 22298-22425/com.chiaro.elviepump.debug W/RxBle#ConnectionOperationQueue: Queue's awaitRelease() has been interrupted abruptly while it wasn't released by the release() method.
    java.lang.InterruptedException
        at java.lang.Object.wait(Native Method)
        at com.polidea.rxandroidble2.internal.serialization.QueueSemaphore.awaitRelease(QueueSemaphore.java:15)
        at com.polidea.rxandroidble2.internal.serialization.ConnectionOperationQueueImpl$1.run(ConnectionOperationQueueImpl.java:71)
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:458)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:764)
2019-02-26 14:10:05.667 22298-22389/com.chiaro.elviepump.debug D/RxBle#ConnectionOperationQueue: QUEUED   CharacteristicWriteOperation(178693411)
2019-02-26 14:10:05.673 22298-22389/com.chiaro.elviepump.debug W/System.err: io.reactivex.exceptions.OnErrorNotImplementedException: The exception was not handled due to missing onError handler in the subscribe() method call. Further reading: https://github.com/ReactiveX/RxJava/wiki/Error-Handling | com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from C5:C1:C4:74:61:88 with status 0 (GATT_SUCCESS)
2019-02-26 14:10:05.674 22298-22389/com.chiaro.elviepump.debug W/System.err:     at io.reactivex.internal.functions.Functions$OnErrorMissingConsumer.accept(Functions.java:704)
2019-02-26 14:10:05.674 22298-22389/com.chiaro.elviepump.debug W/System.err:     at io.reactivex.internal.functions.Functions$OnErrorMissingConsumer.accept(Functions.java:701)
2019-02-26 14:10:05.674 22298-22389/com.chiaro.elviepump.debug W/System.err:     at io.reactivex.internal.observers.ConsumerSingleObserver.onError(ConsumerSingleObserver.java:46)
2019-02-26 14:10:05.674 22298-22389/com.chiaro.elviepump.debug W/System.err:     at com.akaita.java.rxjava2debug.extensions.SingleOnAssembly$OnAssemblySingleObserver.onError(SingleOnAssembly.java:73)
2019-02-26 14:10:05.674 22298-22389/com.chiaro.elviepump.debug W/System.err:     at io.reactivex.internal.operators.single.SingleFlatMap$SingleFlatMapCallback$FlatMapSingleObserver.onError(SingleFlatMap.java:116)
2019-02-26 14:10:05.674 22298-22389/com.chiaro.elviepump.debug W/System.err:     at com.akaita.java.rxjava2debug.extensions.SingleOnAssembly$OnAssemblySingleObserver.onError(SingleOnAssembly.java:73)
2019-02-26 14:10:05.674 22298-22389/com.chiaro.elviepump.debug W/System.err:     at io.reactivex.internal.operators.observable.ObservableElementAtSingle$ElementAtObserver.onError(ObservableElementAtSingle.java: [... cut ...]
2019-02-26 14:10:05.814 22298-22380/com.chiaro.elviepump.debug W/System.err: 	[CIRCULAR REFERENCE:com.akaita.java.rxjava2debug.extensions.RxJavaAssemblyException]
2019-02-26 14:10:05.814 22298-22380/com.chiaro.elviepump.debug E/FirmwareRequestsServiceImpl$startResponsesListener: Error during notification setting: kotlin.Unit
2019-02-26 14:10:05.815 22298-22360/com.chiaro.elviepump.debug D/BluetoothGatt: close()
2019-02-26 14:10:05.815 22298-22360/com.chiaro.elviepump.debug D/BluetoothGatt: unregisterApp() - mClientIf=7
```

dariuszseweryn, 2019-02-26:
Still the library seems to remain functional as you can see in logs from this below:
```
2019-02-26 14:10:05.806 22298-22380/com.myapp.debug D/RxBle#ClientOperationQueue: QUEUED   DisconnectOperation(143355081)
```

Try rewriting the code to get rid of the subject or always have it subscribed with just a log message and see what will happen

dziubekUK, 2019-02-27:
I'll try that rewrite and post more info, but I had checked the subject you're talking about before and no error was propagated on it

dziubekUK, 2019-02-27:
No more intel on this after suggested rewrite. Crash is still happening

dariuszseweryn, 2019-02-27:
The error comes from `FIFORunnableEntry` which is directly routed to all `RxBleConnection` functions. Can you confirm that in the code you use _every_ `.subscribe()` has an on error part?

If yes — I will need a minimal application that would exhibit the problem which I could build locally

dziubekUK, 2019-02-28:
I double checked - all `.subscribe()` methods in the entire project have an on error part. I'll check the feasibility of creating such minimal application and get back to you

dziubekUK, 2019-02-28:
I've turned off one of the service using bluetooth inside app and replicated crash scenario. App crashed, but more logs appeared pointing to my code. Closing the issue

closed 2019-02-28 by dziubekUK

dariuszseweryn, 2019-02-28:
Glad to hear that you have found it. Feel free to give more details as it may be easier for me to spot those situations in the future