dariuszseweryn/RxAndroidBle #552: NullPointerException is thrown sometimes

offwhiite, 2019-03-04:
### Summary
java.lang.NullPointerException is thrown sometimes 

I'm using this library for application development.
However, I occasionally observe the following CrashLog.

```
    java.lang.NullPointerException: Attempt to invoke interface method 'void io.reactivex.ObservableEmitter.onNext(java.lang.Object)' on a null object reference
        at com.polidea.rxandroidble2.internal.connection.DisconnectionRouter.notifySubscribersAboutException(DisconnectionRouter.java:123)
        at com.polidea.rxandroidble2.internal.connection.DisconnectionRouter.access$100(DisconnectionRouter.java:30)
        at com.polidea.rxandroidble2.internal.connection.DisconnectionRouter$1.accept(DisconnectionRouter.java:65)
        at com.polidea.rxandroidble2.internal.connection.DisconnectionRouter$1.accept(DisconnectionRouter.java:60)
        at io.reactivex.internal.operators.maybe.MaybeCallbackObserver.onSuccess(MaybeCallbackObserver.java:71)
        at io.reactivex.internal.operators.observable.ObservableElementAtMaybe$ElementAtObserver.onNext(ObservableElementAtMaybe.java:82)
        at io.reactivex.internal.operators.observable.ObservableMap$MapObserver.onNext(ObservableMap.java:62)
        at io.reactivex.internal.operators.observable.ObservableFilter$FilterObserver.onNext(ObservableFilter.java:52)
        at io.reactivex.internal.operators.observable.ObservableConcatMap$ConcatMapDelayErrorObserver$DelayErrorInnerObserver.onNext(ObservableConcatMap.java:506)
        at io.reactivex.internal.operators.observable.ObservableMap$MapObserver.onNext(ObservableMap.java:62)
        at com.polidea.rxandroidble2.RxBleAdapterStateObservable$1.onReceive(RxBleAdapterStateObservable.java:62)
        at android.app.LoadedApk$ReceiverDispatcher$Args.run(LoadedApk.java:1126)
        at android.os.Handler.handleCallback(Handler.java:751)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:159)
        at android.app.ActivityThread.main(ActivityThread.java:6097)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:865)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:755)
```

This crash seems to be occurring in the following places.

``` 
    private void notifySubscribersAboutException() {
        if (adapterMonitoringDisposable != null) {
            adapterMonitoringDisposable.dispose();
        }

        while (!exceptionEmitters.isEmpty()) {
            final ObservableEmitter<BleException> exceptionEmitter = exceptionEmitters.poll();
            exceptionEmitter.onNext(exceptionOccurred); // crash 
            exceptionEmitter.onComplete();
        }
    }
```

I think that notifySubscribersAboutException () is called multi thread.
However, since the exceptionEmitters queue is thread safe, the processing is executed one by one.

If ThreadA, ThreadB calls this method at the same time, I think through "!ExceptionEmitters.isEmpty ()"  and executing exceptionEmitters.poll () and getting null.

So I want to fix it as follows.

```
　  // add synchronized
    private synchronized void notifySubscribersAboutException() {
        if (adapterMonitoringDisposable != null) {
            adapterMonitoringDisposable.dispose();
        }

        while (!exceptionEmitters.isEmpty()) {
            final ObservableEmitter<BleException> exceptionEmitter = exceptionEmitters.poll();
            exceptionEmitter.onNext(exceptionOccurred); // crash 
            exceptionEmitter.onComplete();
        }
    }
``` 

other

```
　  // add synchronized
    private synchronized void notifySubscribersAboutException() {
        if (adapterMonitoringDisposable != null) {
            adapterMonitoringDisposable.dispose();
        }

        while (!exceptionEmitters.isEmpty()) {
            final ObservableEmitter<BleException> exceptionEmitter = exceptionEmitters.poll();
            if (exceptionEmitter != null) { // null check
                exceptionEmitter.onNext(exceptionOccurred); // crash 
            }
            exceptionEmitter.onComplete();

        }
    }
``` 

Do you think this is correct?

### Library version
`2.1.0`

#### Steps to reproduce actual result
unknown (get log from crashlytics)

dariuszseweryn, 2019-03-11:
Hey! Thanks for the report. I have pushed a branch that should fix the issue. I have removed the manual handling of subscribers — it should all be handled now by the RxJava implementation. Feel free to look on the PR before I merge to double check if it looks correct.

closed 2019-03-15 by dariuszseweryn

dariuszseweryn, 2019-03-15:
I have just released `1.8.2` which should address this problem. Feel free to integrate it and report back if it fixed the issue.

PR #553 (2019-03-15): Fixed NPE in DisconnectionRouter
Due to a race condition it was possible that when iterating through disconnection error subscribers one of them has unsubscribed resulting in a null reference. Now all of it should be handled by RxJava operator implementations.

Fixes #552