dariuszseweryn/RxAndroidBle #554: LocationSettingsRequest says Ble is allowed, but we still can't scan

ncapdevi, 2019-03-19:
### Summary
For API > 23, with location services turned off, we are required to turn them on in order to continue. However,  following this guide https://developers.google.com/android/reference/com/google/android/gms/location/SettingsClient  and setting the builder to ` builder.setNeedBle(true);` I am getting results back that ble scan is allowed, however the library prevents it.  Is it possible that the library should be using the SettingsClient to determine if locationServices are required to be turned on?

### Library version
`1.8.2`

dariuszseweryn, 2019-03-20:
Hello,

What is the exact error you get? I expect it is `LOCATION_SERVICES_DISABLED`?

From the docs of `LocationSettingsRequest`:

> public LocationSettingsRequest.Builder setNeedBle (boolean needBle)
> 
> Sets whether the client wants BLE scan to be enabled. When this flag is set to true, if the platform supports BLE scan mode and Bluetooth is off, the dialog will prompt the user to enable BLE scan. If the platform doesn't support BLE scan mode, the dialog will prompt to enable Bluetooth.

It seems that it only handles turning the Bluetooth ON/OFF, not the Location Services?

ncapdevi, 2019-03-20:
Correct, for `LOCATION_SERVICES_DISABLED`

> It seems that it only handles turning the Bluetooth ON/OFF, not the Location Services?

You're correct.

After digging in a bit more, LocationSettingsRequest will return successfully if you:

- Don't have a specific `locationRequest` added
or 
- Have a location request added with `PRIORITY_NO_POWER` as the priority setting.

However, if you add a location request with `PRIORITY_LOW_POWER` you are able to confirm that location settings are off, and follow the provided documentation to request that the user turns it on. (It provides a system dialog).  I think this issue can be considered closed, but might be helpful for people who want address when `LOCATION_SERVICES_DISABLED` is the Client Status

closed 2019-03-20 by ncapdevi