dariuszseweryn/RxAndroidBle #559: BleCharacteristicNotFoundException: Characteristic not found with UUID

SeloSlav, 2019-03-28:
I am unable to retrieve data from my device despite supplying what I believe to be the correct UUID: ``D/ERROR:: com.polidea.rxandroidble2.exceptions.BleCharacteristicNotFoundException: Characteristic not found with UUID 00001101-0000-1000-8000-00805f9b34fb``. What should I do instead?

```
// Get device from scan
rxBleDevice = scanResult.bleDevice

val charUUID = UUID.fromString(scanResult.bleDevice.bluetoothDevice.uuids[0].toString())

/* Establish connection to device */
rxBleDevice!!.establishConnection(false) ?
 .doOnNext {
  _ ->
   Log.d("Device: ", "Connection Established")
 } ?
 .flatMapSingle {
  rxBleConnection ->
   charUUID ? .let {
    rxBleConnection.readCharacteristic(it)
   }
 } ? .subscribe({
  count ->
  // count should be in bytes
  println("SUCCESS: $count")

 }, {
  throwable ->
  Log.d("SHOT COUNT ERROR: ", "$throwable")
 })
```

My expected output is a byte array that reads "datc00099", for example.

jreehuis, 2019-03-28:
You are trying to read from a characteristic but giving a service UUID. As usually the characteristic and service UUIDs are different so I guess you have to find the right characteristic UUID.
Which UUID you have to use depends on the BT Device and Firmware Implementation you are using.

SeloSlav, 2019-03-28:
Which one is the service UUID?

This one?

val deviceUUID = UUID.fromString(scanResult.bleDevice.bluetoothDevice.uuids[0].toString())

Or this one?

scanResult.scanRecord.serviceUuids, which prints to null.

Both?

dariuszseweryn, 2019-03-28:
Closing as there is [a related general usage question on Stackoverflow](https://stackoverflow.com/questions/55392077)

closed 2019-03-28 by dariuszseweryn