dariuszseweryn/RxAndroidBle #561: BleScanException: Scan failed because feature unsupported on api 23+ due to CALLBACK_TYPE_FIRST_MATCH | CALLBACK_TYPE_MATCH_LOST

EFCniux, 2019-04-12:
### Summary
Starting a scan with `CALLBACK_TYPE_FIRST_MATCH | CALLBACK_TYPE_MATCH_LOST` errors with `SCAN_FAILED_FEATURE_UNSUPPORTED` or `SCAN_FAILED_OUT_OF_HARDWARE_RESOURCES`

Testing on API < 21 works without problems, and while looking at the code in `ScanSetupBuilderImplApi18` it is working because the behavior is being emulated.

`ScanSetupBuilderImplApi21` also emulates said behavior, although I cannot test it.

`ScanSetupBuilderImplApi23` does NOT emulate callbacks, on the basis that they are natively available.

Thing is all my available Android API 23+ test devices are old or just plain cheap and their Bluetooth chips likely do not support said behavior, which would make sense given the thrown errors.

I think `ScanSetupBuilderImplApi23` should retry on error with the emulated behavior, or at least give access to it outside of the library, so it can be implemented without having to copy the same functionality.

Thoughts?

### Library version
`1.9.1`

#### Preconditions
```
ScanSettings.Builder()
                .setCallbackType(ScanSettings.CALLBACK_TYPE_FIRST_MATCH or ScanSettings.CALLBACK_TYPE_MATCH_LOST)
                .build()
```

#### Steps to reproduce actual result

#### Minimum code snippet reproducing the issue
```
client
        .scanBleDevices(
            ScanSettings.Builder()
                .setCallbackType(ScanSettings.CALLBACK_TYPE_FIRST_MATCH or ScanSettings.CALLBACK_TYPE_MATCH_LOST)
                .setScanMode(ScanSettings.SCAN_MODE_BALANCED)
                .build(),
            *filters
        )
        .subscribe()
```
#### Logs from the application running with setting 
```
2019-04-12 13:05:19.540 10603-10636/[..] D/RxBle#ClientOperationQueue: QUEUED   ScanOperationApi21(158477343)
2019-04-12 13:05:19.541 10603-10627/[..] D/RxBle#ClientOperationQueue: STARTED  ScanOperationApi21(158477343)
2019-04-12 13:05:19.542 10603-10627/[..] I/RxBle#ClientOperationQueue: RUNNING  ScanOperationApi21{ANY_MUST_MATCH -> nativeFilters=[BluetoothLeScanFilter [mDeviceName=null, MAC=null, mUuid=null, mUuidMask=null, mServiceDataUuid=null, mServiceData=null, mServiceDataMask=null, mManufacturerId=-1, mManufacturerData=null, mManufacturerDataMask=null]]}
2019-04-12 13:05:19.549 10603-10637/[..] I/RxBle#QueueOperation: Scan operation is requested to start.
2019-04-12 13:05:19.549 10603-10637/[..] D/RxBle#ScanOperationApi21: No library side filtering —> debug logs of scanned devices disabled
2019-04-12 13:05:19.557 10603-10627/[..] D/RxBle#ClientOperationQueue: FINISHED ScanOperationApi21(158477343) in 16 ms
2019-04-12 13:05:19.598 10603-10637/[..] I/RxBle#CancellableDisposable: Scan operation is requested to stop.
2019-04-12 13:05:19.601 10603-10637/[..] D/BluetoothLeScanner: could not find callback wrapper
2019-04-12 13:05:19.601 10603-10603/[..] W/System.err: io.reactivex.exceptions.OnErrorNotImplementedException: The exception was not handled due to missing onError handler in the subscribe() method call. Further reading: https://github.com/ReactiveX/RxJava/wiki/Error-Handling | com.polidea.rxandroidble2.exceptions.BleScanException: Scan failed because feature unsupported (code 8)
2019-04-12 13:05:19.602 10603-10603/[..] W/System.err:     at io.reactivex.internal.functions.Functions$OnErrorMissingConsumer.accept(Functions.java:704)
2019-04-12 13:05:19.602 10603-10603/[..] W/System.err:     at io.reactivex.internal.functions.Functions$OnErrorMissingConsumer.accept(Functions.java:701)
2019-04-12 13:05:19.602 10603-10603/[..] W/System.err:     at io.reactivex.internal.observers.LambdaObserver.onError(LambdaObserver.java:77)
2019-04-12 13:05:19.602 10603-10603/[..] W/System.err:     at io.reactivex.internal.operators.observable.ObservableObserveOn$ObserveOnObserver.checkTerminated(ObservableObserveOn.java:281)
2019-04-12 13:05:19.603 10603-10603/[..] W/System.err:     at io.reactivex.internal.operators.observable.ObservableObserveOn$ObserveOnObserver.drainNormal(ObservableObserveOn.java:172)
2019-04-12 13:05:19.603 10603-10603/[..] W/System.err:     at io.reactivex.internal.operators.observable.ObservableObserveOn$ObserveOnObserver.run(ObservableObserveOn.java:255)
2019-04-12 13:05:19.603 10603-10603/[..] W/System.err:     at io.reactivex.android.schedulers.HandlerScheduler$ScheduledRunnable.run(HandlerScheduler.java:124)
2019-04-12 13:05:19.603 10603-10603/[..] W/System.err:     at android.os.Handler.handleCallback(Handler.java:751)
2019-04-12 13:05:19.603 10603-10603/[..] W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:95)
2019-04-12 13:05:19.603 10603-10603/[..] W/System.err:     at android.os.Looper.loop(Looper.java:154)
2019-04-12 13:05:19.603 10603-10603/[..] W/System.err:     at android.app.ActivityThread.main(ActivityThread.java:6126)
2019-04-12 13:05:19.604 10603-10603/[..] W/System.err:     at java.lang.reflect.Method.invoke(Native Method)
2019-04-12 13:05:19.604 10603-10603/[..] W/System.err:     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:886)
2019-04-12 13:05:19.604 10603-10603/[..] W/System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:776)
2019-04-12 13:05:19.604 10603-10603/[..] W/System.err: Caused by: com.polidea.rxandroidble2.exceptions.BleScanException: Scan failed because feature unsupported (code 8)
2019-04-12 13:05:19.605 10603-10603/[..] W/System.err:     at com.polidea.rxandroidble2.internal.operations.ScanOperationApi21$1.onScanFailed(ScanOperationApi21.java:92)
2019-04-12 13:05:19.605 10603-10603/[..] W/System.err:     at android.bluetooth.le.BluetoothLeScanner$1.run(BluetoothLeScanner.java:461)
2019-04-12 13:05:19.605 10603-10603/[..] W/System.err: 	... 7 more
```
#### Actual result
`com.polidea.rxandroidble2.exceptions.BleScanException: Scan failed because feature unsupported (code 8)`

#### Expected result
Library emulates behavior (on error) in API 23 just like pre API 23

EFCniux, 2019-04-16:
I have implemented my [own version of the behavior](https://gist.github.com/EFCniux/9ee9c5f05c667a051aa6fb1438e4e17b) for now, and it seems to work without any problems, although this approach makes it easier to trigger `BleScanException.UNDOCUMENTED_SCAN_THROTTLE` due to the initial attempt also counting for the 5 attempt limit.

dariuszseweryn, 2019-04-16:
As far as I am concerned the OS will count every try to run a scan, whether it was successful or not

dariuszseweryn, 2019-09-17:
Long time no see. I finally have some time to revisit this issue.

Initially I planned to give more knobs for the user to decide if `ScanSettings` or `ScanFilters` should get emulated.

Now I looked again and it seems that the issue here is pretty simple — the native interface is requested to match first and the last appearance of peripherals. Unfortunately `ScanFilter`s are not specified (i.e. they are empty and match all advertisements) and the hardware (offloaded filters) will likely overwhelm the BT chip — this is why `SCAN_FAILED_FEATURE_UNSUPPORTED` or `SCAN_FAILED_OUT_OF_HARDWARE_RESOURCES ` is the result. If that is the case I could simply revert to emulated matching if `ScanFilter`s are empty.

I know it has been a while but could I ask you to test my hypothesis? Run the same code on the same devices but while using non-empty `ScanFilter`?

EFCniux, 2019-09-20:
Hello!
Sadly, I no longer have access to the devices I was using.

In case it helps, I do not remember trying to use filters while I was looking into this, and your reasoning seems solid.

dariuszseweryn, 2019-09-20:
Ok, so I consider this case closed. The fix should be available in `1.11.0-SNAPSHOT` in several minutes and will be available once the stable version is released as well

PR #612 (2019-09-20): Workaround scan failure for non-default callback type and empty filters
Android API may be requested to match first or the last appearance of peripherals. When `ScanFilter`s are not specified or they are empty (match all advertisements) — the hardware/offloaded filtering will likely overwhelm the BT chip — this is why scan is not started with `SCAN_FAILED_FEATURE_UNSUPPORTED` or `SCAN_FAILED_OUT_OF_HARDWARE_RESOURCES` error in the result.

This commit adds a workaround by falling back to emulated callback type in the described scenario — similarly as it is working on API 21 and API 18.

Closes [... cut]
closed 2019-09-20 by dariuszseweryn