dariuszseweryn/RxAndroidBle #564: Sample app broken by change in rxbinding library

RobLewis, 2019-04-25:
It appears that a change in Jake Wharton's `rxbinding` library has caused a compile error in the sample app. The class `AdvancedCharacteristicOperationExampleActivity.java` has a failed import: 
`import com.jakewharton.rxbinding3.view.RxView;` 

I'm not familiar enough with the code to identify the exact cause of the error, but the method `activatedClicksObservable` can't resolve `RxView`.

dariuszseweryn, 2019-04-25:
Hmmm... I just built and ran the app without issues. Are you sure that a `Gradle Sync` (File => Sync Project with Gradle Files) was run after you have pulled the changes?

RobLewis, 2019-04-25:
Didn't know that it was necessary to explicitly sync, as it seems to me that it normally happens automatically? Anyway, that seems to have fixed it.

closed 2019-04-25 by RobLewis

dariuszseweryn, 2019-04-26:
From my knowledge it should happen automatically. Good that it is solved.