dariuszseweryn/RxAndroidBle #566: Using existing paired connection at the OS level

scotzark, 2019-04-29:
When I have an existing connection that the user has paired and bonded at the OS level I am observing that the stack is initiating an entirely new connection to the target device, instead of using the existing connection that the user has created.  2 questions:

1.) Is this working as designed?
2.) If so do I have to tear down the existing connection before establishing a new connection?

I have looked through the documentation and Stack Overflow and I think that in general this shouldn't be an issue for most equipment, but the particular piece of hardware we are working with is not liking the fact that the existing paired connection is not being reused.

thanks,

Scot

dariuszseweryn, 2019-04-29:
> Is this working as designed?

I can only guess what is the exact situation just with your description and without logs. I think the answer is `Yes, it is working as designed`.

> If so do I have to tear down the existing connection before establishing a new connection?

There is a difference between a new `BluetoothGatt` being created and a physical connection. Multiple `BluetoothGatt` objects will share the same connection. In this situation — no, you do not have to tear down the existing connection.

scotzark, 2019-04-29:
Ok the issue really seems to be related to Android 9.0.  We are having good success with older versions of Android, but on Pie the device is disconnecting and reconnecting in a loop.  Yet this same code and device with an Android 8.0, 7.0, 6.0 device works perfectly.  Are there any known issues w/ the stack and Android Pie?

dariuszseweryn, 2019-04-29:
Not yet — could you elaborate more on what you are experiencing? Example code, logs — it could be useful.

scotzark, 2019-04-29:
Here is the log information.  As I mentioned the code works fine on Android 5.0 - 8.0.  But for some reason on Pie the scan operation never seems to complete.

2019-04-29 13:22:42.909 10672-10672/com.harley_davidson.ride_planner_blue D/RxBle#ClientOperationQueue: QUEUED   ScanOperationApi21(239251550)
2019-04-29 13:22:42.911 10672-10817/com.harley_davidson.ride_planner_blue D/RxBle#ClientOperationQueue: STARTED  ScanOperationApi21(239251550)
2019-04-29 13:22:42.913 10672-10819/com.harley_davidson.ride_planner_blue I/RxBle#QueueOperation: Scan operation is requested to start.
2019-04-29 13:22:42.922 10672-10817/com.harley_davidson.ride_planner_blue D/RxBle#ClientOperationQueue: FINISHED ScanOperationApi21(239251550) in 11 ms
2019-04-29 13:22:48.643 10672-10819/com.harley_davidson.ride_planner_blue I/RxBle#CancellableDisposable: Scan operation is requested to stop.
2019-04-29 13:22:50.733 10672-10989/com.harley_davidson.ride_planner_blue D/RxBle#ClientOperationQueue: QUEUED   ConnectOperation(214580364)
2019-04-29 13:22:50.737 10672-10817/com.harley_davidson.ride_planner_blue D/RxBle#ClientOperationQueue: STARTED  ConnectOperation(214580364)
2019-04-29 13:22:57.527 10672-10690/com.harley_davidson.ride_planner_blue D/RxBle#BluetoothGatt$1: onConnectionStateChange newState=2 status=0
2019-04-29 13:22:57.578 10672-10989/com.harley_davidson.ride_planner_blue D/RxBle#ConnectionOperationQueue: QUEUED   ServiceDiscoveryOperation(56280050)
2019-04-29 13:22:57.583 10672-10988/com.harley_davidson.ride_planner_blue D/RxBle#ConnectionOperationQueue: STARTED  ServiceDiscoveryOperation(56280050)
2019-04-29 13:22:57.584 10672-10817/com.harley_davidson.ride_planner_blue D/RxBle#ClientOperationQueue: FINISHED ConnectOperation(214580364) in 6849 ms
2019-04-29 13:23:17.603 10672-10989/com.harley_davidson.ride_planner_blue D/RxBle#ClientOperationQueue: QUEUED   DisconnectOperation(40269473)
2019-04-29 13:23:17.606 10672-10817/com.harley_davidson.ride_planner_blue D/RxBle#ClientOperationQueue: STARTED  DisconnectOperation(40269473)
2019-04-29 13:23:17.608 10672-10989/com.harley_davidson.ride_planner_blue I/RxBle#ConnectionOperationQueue: Connection operations queue to be terminated (00:16:A4:4A:26:9E)
2019-04-29 13:23:17.609 10672-10988/com.harley_davidson.ride_planner_blue D/RxBle#ConnectionOperationQueue: FINISHED ServiceDiscoveryOperation(56280050) in 20017 ms
2019-04-29 13:23:17.612 10672-10988/com.harley_davidson.ride_planner_blue D/RxBle#Executors$RunnableAdapter: Terminated.
2019-04-29 13:23:17.647 10672-10989/com.harley_davidson.ride_planner_blue D/RxBle#ClientOperationQueue: QUEUED   ConnectOperation(31134079)
2019-04-29 13:23:17.654 10672-10828/com.harley_davidson.ride_planner_blue D/RxBle#BluetoothGatt$1: onConnectionStateChange newState=0 status=0
2019-04-29 13:23:17.660 10672-10817/com.harley_davidson.ride_planner_blue D/RxBle#ClientOperationQueue: FINISHED DisconnectOperation(40269473) in 55 ms
2019-04-29 13:23:17.660 10672-10817/com.harley_davidson.ride_planner_blue D/RxBle#ClientOperationQueue: STARTED  ConnectOperation(31134079)
2019-04-29 13:23:17.687 10672-10828/com.harley_davidson.ride_planner_blue D/RxBle#BluetoothGatt$1: onConnectionStateChange newState=2 status=0
2019-04-29 13:23:17.701 10672-10989/com.harley_davidson.ride_planner_blue D/RxBle#ConnectionOperationQueue: QUEUED   ServiceDiscoveryOperation(261794197)
2019-04-29 13:23:17.707 10672-10988/com.harley_davidson.ride_planner_blue D/RxBle#ConnectionOperationQueue: STARTED  ServiceDiscoveryOperation(261794197)
2019-04-29 13:23:17.709 10672-10817/com.harley_davidson.ride_planner_blue D/RxBle#ClientOperationQueue: FINISHED ConnectOperation(31134079) in 48 ms
2019-04-29 13:23:20.639 10672-10828/com.harley_davidson.ride_planner_blue D/RxBle#BluetoothGatt$1: onConnectionStateChange newState=0 status=22
2019-04-29 13:23:20.647 10672-10828/com.harley_davidson.ride_planner_blue I/RxBle#ConnectionOperationQueue: Connection operations queue to be terminated (00:16:A4:4A:26:9E)
2019-04-29 13:23:20.676 10672-10828/com.harley_davidson.ride_planner_blue D/RxBle#ClientOperationQueue: QUEUED   DisconnectOperation(49014646)
2019-04-29 13:23:20.677 10672-10817/com.harley_davidson.ride_planner_blue D/RxBle#ClientOperationQueue: STARTED  DisconnectOperation(49014646)
2019-04-29 13:23:20.689 10672-10988/com.harley_davidson.ride_planner_blue D/RxBle#ConnectionOperationQueue: FINISHED ServiceDiscoveryOperation(261794197) in 2986 ms
2019-04-29 13:23:20.689 10672-10988/com.harley_davidson.ride_planner_blue D/RxBle#Executors$RunnableAdapter: Terminated.
2019-04-29 13:23:20.716 10672-10817/com.harley_davidson.ride_planner_blue D/RxBle#ClientOperationQueue: FINISHED DisconnectOperation(49014646) in 39 ms
2019-04-29 13:23:20.729 10672-10989/com.harley_davidson.ride_planner_blue D/RxBle#ClientOperationQueue: QUEUED   ConnectOperation(204467731)
2019-04-29 13:23:20.732 10672-10817/com.harley_davidson.ride_planner_blue D/RxBle#ClientOperationQueue: STARTED  ConnectOperation(204467731)
2019-04-29 13:23:23.670 10672-10828/com.harley_davidson.ride_planner_blue D/RxBle#BluetoothGatt$1: onConnectionStateChange newState=2 status=0
2019-04-29 13:23:23.711 10672-10989/com.harley_davidson.ride_planner_blue D/RxBle#ConnectionOperationQueue: QUEUED   ServiceDiscoveryOperation(10770512)
2019-04-29 13:23:23.714 10672-10988/com.harley_davidson.ride_planner_blue D/RxBle#ConnectionOperationQueue: STARTED  ServiceDiscoveryOperation(10770512)
[... cut ...]
2019-04-29 13:24:11.978 10672-10744/com.harley_davidson.ride_planner_blue D/RxBle#BluetoothGatt$1: onConnectionStateChange newState=2 status=0
2019-04-29 13:24:12.011 10672-10989/com.harley_davidson.ride_planner_blue D/RxBle#ConnectionOperationQueue: QUEUED   ServiceDiscoveryOperation(47063742)
2019-04-29 13:24:12.014 10672-10988/com.harley_davidson.ride_planner_blue D/RxBle#ConnectionOperationQueue: STARTED  ServiceDiscoveryOperation(47063742)
2019-04-29 13:24:12.024 10672-10817/com.harley_davidson.ride_planner_blue D/RxBle#ClientOperationQueue: FINISHED ConnectOperation(16631161) in 519 ms

scotzark, 2019-04-29:
Some additional logging information:

2019-04-29 13:27:50.210 10672-10690/com.harley_davidson.ride_planner_blue D/RxBle#BluetoothGatt$1: onConnectionStateChange newState=2 status=0
2019-04-29 13:27:50.239 10672-10819/com.harley_davidson.ride_planner_blue D/BluetoothGatt: discoverServices() - device: 00:16:A4:4A:26:9E
2019-04-29 13:27:50.924 10672-10690/com.harley_davidson.ride_planner_blue D/BluetoothGatt: onConnectionUpdated() - Device=00:16:A4:4A:26:9E interval=6 latency=0 timeout=500 status=0
2019-04-29 13:28:10.256 10672-10819/com.harley_davidson.ride_planner_blue D/BluetoothManager: getConnectionState()
2019-04-29 13:28:10.256 10672-10819/com.harley_davidson.ride_planner_blue D/BluetoothManager: getConnectedDevices
2019-04-29 13:28:10.264 10672-10819/com.harley_davidson.ride_planner_blue D/BluetoothGatt: cancelOpen() - device: 00:16:A4:4A:26:9E
2019-04-29 13:28:10.269 10672-10690/com.harley_davidson.ride_planner_blue D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=00:16:A4:4A:26:9E
2019-04-29 13:28:10.271 10672-10690/com.harley_davidson.ride_planner_blue D/RxBle#BluetoothGatt$1: onConnectionStateChange newState=0 status=0
2019-04-29 13:28:10.277 10672-10819/com.harley_davidson.ride_planner_blue D/BluetoothGatt: close()
2019-04-29 13:28:10.277 10672-10819/com.harley_davidson.ride_planner_blue D/BluetoothGatt: unregisterApp() - mClientIf=7
2019-04-29 13:28:10.292 10672-10819/com.harley_davidson.ride_planner_blue D/BluetoothGatt: connect() - device: 00:16:A4:4A:26:9E, auto: false
2019-04-29 13:28:10.292 10672-10819/com.harley_davidson.ride_planner_blue D/BluetoothGatt: registerApp()
2019-04-29 13:28:10.294 10672-10819/com.harley_davidson.ride_planner_blue D/BluetoothGatt: registerApp() - UUID=eaa22739-71e3-451e-a356-5e7f0447516d
2019-04-29 13:28:10.296 10672-10690/com.harley_davidson.ride_planner_blue D/BluetoothGatt: onClientRegistered() - status=0 clientIf=7
2019-04-29 13:28:10.299 10672-10690/com.harley_davidson.ride_planner_blue D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=00:16:A4:4A:26:9E
2019-04-29 13:28:10.300 10672-10690/com.harley_davidson.ride_planner_blue D/RxBle#BluetoothGatt$1: onConnectionStateChange newState=2 status=0
2019-04-29 13:28:10.327 10672-10819/com.harley_davidson.ride_planner_blue D/BluetoothGatt: discoverServices() - device: 00:16:A4:4A:26:9E
2019-04-29 13:28:10.379 10672-10690/com.harley_davidson.ride_planner_blue D/BluetoothGatt: onConnectionUpdated() - Device=00:16:A4:4A:26:9E interval=48 latency=0 timeout=500 status=0
2019-04-29 13:28:11.163 10672-10690/com.harley_davidson.ride_planner_blue D/BluetoothGatt: onConnectionUpdated() - Device=00:16:A4:4A:26:9E interval=6 latency=0 timeout=500 status=0
2019-04-29 13:28:13.240 10672-10690/com.harley_davidson.ride_planner_blue D/BluetoothGatt: onClientConnectionState() - status=22 clientIf=7 device=00:16:A4:4A:26:9E
2019-04-29 13:28:13.243 10672-10690/com.harley_davidson.ride_planner_blue D/RxBle#BluetoothGatt$1: onConnectionStateChange newState=0 status=22
2019-04-29 13:28:13.269 10672-10819/com.harley_davidson.ride_planner_blue D/BluetoothManager: getConnectionState()
2019-04-29 13:28:13.269 10672-10819/com.harley_davidson.ride_planner_blue D/BluetoothManager: getConnectedDevices
2019-04-29 13:28:13.274 10672-10819/com.harley_davidson.ride_planner_blue D/BluetoothGatt: close()
2019-04-29 13:28:13.274 10672-10819/com.harley_davidson.ride_planner_blue D/BluetoothGatt: unregisterApp() - mClientIf=7
2019-04-29 13:28:13.302 10672-10819/com.harley_davidson.ride_planner_blue D/BluetoothGatt: connect() - device: 00:16:A4:4A:26:9E, auto: false
2019-04-29 13:28:13.302 10672-10819/com.harley_davidson.ride_planner_blue D/BluetoothGatt: registerApp()
2019-04-29 13:28:13.303 10672-10819/com.harley_davidson.ride_planner_blue D/BluetoothGatt: registerApp() - UUID=a89f5410-819c-482e-a332-be3bce3547df
2019-04-29 13:28:13.305 10672-10690/com.harley_davidson.ride_planner_blue D/BluetoothGatt: onClientRegistered() - status=0 clientIf=7

scotzark, 2019-04-29:
I do see this crash:

2019-04-29 14:17:21.837 18928-19690/com.harley_davidson.ride_planner_blue W/ConnectedVehicleService:     at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.onNext(ObservableFlatMap.java:139)
        at io.reactivex.internal.util.NotificationLite.accept(NotificationLite.java:246)
        at io.reactivex.internal.operators.observable.ObservableCache$ReplayDisposable.replay(ObservableCache.java:350)
        at io.reactivex.internal.operators.observable.ObservableCache$CacheState.onNext(ObservableCache.java:225)
        at io.reactivex.internal.operators.observable.ObservableElementAt$ElementAtObserver.onNext(ObservableElementAt.java:87)
        at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.tryEmit(ObservableFlatMap.java:262)
        at io.reactivex.internal.operators.observable.ObservableFlatMap$InnerObserver.onNext(ObservableFlatMap.java:559)
        at com.jakewharton.rxrelay2.PublishRelay$PublishDisposable.onNext(PublishRelay.java:176)
        at com.jakewharton.rxrelay2.PublishRelay.accept(PublishRelay.java:141)
        at com.polidea.rxandroidble2.internal.connection.DisconnectionRouter.onDisconnectedException(DisconnectionRouter.java:86)
        at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:77)
        at android.bluetooth.BluetoothGatt$1$4.run(BluetoothGatt.java:262)
        at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:770)
        at android.bluetooth.BluetoothGatt.access$200(BluetoothGatt.java:39)
        at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:257)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:71)
        at android.os.Binder.execTransact(Binder.java:731)

dariuszseweryn, 2019-04-30:
Hmm... I have not seen this crash — it may be related to some issue in `RxReplayingShare` or different. Hard to say without further investigation.

As for the logs they look a bit odd at first when the disconnections seems to be triggered by the application and subsequent ones are probably OS related. It does look preventable still

scotzark, 2019-04-30:
I implemented what this thread suggests:

https://stackoverflow.com/questions/23076641/ble-device-bonding-remove-automatically-in-android

But still it disconnects.  On the disconnect we are getting:

GATT exception from MAC address 00:16:A4:4A:26:9E, with type BleGattOperation{description='SERVICE_DISCOVERY'}

It seems that the service discovery process is causing the disconnection somehow.

dariuszseweryn, 2019-04-30:
Hmmm... Try postponing service discovery by about 2 seconds and see what will happen

scotzark, 2019-04-30:
With this logic I was able to get a connection once to stay stable, but I was not able to write any characteristics to the device.  Other than that one time though it is pretty consistent in the prior behavior of disconnecting repeatedly.  

I assume though that you wanted me to do the testing with the delay w/ the unpair behavior as spelled out in the above link?  I tried both ways, with and without the unpair, and it exhibited the same behavior.

Any other ideas?

dariuszseweryn, 2019-04-30:
You can try getting the HCI logs and sharing them here. It seems to be an issue with the Android BLE stack — but I have no idea what exactly is happening without logs.

scotzark, 2019-04-30:
[btsnoop_hci.log](https://github.com/Polidea/RxAndroidBle/files/3132540/btsnoop_hci.log)

scotzark, 2019-04-30:
It looks to me that one of the critical attributes for the HID is missing so Android disconnects to me.

dariuszseweryn, 2019-05-02:
It just seems that the discovery process has reached an end as there are no more requests. Funny enough the process is repeated after ~20 seconds. What may have happened:
The Android prior to Pie (9.0) was more flexibly interpreting the Service Discovery process and now it is more strict (in which case the problem lies in the peripheral's implementation)
I have not seen too much complaints for BLE on Pie so far

scotzark, 2019-05-02:
That is correct Darius.  I spoke with the equipment manufacturer and their implementation of HOGP had an error.  They provided a new build and things now work with Pie.  

Thanks for all the help.   Nice stack you have put together!

dariuszseweryn, 2019-05-02:
Glad you have found it