dariuszseweryn/RxAndroidBle #567: Checkstyle rule violations were found.

gaganAriel, 2019-04-30:
Checkstyle rule violations were found. See the report at: file:///C:/Users//Downloads/RxAndroidBle-master/rxandroidble/build/reports/checkstyle/checkstyle.html
Checkstyle files with violations: 1
Checkstyle violations by severity: [error:1]

i'm getting this kind of issue please help me i'm new in android

dariuszseweryn, 2019-04-30:
Have you checked the report?

gaganAriel, 2019-04-30:
report ?

gaganAriel, 2019-04-30:
i'm not getting your point

gaganAriel, 2019-04-30:
i just download source code and try to run on android studio 3.3

dariuszseweryn, 2019-04-30:
> Checkstyle rule violations were found. See the **report** at: file:///C:/Users//Downloads/RxAndroidBle-master/rxandroidble/build/reports/checkstyle/checkstyle.html
> Checkstyle files with violations: 1
> Checkstyle violations by severity: [error:1]

Current [builds](https://travis-ci.org/Polidea/RxAndroidBle/builds) seem to work correctly and pass the `checkstyle` task so I am fairly sure that that you must have made some changes to the source code

gaganAriel, 2019-04-30:
its not working my friend also try to run on his system but same error occured

dariuszseweryn, 2019-04-30:
Could you provide some more information? Contents of the report, what commit you try to build, what target?

gaganAriel, 2019-04-30:
can i send you report through email?

dariuszseweryn, 2019-04-30:
The contents of the report should not contain any private data (only projects files which are already published in this repository) — you can safely paste them here

gaganAriel, 2019-04-30:
okay i'll

gaganAriel, 2019-04-30:
sorry i'm not able to find out here please profile me specific path

dariuszseweryn, 2019-04-30:
You can simply post the contents of the file as a comment below inside as code

gaganAriel, 2019-04-30:
[checkstyle.zip](https://github.com/Polidea/RxAndroidBle/files/3131279/checkstyle.zip)

dariuszseweryn, 2019-04-30:
I see nothing unusual. Try running `./gradlew :checkstyle` command in the project's root folder and paste the output here

dariuszseweryn, 2019-05-14:
Closing due to lack of information and inactivity of the original poster.

When more data will be available—feel free to add that information and reopen this issue.

Best Regards

abir-hasan, 2019-05-16:
I had the same issue, didn't know what was causing it. so i modified checkstyle.gradle file to make a workaround.

`task checkstyle(type: Checkstyle) {
        configFile rootProject.file('checkstyle.xml')
        source 'src/main/java'
        ignoreFailures true
        showViolations false
        include '**/*.java'

        classpath = files()
    }`

after setting ignoreFailure true and showViolations false, now i can run the sample apps.

dariuszseweryn, 2019-05-16:
Could you try running `./gradlew :sample:assembleDebug` in the console and paste here the output? `checkstyle` should not run for samples and on the CI and my machine it all works fine so I cannot debug it.

abir-hasan, 2019-05-16:
Okay i ran this ` ./gradlew :sample-kotlin:assembleDebug` cause i'm exploring the kotlin sample. And this is what i got: 

> > Task :rxandroidble:checkstyle
> [ant:checkstyle] [ERROR] D:\workspace\GitHub\RxAndroidBle\rxandroidble\src\main\java\com\polidea\rxandroidble2\internal\connection
> \DisconnectionRouter.java:125: Line is longer than 140 characters (found 141). [LineLength]
> 
> > Task :rxandroidble:checkstyle FAILED
> 
> FAILURE: Build failed with an exception.
> 
> * What went wrong:
> Execution failed for task ':rxandroidble:checkstyle'.
> > Checkstyle rule violations were found. See the report at: file:///D:/workspace/GitHub/RxAndroidBle/rxandroidble/build/reports/ch
> eckstyle/checkstyle.html
>   Checkstyle files with violations: 1
>   Checkstyle violations by severity: [error:1]
> 
> 
> * Try:
> Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to
> get full insights.
> 
> * Get more help at https://help.gradle.org
> 
> Deprecated Gradle features were used in this build, making it incompatible with Gradle 5.0.
> Use '--warning-mode all' to show the individual deprecation warnings.
> See https://docs.gradle.org/4.10.1/userguide/command_line_interface.html#sec:command_line_warnings
> 
> BUILD FAILED in 10s
> 3 actionable tasks: 1 executed, 2 up-to-date

dariuszseweryn, 2019-05-16:
Hmmm... That is interesting. At least my machine says it is exactly 140 characters long. Not 141...
I will check on this.
[image]

dariuszseweryn, 2019-05-16:
@abir-hasan I am trying to figure out what is the issue here. Could you:

1. Paste here the full line (including whitespaces) from the file that causes the issue?
2. Confirm that you have just checked out the project, did not opened it in the IDE (which could potentially reformat the code) and run `./gradlew :sample-kotlin:assembleDebug` and the issue appeared?

abir-hasan, 2019-05-17:
Ohh, I actually did check out the project through Android Studio.

dariuszseweryn, 2019-05-17:
So are there any changes in your checked out repository vs the origin? Maybe there are additional whitespaces? Any more info is appreciated as I have no ideas at the moment.

abir-hasan, 2019-05-18:
When I first checked out the project through the Android Studio I wasn't able to run the sample-kotlin app. Then as I read this thread I thought that some file might have changed, so then I opened the directory through GitKraken to see if there really was any change in files due to Android Studio. But there wasn't any change. But as you are asking about whitespaces, if it's in any of the gitignored files then I also have no clue. The only silver lining is that I was able to tweak the checkstyle.gradle file to run the sample project.

And kudos on the library and the sample-kotlin app. It really is well-organized demo. Loved going through it.

dariuszseweryn, 2019-05-20:
Tweaking or disabling the `checkstyle` plugin is not a problem. Thing is that according to what you have write above the error is in:
```
D:\workspace\GitHub\RxAndroidBle\rxandroidble\src\main\java\com\polidea\rxandroidble2\internal\connection\DisconnectionRouter.java:125: Line is longer than 140 characters (found 141). [LineLength]
```
This file is not ignored. More over in the source code I see 139 characters on that particular line of code. The CI also sees no issue with the length of this line. However on your machine and @gaganAriel  's it seems to be longer than that. I am trying to figure it out what may be the cause of it. That is why I asked you to paste the whole line in here — maybe then we would be able to figure it out.

jreehuis, 2019-05-20:
I am not in CheckStyle in any means, but it might be caused that Windows is using `\l\n` as line break but *nx systems just `\n`.
In GIT you can configure that all *nx line breaks are converted to Windows and vise versa upon push.
(https://stackoverflow.com/questions/10418975/how-to-change-line-ending-settings)
So if CheckStyle also checks the line break chars on Windows it would find `141` chars in this line.

dariuszseweryn, 2019-05-20:
This is very much possible. I will dig deeper in this direction. Thank you for the hint @jreehuis 🥇

dariuszseweryn, 2019-05-21:
@abir-hasan I do not have access to a Windows machine right now. I wonder if the updated version of `checkstyle` (7.6.1 => 8.20) also displays the same behaviour or is it fixed. If you could:
1. check out branch `feature/checkstyle_version`
2. run `./gradlew :rxandroidble:checkstyle` and again paste the results here or just say if the output is the same
This could help me.

Alternatively you could set in your project settings to use the unix/os x line separator (`Android Studio` => `Preferences` => `Editor/Code Style`=> `Line separator`) and check if it would change anything

closed 2019-05-31 by dariuszseweryn

MWals, 2019-06-18:
Same issue here, checked out the repo, workaround:

in checkstyle.gradle:
subprojects { project ->
    group = GROUP
    version = VERSION_NAME

    apply plugin: 'checkstyle'

    checkstyle {
        toolVersion = '7.6.1'
    }

    task checkstyle(type: Checkstyle) {
        configFile rootProject.file('checkstyle.xml')
        source 'src/main/java'
        **ignoreFailures true**
        showViolations true
        include '**/*.java'

        classpath = files()
    }
}

dariuszseweryn, 2019-09-20:
This should be fixed now. Further checkouts of the repository should always have `LF` EOL markers which should make the `checkstyle` happy

PR #619 (2019-09-20): Fixed changing EOL markers on checkout
Git may automatically convert line endings to those used by the host system (CRLF on Windows, LF otherwise). This proved to be an issue for developers checking out the repository on Windows machines where `checkstyle` plugin would complain that some lines are too long because the EOL marker is longer. This `.gitattributes` file should make git always set the EOL marker to LF.

Fixes #567