dariuszseweryn/RxAndroidBle #569: Not using Bonded OS Connection

scotzark, 2019-05-10:
It appears that the stack isn’t reusing a bonded connection created by the OS, but is instead creating a new bonded connection.  From the hardware instrument we can observe that the phone has an existing bonded connection.  When the user connects with our app using the stack that existing connection is not being used.  Instead an entirely new bonded connection is being created.

dariuszseweryn, 2019-05-10:
At the same time there can be only one connection (as it is defined in Bluetooth Core Specification) established between any two BLE devices. Could you describe what do you mean by:
> an entirely new bonded connection is being created

And how does it affect your use case?

scotzark, 2019-05-10:
It's slowing the re connection speed down, because we need to exchange link keys again and it's creating more entrees in cached device list on the device.  The MAC address is rolled on successive bonds for security reasons.  I am not sure if we are doing something wrong that is causing the re-bond....or that is intended.

dariuszseweryn, 2019-05-11:
This behaviour is not dependent on the library. If a "physical" connection is established by the OS the `RxBleDevice.establishConnection()` will bind to it as well. If your peripheral rolls its MAC address right after the bond it makes me think — is that roll according to the Core Bluetooth Specification? If not then the next connection will probably be treated by the OS as a connection to an absolutely new device

dariuszseweryn, 2019-05-14:
@scotzark Any more info?

closed 2019-05-21 by dariuszseweryn