dariuszseweryn/RxAndroidBle #570: GATT exception without Status Code: Where do I go from here?

edster321, 2019-05-10:
Short summary: I have received the following error message and it doesn't have a specific code which would help to identify the underlying issue. What do I do to get to the bottom of this?

Error Message:
GATT exception from MAC address FB:85:88:D9:9B:73, with type BleGattOperation{description='CHARACTERISTIC_READ'}

Generated by Code:
    public void onClick(View v) {
        final RxBleDevice rxBleDevice = (RxBleDevice) v.getTag();
        Log.d(TAG, "Connecting to : " + rxBleDevice.getName() + " " + rxBleDevice.getMacAddress());

        Subscription powerSubscription = rxBleDevice.establishConnection(this,false)
                .flatMap(rxBleConnection -> rxBleConnection.readCharacteristic(powerCharacteristicUuid))
                .doOnError(error -> Log.d(TAG,"The error message is: " + error.getMessage()))
                .subscribe( (byte[] powerValue) -> {
                    // Read the value
                    Log.d(TAG, "power Subscription");
                    processPower(powerValue);
                });

... snip ...

Background:
Java-based Android app successfully uses RxAndroidBT library to talk to another BT device. I am adding a second device to obtain different information while retaining communication with the first. onClick() is being called after user selects the additional BT device to add. In other code I can successfully Discover services and BT characteristics on the devices. The BT codes match the ones I am looking for per spec, i.e basic communicates where established on basic scanning. Once scanned, I intended for app to drop the communications and re-establish communications with it assuming the user clicks on the new device.

Thank you for any debugging suggestions you might have.

dariuszseweryn, 2019-05-11:
Hello,

Check the class of the exception. Not only the message.

edster321, 2019-05-12:
Thank you for the insight. Looks like the exception is:
com.polidea.rxandroidble.exceptions.BleGattCannotStartException

edster321, 2019-05-13:
(Update) I think I've cracked the code on his one. My GATT characteristic required a setupNotification() call rather than a read() call. I'm now able to read data. I still likely doing some things poorly, but it's a breakthrough. I appreciate the advice.

Updated code to:

        final Subscription powerNotification = rxBleDevice.establishConnection(this,false)
                .flatMap(rxBleConnection -> rxBleConnection.setupNotification(powerCharacteristicUuid))
                .flatMap(observable -> observable)
                .observeOn(AndroidSchedulers.mainThread())
                .subscribe((byte[] powerValue) -> {
                    // Read the value
                    Log.d(TAG, "power Subscription");
                    processPower(powerValue);
                });

dariuszseweryn, 2019-05-13:
Glad you have found it

closed 2019-05-13 by dariuszseweryn

gauravdhim1990, 2023-03-28:
@edster321 thanks, same issue i was facing and i follow the same. 
readCharacteristic was throwing the same error for me 
**"GATT exception from MAC address *******, with type BleGattOperation{description='CHARACTERISTIC_READ'}"**

I finally used setupNotification and it's worked fine me 
thank you