dariuszseweryn/RxAndroidBle #571: Can't scan to device on the phone that close system log

xaxasemail, 2019-05-10:
### Disclaimer
Before you will post a new issue to the RxAndroidBle library consider if you are filling a bug/feature request or a general question 
about:
* Usage of RxJava
* Usage of Bluetooth Low Energy
* Usage of RxAndroidBle
* Weird behaviour of the peripheral you are using
* Other not directly related to a new feature request or a bug

If any of the above points seems to match your use case — consider using Google or creating a question on www.stackoverflow.com with 
a tag `rxandroidble` where it will be easier to access by other people with similar questions. Issues that are not bugs or feature requests
will be closed.

### Summary

### Library version
`X.X.X`

#### Preconditions

#### Steps to reproduce actual result
<br />1.
<br />2.
<br />3.
#### Minimum code snippet reproducing the issue

#### Logs from the application running with setting `RxBleLog.setLogLevel(RxBleLog.VERBOSE)`

#### Actual result

#### Expected result
I notice that the system log is closed in default on some phone like chinese phone meilan s6 . And can't scan to any device on that phone.How could I solve the problem? Thanks.

dariuszseweryn, 2019-05-11:
Hello,

Could you describe what you are facing in a bit more detail way?
I do not have access to the phone you mentioned. Check if no errors are captured during the scan.
I am not sure what you reference by:
> I notice that the system log is closed in default

xaxasemail, 2019-05-11:
The system log switch button is a switch button in Developer options ,we could print log in android studio when you turn it on.
Once we turn it off, we could not get the log from Log.d("asdf") in android studio cause we turn it off.
In a word, the switch button could control the output of the log.
I see , the function is less common on android phone. so very few people pay attention to this. Some huawei phones also have this function , but the switch button is in Engineering mode.

dariuszseweryn, 2019-05-12:
Is there any problem with turning the logs on for debugging purposes? If so you can set your custom logger to check the library logs.
If something goes really wrong you would get an error in the `.subscribe()` function callback.

xaxasemail, 2019-05-12:
My problem is that some devices cannot be scanned when the log is turned off, and the log switch on some Chinese mobile phones is turned off by default, so I wonder if I can normally scan the device when the log is turned off.

dariuszseweryn, 2019-05-12:
I see no reason why one thing would interfere with another

dariuszseweryn, 2019-05-13:
I can only see one reason for this to happen. _If_ a call to any `Log` method would throw a `RuntimeException` but it seems unlikely. Feel free to dig deeper into the issue and report back if you find a culprit.

dariuszseweryn, 2019-05-17:
@xaxasemail Any news?

closed 2019-05-21 by dariuszseweryn