dariuszseweryn/RxAndroidBle #572: LocationServicesOkObservable crash

ened, 2019-05-13:
### Disclaimer

This seem to have happened on a R17 phone running Android 8.1.0:
```
Fatal Exception: java.lang.IllegalArgumentException: Receiver not registered: com.polidea.rxandroidble2.internal.util.LocationServicesOkObservableApi23Factory$1$1@d23b395
       at android.app.LoadedApk.forgetReceiverDispatcher(LoadedApk.java:1247)
       at android.app.ContextImpl.unregisterReceiver(ContextImpl.java:1471)
       at android.content.ContextWrapper.unregisterReceiver(ContextWrapper.java:660)
       at com.polidea.rxandroidble2.internal.util.LocationServicesOkObservableApi23Factory$1$2.cancel(LocationServicesOkObservableApi23Factory.java:47)
       at io.reactivex.internal.disposables.CancellableDisposable.dispose(CancellableDisposable.java:48)
       at io.reactivex.internal.disposables.DisposableHelper.dispose(DisposableHelper.java:124)
       at io.reactivex.internal.operators.observable.ObservableCreate$CreateEmitter.dispose(ObservableCreate.java:121)
       at io.reactivex.internal.observers.BasicFuseableObserver.dispose(BasicFuseableObserver.java:152)
       at io.reactivex.internal.observers.BasicFuseableObserver.dispose(BasicFuseableObserver.java:152)
       at io.reactivex.internal.disposables.DisposableHelper.dispose(DisposableHelper.java:124)
       at io.reactivex.internal.operators.observable.ObservableSwitchMap$SwitchMapInnerObserver.cancel(ObservableSwitchMap.java:392)
       at io.reactivex.internal.operators.observable.ObservableSwitchMap$SwitchMapObserver.disposeInner(ObservableSwitchMap.java:174)
       at io.reactivex.internal.operators.observable.ObservableSwitchMap$SwitchMapObserver.dispose(ObservableSwitchMap.java:159)
       at io.reactivex.internal.operators.observable.ObservableSkip$SkipObserver.dispose(ObservableSkip.java:72)
       at io.reactivex.internal.disposables.DisposableHelper.dispose(DisposableHelper.java:124)
       at io.reactivex.internal.operators.mixed.SingleFlatMapObservable$FlatMapObserver.dispose(SingleFlatMapObservable.java:84)
       at io.reactivex.internal.observers.BasicFuseableObserver.dispose(BasicFuseableObserver.java:152)
       at io.reactivex.internal.observers.DisposableLambdaObserver.dispose(DisposableLambdaObserver.java:90)
       at io.reactivex.internal.operators.observable.ObservableDoOnEach$DoOnEachObserver.dispose(ObservableDoOnEach.java:79)
       at io.reactivex.internal.observers.DisposableLambdaObserver.dispose(DisposableLambdaObserver.java:90)
       at io.reactivex.internal.disposables.DisposableHelper.dispose(DisposableHelper.java:124)
       at io.reactivex.internal.operators.observable.ObservableConcatMap$ConcatMapDelayErrorObserver$DelayErrorInnerObserver.dispose(ObservableConcatMap.java:531)
       at io.reactivex.internal.operators.observable.ObservableConcatMap$ConcatMapDelayErrorObserver.dispose(ObservableConcatMap.java:382)
       at io.reactivex.internal.observers.DisposableLambdaObserver.dispose(DisposableLambdaObserver.java:90)
       at io.reactivex.internal.operators.observable.ObservableSwitchMap$SwitchMapObserver.dispose(ObservableSwitchMap.java:158)
       at io.reactivex.internal.disposables.DisposableHelper.dispose(DisposableHelper.java:124)
       at io.reactivex.internal.operators.observable.ObservableRetryWhen$RepeatWhenObserver.dispose(ObservableRetryWhen.java:123)
       at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.disposeAll(ObservableFlatMap.java:504)
       at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.dispose(ObservableFlatMap.java:310)
       at io.reactivex.internal.operators.observable.ObservableDoOnEach$DoOnEachObserver.dispose(ObservableDoOnEach.java:79)
       at io.reactivex.internal.observers.BasicFuseableObserver.dispose(BasicFuseableObserver.java:152)
       at io.reactivex.internal.disposables.DisposableHelper.dispose(DisposableHelper.java:124)
       at io.reactivex.internal.operators.observable.ObservableCombineLatest$CombinerObserver.dispose(ObservableCombineLatest.java:320)
       at io.reactivex.internal.operators.observable.ObservableCombineLatest$LatestCoordinator.cancelSources(ObservableCombineLatest.java:139)
       at io.reactivex.internal.operators.observable.ObservableCombineLatest$LatestCoordinator.dispose(ObservableCombineLatest.java:125)
       at io.reactivex.internal.operators.observable.ObservableDoOnEach$DoOnEachObserver.dispose(ObservableDoOnEach.java:79)
       at io.reactivex.internal.observers.BasicFuseableObserver.dispose(BasicFuseableObserver.java:152)
       at io.reactivex.internal.observers.BasicFuseableObserver.dispose(BasicFuseableObserver.java:152)
       at io.reactivex.internal.operators.observable.ObservableDoOnEach$DoOnEachObserver.dispose(ObservableDoOnEach.java:79)
       at io.reactivex.internal.observers.DisposableLambdaObserver.dispose(DisposableLambdaObserver.java:90)
       at io.reactivex.internal.observers.DisposableLambdaObserver.dispose(DisposableLambdaObserver.java:90)
       at io.reactivex.internal.operators.observable.ObservableTakeUntilPredicate$TakeUntilPredicateObserver.onNext(ObservableTakeUntilPredicate.java:72)
       at io.reactivex.internal.observers.DisposableLambdaObserver.onNext(DisposableLambdaObserver.java:58)
       at io.reactivex.internal.observers.DisposableLambdaObserver.onNext(DisposableLambdaObserver.java:58)
       at io.reactivex.internal.operators.observable.ObservableDoOnEach$DoOnEachObserver.onNext(ObservableDoOnEach.java:101)
       at io.reactivex.internal.operators.observable.ObservableDistinctUntilChanged$DistinctUntilChangedObserver.onNext(ObservableDistinctUntilChanged.java:85)
       at io.reactivex.internal.operators.observable.ObservableMap$MapObserver.onNext(ObservableMap.java:62)
       at io.reactivex.internal.operators.observable.ObservableDoOnEach$DoOnEachObserver.onNext(ObservableDoOnEach.java:101)
       at io.reactivex.internal.operators.observable.ObservableCombineLatest$LatestCoordinator.drain(ObservableCombineLatest.java:208)
       at io.reactivex.internal.operators.observable.ObservableCombineLatest$LatestCoordinator.innerNext(ObservableCombineLatest.java:237)
       at io.reactivex.internal.operators.observable.ObservableCombineLatest$CombinerObserver.onNext(ObservableCombineLatest.java:306)
       at io.reactivex.internal.operators.observable.ObservableMap$MapObserver.onNext(ObservableMap.java:62)
       at io.reactivex.internal.observers.DisposableLambdaObserver.onNext(DisposableLambdaObserver.java:58)
       at io.reactivex.internal.operators.observable.ObservableInterval$IntervalObserver.run(ObservableInterval.java:82)
       at io.reactivex.internal.schedulers.ScheduledDirectPeriodicTask.run(ScheduledDirectPeriodicTask.java:38)
       at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:457)
       at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:307)
       at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:302)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
       at java.lang.Thread.run(Thread.java:764)
```

I'm not yet sure how to reproduce it.

### Library version

`1.9.1`.

dariuszseweryn, 2019-05-13:
Do you use `LocationServicesOkObservable.createInstance()` function in your code?

ened, 2019-05-13:
@dariuszseweryn nope, nowhere. Should I?

dariuszseweryn, 2019-05-13:
I have looked into the stacktrace and I see the culprit. Apparently there may be a race condition in which a `BroadcastReceiver` may be unregistered before it was registered. Rather unlikely but as you can see it happened. Should be easy to fix.

ened, 2019-05-13:
Thank you. We are launching a new App this week. Any bug less to worry about is helpful. :-)

dariuszseweryn, 2019-05-13:
Do you use `RxBleClient.observeStateChanges()`?

ened, 2019-05-13:
Yes, we use it for for the scan observable. 
Final connection is NOT done through `RxAndroidBle` though, as we need to connect BTc.

dariuszseweryn, 2019-05-13:
A quick-fix is to add `.subscribeOn()` and `.unsubscribeOn()` with the same single-threaded `Scheduler`

closed 2019-05-13 by dariuszseweryn

dariuszseweryn, 2019-05-13:
And it is released. Should be fixed now

ened, 2019-05-13:
Thank you so much for the amazingly quick turnaround. We will upgrade today.

PR #573 (2019-05-13): Fixed IllegalArgumentException in LocationServicesOkObservableApi23Factory
Due to a race condition with subscribing/unsubscribing from different threads it may have happened that unregistering was made before registering finished resulting in an exception being thrown.

Fixes #572