dariuszseweryn/RxAndroidBle #576: Failed or timed out. Client disconnecting from SearchService!

Visstaralax, 2019-05-17:
### Summary
An app that links a mobile and a relay. The mobile sends hexadecimal commands. I have worked little with Android Studio and I do not know neither Observable nor RxJava.

### Library version
Sample project (merge functions of connectivity and write in same activity)

#### Preconditions
- Relay device with Mac address

#### Steps to reproduce actual result
<br />1. Create client with SampleApplication
<br />2. Hit button creates a device with "getBleDevice (macAddress)" and a Disposable with establishConnection (false).
<br />3. Sends command with bleDevice.establishConnection(false) and flatMapSingle(...write...bytes....)

#### Minimum code snippet reproducing the issue
```
public void Connect() {

       bleDevice = rxBleClient.getBleDevice(address);

       connectionDisposable = bleDevice.establishConnection(false)
               .observeOn(AndroidSchedulers.mainThread())
               .doFinally(this::SendCommand)
               .subscribe(this::onConnectionReceived, this::onConnectionFailure);

if (connectionDisposable != null)
   connectionDisposable.dispose();
}

private void SendCommand() {

   OpenCloseDevice( "C******************A", "open"); //abrir
   Handler handler = new Handler();
   handler.postDelayed (new Runnable()
   {
       public void run() {
           OpenCloseDevice("C******************A","close"); //cerrar
       }
   },1500);
}
   }

private void OpenCloseDevice(String command, String messenger) {
   byte[] data = hexStringToByteArray(command);

bleDevice.establishConnection(false)
       .flatMapSingle(rxBleConnection -> rxBleConnection.writeCharacteristic(UUID.fromString("0000ffe1-0000-1000-

8000-00805f9b34fb"), data))
           .take(1)
           .subscribe(
                   characteristicValue -> {
                       ScanActivity.this.runOnUiThread(new Runnable() {
                           @Override
                           public void run() {
                               if (messenger.equals("open"))
                                   Snackbar.make(findViewById(android.R.id.content), "Abriendo puerta...", Snackbar.LENGTH_SHORT).show();
                               else if (messenger.equals("close"))
                                   Snackbar.make(findViewById(android.R.id.content), "Cerrando puerta...", Snackbar.LENGTH_SHORT).show();

                           }
                       });
                   },
                   throwable -> {
                       Snackbar.make(findViewById(android.R.id.content), "Error " + throwable, Snackbar.LENGTH_SHORT).show();
                       Log.e ("E","rxjava error: " + throwable);
                   }
           );

}
```
#### Logs from the application running with setting 
```

****ERROR

05-17 08:40:19.227 10690-10712/? E/BluetoothRemoteDevices: state12newState0
05-17 08:40:19.289 10690-10859/? E/bt_btif: bta_gattc_mark_bg_con- unable to find the bg connection mask for: bb*****
05-17 08:40:19.324 10690-10712/? E/BluetoothRemoteDevices: state12newState1
05-17 08:40:19.910 3440-3861/? E/WifiConfigStore: updateConfiguration freq=2437 BSSID=***** RSSI=-53 "Goufone_*****"WPA_PSK
05-17 08:40:20.174 17222-17222/? E/SearchServiceStarter: Task COMMON_BROADCAST_MESSAGE_RECEIVED failed or timed out. Client 75744714928987424 disconnecting from SearchService!
java.util.concurrent.CancellationException: Task was cancelled.

***** VERBOSE

05-17 08:43:59.266 26210-26380/com.polidea.rxandroidble.sample I/RxBle#ClientOperationQueue: RUNNING ConnectOperation{MAC='BB*****', autoConnect=false}
05-17 08:43:59.267 26210-26380/com.polidea.rxandroidble.sample I/RxBle#ClientOperationQueue: RUNNING DisconnectOperation{MAC='BB:*****'}
05-17 08:43:59.268 26210-26633/com.polidea.rxandroidble.sample W/RxBle#QueueOperation: Disconnect operation has been executed but GATT instance was null - considering disconnected.
05-17 08:43:59.274 26210-26380/com.polidea.rxandroidble.sample I/RxBle#ClientOperationQueue: RUNNING ConnectOperation{MAC='BB*****', autoConnect=false}
05-17 08:43:59.277 26210-26633/com.polidea.rxandroidble.sample D/BluetoothGatt: connect() - device: *******, auto: false
05-17 08:43:59.277 26210-26633/com.polidea.rxandroidble.sample D/BluetoothGatt: registerApp()
05-17 08:43:59.277 26210-26633/com.polidea.rxandroidble.sample D/BluetoothGatt: registerApp() - UUID=e6969390-****-497f-a063-57b897cad446
05-17 08:43:59.277 10690-10734/? I/Bluetooth: BluetoothProfileService:isAvailable(), mStartError=false, mCleaningUp=false
05-17 08:43:59.278 10690-10734/? I/Bluetooth: BtGatt.GattService:registerClient() - UUID=e6969390-****-497f-a063-57b897cad446
05-17 08:43:59.278 10690-10859/? I/bt_att: GATT_Register
05-17 08:43:59.279 10690-10859/? D/bt_att: UUID=[0xe69693903124497fa06357b******]
05-17 08:43:59.279 10690-10859/? I/bt_att: allocated gatt_if=6
05-17 08:43:59.279 10690-10859/? I/bt_att: GATT_StartIf gatt_if=6
05-17 08:43:59.279 10690-10859/? D/bt_att: gatt_find_the_connected_bda start_idx=0
05-17 08:43:59.279 10690-10859/? D/bt_att: gatt_find_the_connected_bda found=0 found_idx=7
05-17 08:43:59.279 10690-10712/? W/bt_btif: HAL bt_gatt_callbacks->client->register_client_cb
05-17 08:43:59.279 10690-10712/? I/Bluetooth: BtGatt.GattService:onClientRegistered() - UUID=e6969390-****-497f-a063-57b897cad446, clientIf=6
05-17 08:43:59.279 26210-26294/com.polidea.rxandroidble.sample D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
05-17 08:43:59.280 10690-10701/? I/Bluetooth: BluetoothProfileService:isAvailable(), mStartError=false, mCleaningUp=false
05-17 08:43:59.281 10690-10701/? I/Bluetooth: BtGatt.GattService:clientConnect() - address=******, isDirect=true
05-17 08:43:59.282 10690-10712/? D/bt_btif_config: btif_get_address_type: Device [******] address type 0
05-17 08:43:59.282 10690-10712/? D/bt_btif_config: btif_get_device_type: Device [***** type 2
05-17 08:43:59.282 10690-10859/? I/bt_btm: BTM_InqDbRead: bd addr [*****]
05-17 08:43:59.282 10690-10859/? I/bt_att: GATT_Connect gatt_if=6
05-17 08:43:59.282 10690-10859/? D/bt_att: gatt_get_ch_state: ch_state=0
05-17 08:43:59.282 10690-10859/? D/bt_att: gatt_set_ch_state: old=0 new=2
05-17 08:43:59.282 10690-10859/? I/bt_btm: BTM_InqDbRead: bd addr [*****]
05-17 08:43:59.282 10690-10859/? I/bt_btm: btm_find_or_alloc_dev
05-17 08:43:59.282 10690-10859/? D/bt_btif_dm: check_unuse_private_addr_bl: ******* is not in blacklist for unused private address

```
#### Actual result

Unstable connections, sometimes just open the door, sometimes just close the door and sometimes it does not do anything. Also the bluetooth connection stays open (fixed led light)

#### Expected result

Stable connections and just open and close relay (close with delay of 1500 ms).

#### Questions
- When I use dispose?
- When I use retry (to retry connection)?

dariuszseweryn, 2019-05-17:
Hello!

Thank you for using the library.

This is a general usage question and as such should be posted on [stackoverflow](https://stackoverflow.com/questions/tagged/rxandroidble) with `rxandroidble` tag—it will be easier to find by other users. Also feel free to check already posted questions as a similar one may be already answered.

Best Regards

P.S. If you are new to the reactive programming you can check some of [RxJava 2 tutorials](https://www.google.com/search?q=rxjava+2+tutorial)

closed 2019-05-17 by dariuszseweryn

Visstaralax, 2019-05-17:
Hello. Thank you for response and for create library :)