dariuszseweryn/RxAndroidBle #582: BLE long read (> 600 bytes)

GuillaumeLefevre, 2019-05-20:
Hello,

I'm struggling with reading data longer than 600 bytes..
I tried the solution of this post with no result : [https://stackoverflow.com/questions/48741196/android-ble-client-only-returns-600-byte-of-data-in-oncharacteristicread](url)

The last bytes are getting discarded whatever MTU i'm using (any relation with the MTU ?).

MTU -> Data length
247 (Max of my ble device) -> 600
201 -> 514
200 -> 512
185 (iOS MTU) -> 482
151 -> 414
101 -> 300
92 -> 273
61 -> 180
51 -> 150
23 (Min of my ble device) -> 66

It's working well on MAC and iOS (BlueSee / Nrf Connect), i'm getting 512 bytes on the first read and the remaining chunks in a second read.

I don't know if it's a side effect of the library or not...
Any idea ?

GL

dariuszseweryn, 2019-05-20:
Hello,

Thank you for using the library.
Your post seems to be a bug report but it lacks some crucial information. Could you edit to contain information according to the [issue template](https://github.com/Polidea/RxAndroidBle/blob/master/.github/ISSUE_TEMPLATE.md)?

Best Regards

GuillaumeLefevre, 2019-05-20:
@dariuszseweryn Sry about that, just edited my post :)

dariuszseweryn, 2019-05-20:
According to Core `BLUETOOTH SPECIFICATION Version 5.0 | Vol 3, Part F`, `3.2.9 Long Attribute Values`:

> The maximum length of an attribute value shall be 512 octets.

Having longer values is against specification and it is hard to say what will happen

dariuszseweryn, 2019-05-20:
From what I see the output is more or less according to what you have implemented judging from the code snippet. First of all I do not see a reason to create a custom operation for this one but just subscribe two times to the `readCharacteristic()` observable. Anyway this is not an issue with the library. This is a general usage question and as such should be posted on [stackoverflow](https://stackoverflow.com/questions/tagged/rxandroidble) with `rxandroidble` tag—it will be easier to find by other users. Also feel free to check already posted questions as a similar one may be already answered.

closed 2019-05-20 by dariuszseweryn

VictorVanWeyenberg, 2021-09-13:
Also bluetooth spec, couple of lines above the line you quotes, it says:

> An attribute value may be defined to be larger than (ATT_MTU-1) octets in size. These attributes are called long attributes. To read the entire value of an attributes larger than (ATT_MTU- 1) octets, the ATT_READ_BLOB_REQ PDU is used. It is possible to read the first (ATT_MTU-1) octets of a long attribute value using the ATT_READ_REQ PDU.

Although the max value length of a characteristic is 512 bytes, this blob behaviour can be used to extend that length to 65536 bytes theoratically.

I've seen this being supported in blueZ. Some tests I ran there only yielded success if I kept my value length around 32kB.

jci-zimm, 2022-02-05:
Anyone know how to get past this 600 byte limitation? I tried changing `ESP_GATT_MAX_ATTR_LEN` [here](https://github.com/espressif/esp-idf/blob/3441dd1e9832a99dbd823e7dc73700169815772f/components/bt/host/bluedroid/api/include/api/esp_gatt_defs.h#L295) and `GATT_MAX_ATTR_LEN` [here](https://github.com/espressif/esp-idf/blob/1cb31e50943bb757966ca91ed7f4852692a5b0ed/components/bt/host/bluedroid/stack/include/stack/gatt_api.h#L137) and `BTA_GATTC_CACHE_SRVR_SIZE` to a larger size without luck. Is there something beneath bluedroid on the ESP32 that is hardcoded to 600?