dariuszseweryn/RxAndroidBle #583: Disconnecting with status 19

arthurmazer, 2019-05-21:
I'm developing an application to pair with two kind of sensors with the app and also these sensors are pairing with each other, we developed our custom bluetooth communication protocol. The connection is working great usually, but there are still some errors that I'm having hard time to fix it.

Sensor 1 paring alone is working great, but every time I'm pairing both of them, then i close the app, pair it again with first sensor, I got disconnected with status 19 just after the connection is established, after I try again one or two times the connection will be established properly. I was thinking that was a problem with Gatt refresh, but I already tried one solution found here and I keep reproducing this error every time.

`
sensorConnectionSubscription?.dispose()  
val connectionSubscription = device.device.establishConnection(false)
                .flatMapSingle { connection ->

                    if (device is Sensor) {
                        sensorConnection = connection
                        connectedSensor = device
                    } else if (device is Dongle) {
                        dongleConnection = connection
                        connectedDongle = device
                    }

                    connection.queue(CustomRefresh())
                            ?.observeOn(Schedulers.io())
                            ?.doOnComplete{
                                BleLogHelper.writeLog("GATT REFRESHED")
                            }
                            ?.subscribe ({
                                BleLogHelper.writeLog("GATT REFRESHED")
                            },{
                                BleLogHelper.writeLog("FAIL REFRESHING GATT")
                            })

                    BleLogHelper.writeLog("Send Request Connection Command $deviceType")
                    val command = BraincareBluetoothCommandProtocol.createRequestConnectionCommandFrame(deviceType)
                    connection.writeCharacteristic(BraincareBluetoothProtocol.rxCharacteristicUUID, command)
                }
                .delay(300, TimeUnit.MILLISECONDS)
                .subscribe({
                    sensorConnection?.queue(CustomRefresh())
                            ?.observeOn(Schedulers.io())
                            ?.doOnComplete{
                                BleLogHelper.writeLog("GATT REFRESHED")
                            }
                            ?.subscribe ({
                                BleLogHelper.writeLog("GATT REFRESHED")
                            },{
                                BleLogHelper.writeLog("FAIL REFRESHING GATT")
                            })

                    BleLogHelper.writeLog("Connection Established ${device.type}")
                    val iscon= this.isConnecting
                    startBlinkingDeviceLed(deviceType, pairColor)
                    connectionFlowListeners.forEach { it.onConnectionEstablished(device) }
                }, {
                    BleLogHelper.writeError("Connection Lost ${device.type}", it)
                    BleLogHelper.writeError("Retrying...", it)
})

sensorConnectionSubscription = connectionSubscription`

Log error:

`2019-05-21 10:54:11.816 11797-11889/io.b4c.brain4care.debug E/BLEBC: 21/05/2019 10:54:11.810 - Connection Lost SENSOR
    com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from D4:57:4F:53:44:E7 with status 19 (UNKNOWN)
        at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:77)
        at android.bluetooth.BluetoothGatt$1$4.run(BluetoothGatt.java:268)
        at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:789)
        at android.bluetooth.BluetoothGatt.-wrap0(Unknown Source:0)
        at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:264)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:70)
        at android.os.Binder.execTransact(Binder.java:682)
2019-05-21 10:54:11.840 11797-11889/io.b4c.brain4care.debug E/BLEBC: 21/05/2019 10:54:11.836 - Retrying...
    com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from D4:57:4F:53:44:E7 with status 19 (UNKNOWN)
        at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:77)
        at android.bluetooth.BluetoothGatt$1$4.run(BluetoothGatt.java:268)
        at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:789)
        at android.bluetooth.BluetoothGatt.-wrap0(Unknown Source:0)
        at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:264)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:70)
        at android.os.Binder.execTransact(Binder.java:682)
`

arthurmazer, 2019-05-21:
[btsnoop_hci.log](https://github.com/Polidea/RxAndroidBle/files/3203008/btsnoop_hci.log)

My snoop_hci log

dariuszseweryn, 2019-05-21:
Hello!

Thank you for using the library.

This is a general usage question and as such should be posted on [stackoverflow](https://stackoverflow.com/questions/tagged/rxandroidble) with `rxandroidble` tag—it will be easier to find by other users. Also feel free to check already posted questions as a similar one may be already answered.

Best Regards

closed 2019-05-21 by dariuszseweryn

arthurmazer, 2019-05-21:
Thanks for the amazing help

dariuszseweryn, 2019-05-21:
Please cut the sarcasm. As I see you have already posted a question on [Stackoverflow](https://stackoverflow.com/questions/56240955) which is good as you have much better chance of getting the _free_ support there either from me or from other Stackoverflow users. 

Unfortunately my time is limited and what you face does not seem to be an issue with the library. Issues here are mostly to describe bugs, feature requests or other issues with the library.

You did not bother to conform to the issue template. If you did you would see where you can expect help with usage of the library/BLE (and why there). You could also try the newest version of the library which could potentially help you.

Please respect my time — or any other open source maintainer for that matter.

Cheers