dariuszseweryn/RxAndroidBle #584: android.permission.BLUETOOTH_ADMIN

scotzark, 2019-05-23:
I have seen frequent references in Stack Overflow that the android.permission.BLUETOOTH_ADMIN is required for the library to work properly.  But in my testing (so far on Android 9 and 8) I have found that this permission is not required.  Is it true that this isn't required?  Or is it required on older OS versions?

dariuszseweryn, 2019-05-23:
I think [this documentation page](https://developer.android.com/reference/android/bluetooth/le/BluetoothLeScanner) should answer your question.

From my knowledge this permission is needed to manage scanning and bluetooth adapter.

scotzark, 2019-05-23:
Hi Darius-

Yeah I am aware of the documentation.  It just seems strange that the library does work properly even without this permission.   At the end of the day it is probably best to just request it given all the different flavors of Android.

Steppschuh, 2019-05-23:
We observed that starting or stopping a `BluetoothLeScanner` doesn't require the `BLUETOOTH_ADMIN` permission. Neither does using this library. You just need it if you want to change the state of a `BluetoothAdapter` (i.e. when turning Bluetooth on or off).

dariuszseweryn, 2019-05-23:
What have you tested on?

scotzark, 2019-05-23:
Pixel 3XL - Android 9.0
Samsung Galaxy S8 - Version 8.0
Samsung Galaxy S8 - Version 7.0 -> Just tried this one...not in the original ticket.

dariuszseweryn, 2019-05-24:
Are you sure your test app does not have `BLUETOOTH_ADMIN` permission? I have just tested on Nexus 5X and got the below when tried to start a scan:
```
2019-05-24 10:56:16.227 13065-13102/com.polidea.rxandroidble.sample W/Binder: Caught a RuntimeException from the binder stub implementation.
    java.lang.SecurityException: Need BLUETOOTH_ADMIN permission: Neither user 10267 nor current process has android.permission.BLUETOOTH_ADMIN.
        at android.os.Parcel.readException(Parcel.java:1942)
        at android.os.Parcel.readException(Parcel.java:1888)
        at android.bluetooth.IBluetoothGatt$Stub$Proxy.startScan(IBluetoothGatt.java:918)
        at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper.onScannerRegistered(BluetoothLeScanner.java:431)
        at android.bluetooth.le.IScannerCallback$Stub.onTransact(IScannerCallback.java:56)
        at android.os.Binder.execTransact(Binder.java:674)
```

dariuszseweryn, 2019-05-24:
Mark that is was not returned to the application. Just the binder printed a warning.

scotzark, 2019-05-24:
I should also add that our user has a requirement that the user must have the device paired at the OS level to use the app.  They do not use the app to pair to the device.

dariuszseweryn, 2019-05-24:
> I should also add that our user has a requirement that the user must have the device paired at the OS level to use the app. They do not use the app to pair to the device.

Could you describe why is this relevant?

dariuszseweryn, 2019-05-31:
@scotzark I have also tested on LG G7 with Android 8.1.0 and it also requires `BLUETOOTH_ADMIN`. I cannot confirm your findings

closed 2019-05-31 by dariuszseweryn

vadimlukutin, 2024-07-15:
@dariuszseweryn Please can u describe how u resolve this issue.