dariuszseweryn/RxAndroidBle #588: Recommended way to handle cross-platform exceptions

devxpy, 2019-05-25:
Hey there! Thanks for this wonderful and easy to use library.

I'm currently developing a [flutter plugin](https://github.com/pycampers/flutter-rx-ble) that wraps your android and ios libraries. 

I have added the exception serialization and re-throwing mechanism by copying over the error classes from RxAndroidBle [here](https://github.com/pycampers/flutter-rx-ble/blob/master/lib/src/exceptions.dart).

But I'm a little lost as to how I can translate swift error codes to conform to the classes above.

The `BluetoothError.swift` file seems to contain quite a lot of cases, which are not present in the java side. 

Could you please guide me as to how I should handle errors in a cross-platform way?

dariuszseweryn, 2019-05-27:
You should probably extract the superset of both worlds and map the native errors appropriately. You may check `Polidea/react-native-ble-plx` library for inspiration.

devxpy, 2019-05-27:
I noticed in your [docs](https://polidea.github.io/react-native-ble-plx/#bleerrorcode) that there are three types of error codes, `BleErrorCode`, `
BleAndroidErrorCode`, and `BleIOSErrorCode`. Are you doing a translation of Platform-specific codes to `BleErrorCode`? Can I please have a link to that code?
Or are these 3 classes mutually exclusive?

Cierpliwy, 2019-05-28:
Hey, there are four types of errors in total:
- ATT ErrorCode - this is an error code which is platform independent and is important for debugging purposes. On Android when error code is in  0x012 – 0x7F range, it is an ATT error code. Otherwise, it is platform specific code. On iOS there is special error namespace for ATT errors.
- iOS ErrorCode - platform specific, might be useful.
- Android ErrorCode - platform specific, might be useful.
- ErrorCode - platform independent mapping of the most common types of errors.

dariuszseweryn, 2019-05-31:
If that is all I close this topic

closed 2019-05-31 by dariuszseweryn