dariuszseweryn/RxAndroidBle #592: Receiver not registered

ened, 2019-06-18:
### Summary

IllegalArgumentException ("Receiver not registered") for BleState

### Library version
`1.10.1`

#### Preconditions

Create, Subscribe and immediately dispose a BleState observable.

#### Steps to reproduce actual result
<br />1.Create a RxBleAdapterStateObservable
<br />2.Subscribe to it
<br />3.Immediately dispose the observable (we are doing this once a specific condition is matched).

#### Actual result

```
 java.lang.IllegalArgumentException: Receiver not registered: com.polidea.rxandroidble2.RxBleAdapterStateObservable$1$1@715f55
 	at android.app.LoadedApk.forgetReceiverDispatcher(LoadedApk.java:1402)
 	at android.app.ContextImpl.unregisterReceiver(ContextImpl.java:1769)
 	at android.content.ContextWrapper.unregisterReceiver(ContextWrapper.java:659)
 	at com.polidea.rxandroidble2.RxBleAdapterStateObservable$1$2.cancel(RxBleAdapterStateObservable.java:73)
 	at io.reactivex.internal.disposables.CancellableDisposable.dispose(CancellableDisposable.java:48)
 	at io.reactivex.internal.disposables.DisposableHelper.set(DisposableHelper.java:53)
 	at io.reactivex.internal.operators.observable.ObservableCreate$CreateEmitter.setDisposable(ObservableCreate.java:106)
 	at io.reactivex.internal.operators.observable.ObservableCreate$CreateEmitter.setCancellable(ObservableCreate.java:111)
 	at com.polidea.rxandroidble2.RxBleAdapterStateObservable$1.subscribe(RxBleAdapterStateObservable.java:70)
 	at io.reactivex.internal.operators.observable.ObservableCreate.subscribeActual(ObservableCreate.java:40)
 	at io.reactivex.Observable.subscribe(Observable.java:12267)
 	at io.reactivex.internal.operators.observable.ObservableSubscribeOn$SubscribeTask.run(ObservableSubscribeOn.java:96)
 	at io.reactivex.internal.schedulers.TrampolineScheduler.scheduleDirect(TrampolineScheduler.java:52)
 	at io.reactivex.internal.operators.observable.ObservableSubscribeOn.subscribeActual(ObservableSubscribeOn.java:36)
 	at io.reactivex.Observable.subscribe(Observable.java:12267)
 	at io.reactivex.internal.operators.observable.ObservableUnsubscribeOn.subscribeActual(ObservableUnsubscribeOn.java:32)
 	at io.reactivex.Observable.subscribe(Observable.java:12267)
 	at io.reactivex.internal.operators.observable.ObservablePublish.connect(ObservablePublish.java:116)
 	at io.reactivex.internal.operators.observable.ObservableRefCount.subscribeActual(ObservableRefCount.java:87)
 	at io.reactivex.Observable.subscribe(Observable.java:12267)
 	at com.polidea.rxandroidble2.RxBleAdapterStateObservable.subscribeActual(RxBleAdapterStateObservable.java:86)
 	at io.reactivex.Observable.subscribe(Observable.java:12267)
 	at io.reactivex.internal.operators.observable.ObservableConcatMap$ConcatMapDelayErrorObserver.drain(ObservableConcatMap.java:475)
 	at io.reactivex.internal.operators.observable.ObservableConcatMap$ConcatMapDelayErrorObserver.onSubscribe(ObservableConcatMap.java:330)
 	at io.reactivex.internal.operators.observable.ObservableFromArray.subscribeActual(ObservableFromArray.java:31)
 	at io.reactivex.Observable.subscribe(Observable.java:12267)
 	at io.reactivex.internal.operators.observable.ObservableConcatMap.subscribeActual(ObservableConcatMap.java:55)
 	at io.reactivex.Observable.subscribe(Observable.java:12267)
 	at io.reactivex.internal.operators.observable.ObservableSwitchMap.subscribeActual(ObservableSwitchMap.java:51)
 	at io.reactivex.Observable.subscribe(Observable.java:12267)
 	at io.reactivex.internal.operators.observable.ObservableDistinctUntilChanged.subscribeActual(ObservableDistinctUntilChanged.java:35)
 	at io.reactivex.Observable.subscribe(Observable.java:12267)
 	at io.reactivex.internal.operators.observable.ObservableSkip.subscribeActual(ObservableSkip.java:29)
 	at io.reactivex.Observable.subscribe(Observable.java:12267)
 	at io.reactivex.internal.operators.mixed.SingleFlatMapObservable$FlatMapObserver.onSuccess(SingleFlatMapObservable.java:109)
 	at io.reactivex.internal.operators.single.SingleMap$MapSingleObserver.onSuccess(SingleMap.java:64)
 	at io.reactivex.internal.operators.observable.ObservableCountSingle$CountObserver.onComplete(ObservableCountSingle.java:82)
 	at io.reactivex.internal.operators.observable.ObservableTakeWhile$TakeWhileObserver.onNext(ObservableTakeWhile.java:84)
 	at io.reactivex.internal.operators.observable.ObservableInterval$IntervalObserver.run(ObservableInterval.java:82)
 	at io.reactivex.internal.schedulers.ScheduledDirectPeriodicTask.run(ScheduledDirectPeriodicTask.java:38)
 	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:458)
 	at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:307)
 	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:302)
 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
 	at java.lang.Thread.run(Thread.java:784)
```

#### Expected result

No crashes.

#### Minimum code snippet reproducing the issue

N/A as I have to extract the code.

Logs and further info will be added asap, but perhaps this is a easy issue to fix (not the first of it's kind @dariuszseweryn ?).

dariuszseweryn, 2019-06-18:
I would need to have a code snippet that subscribes/unsubscribes to recreate/understand/showcase the issue

dariuszseweryn, 2019-06-26:
@ened Ping?

ened, 2019-06-26:
No time yet to build a sample. We did not see this in earlier versions of the library.

Currently 100 users affected in production, across Android 6 to 9.

ened, 2019-06-26:
The code change here: https://github.com/Polidea/RxAndroidBle/pull/575/files#diff-016b41641d542449b5cade3ae114a495R61 could possibly be extended by putting the `receiver` in an `AtomicReference`, and instantiate it as close as possible to the actual `registerReceiver` call.

dariuszseweryn, 2019-06-27:
Still I need the example code to test possible solutions

PR #596 (2019-06-28): Fixed IllegalStateException on RxBleAdapterStateObservable
Setting a cancelable on an `Emitter` did run it immediately if it was already disposed. Postponing the setting until after the `BroadcastReceiver` is registered fixed the issue.

Fixes #592
closed 2019-06-28 by dariuszseweryn