dariuszseweryn/RxAndroidBle #595: BleGattCallbackTimeoutException when  restarting the app

dthib, 2019-06-26:
# Summary
Hi, 

I really need your help :) : I am getting BleGattCallbackTimeoutException when I restart the app (very predictable: once out of two restart, in other words: I am restarting the app i get a timeout, I am restarting agin ok, I am restarting again timeout, etc.)
If I try again when I get a timeout it will succeed. 

Another weird thing is that when it's supposed not to work (1 out of 2):
- I can connect with another phone
- I cannot connect even with an app such as nrfConnect

It seems to me that it's related to BluetoothAdapter state. Any help really appreciated, I am struggling for days....

Please find below the logs. I have reproduced it even by creating an isolated sample app doing only connection.

### Library version
version 2.0.2

#### Logs from the application running with settings:

```
D/TallyJumpBluetoothCon: BLE Client state: READY
E/: [ZeroHung]zrhung_get_config: Get config failed for wp[0x0008]
D/RxBle#ClientOperationQueue: QUEUED   ScanOperationApi21(197577894)
D/RxBle#ClientOperationQueue: STARTED  ScanOperationApi21(197577894)
I/RxBle#ClientOperationQueue: RUNNING  ScanOperationApi21{ANY_MUST_MATCH -> nativeFilters=[BluetoothLeScanFilter [mDeviceName=null, MAC=null, mUuid=..., mUuidMask=null, mServiceDataUuid=null, mServiceData=null, mServiceDataMask=null, mManufacturerId=-1, mManufacturerData=null, mManufacturerDataMask=null]]}
I/RxBle#QueueOperation: Scan operation is requested to start.
D/HwAppInnerBoostImpl: asyncReportData androidtallyjump.tallyjump.com.testble,2,1,2,0 interval=344
D/RxBle#ScanOperationApi21: No library side filtering —> debug logs of scanned devices disabled
I/BluetoothAdapter: getBluetoothLeScanner
D/BluetoothAdapter: isLeEnabled(): ON
D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=8 mScannerId=0
D/RxBle#ClientOperationQueue: FINISHED ScanOperationApi21(197577894) in 17 ms
I/HiTouch_HiTouchSensor: enabledInPad = false,isPcCastMode = false
D/mali_winsys: EGLint new_window_surface(egl_winsys_display *, void *, EGLSurface, EGLConfig, egl_winsys_surface **, EGLBoolean) returns 0x3000
D/HwAppInnerBoostImpl: asyncReportData androidtallyjump.tallyjump.com.testble,2,2,1,21 interval=412
I/ViewRootImpl: jank_removeInvalidNode jank list is null
D/OpenGLRenderer:   HWUI Binary is  enabled
D/HwAppInnerBoostImpl: asyncReportData androidtallyjump.tallyjump.com.testble,2,1,3,0 interval=599
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=tj}, rssi=-66, timestampNanos=689726823475103, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
W/ble: C8:DC:AB:4B:C0:73
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=tj}, rssi=-75, timestampNanos=689727175377707, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
W/ble: F2:D8:80:27:7A:B1
D/AwareBitmapCacher: handleInit switch not opened pid=25830
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=tj}, rssi=-90, timestampNanos=689728426342811, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
W/ble: F2:D8:80:27:7A:B1
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=tj}, rssi=-88, timestampNanos=689728432236561, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
W/ble: DB:09:63:86:CC:04
I/ViewRootImpl: jank_removeInvalidNode all the node in jank list is out of time
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=tj}, rssi=-56, timestampNanos=689729356273020, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
W/ble: C8:DC:AB:4B:C0:73
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=tj}, rssi=-93, timestampNanos=689729693823540, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
W/ble: DB:09:63:86:CC:04
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=tj}, rssi=-82, timestampNanos=689729698001144, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
W/ble: F2:D8:80:27:7A:B1
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=tj}, rssi=-67, timestampNanos=689730631643332, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
W/ble: C8:DC:AB:4B:C0:73
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=tj}, rssi=-66, timestampNanos=689733157575102, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
W/ble: C8:DC:AB:4B:C0:73
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=tj}, rssi=-73, timestampNanos=689733505705831, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
W/ble: F2:D8:80:27:7A:B1
I/RxBle#Client: ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=tj}, rssi=-87, timestampNanos=689734756789165, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...]}
W/ble: DB:09:63:86:CC:04
D/TallyJumpBluetoothCon: BLE Client state: READY
W/ble: stop scanning
I/RxBle#CancellableDisposable: Scan operation is requested to stop.
I/BluetoothAdapter: getBluetoothLeScanner
D/BluetoothAdapter: isLeEnabled(): ON
D/TallyJumpBluetoothCon: Waiting before starting connection just after scan: 689735830
D/TallyJumpBluetoothCon: Now complete scan (12) to connect: 689736346
D/RxBle#ClientOperationQueue: QUEUED   ConnectOperation(185132267)
[... 35 lines cut ...]
        at com.polidea.rxandroidble2.internal.util.DisposableUtil$3.onError(DisposableUtil.java:60)
        at io.reactivex.internal.operators.single.SingleDoFinally$DoFinallyObserver.onError(SingleDoFinally.java:79)
        at io.reactivex.internal.operators.single.SingleTimeout$TimeoutMainObserver$TimeoutFallbackObserver.onError(SingleTimeout.java:97)
        at io.reactivex.internal.operators.single.SingleFromCallable.subscribeActual(SingleFromCallable.java:48)
        at io.reactivex.Single.subscribe(Single.java:3603)
        at io.reactivex.internal.operators.single.SingleTimeout$TimeoutMainObserver.run(SingleTimeout.java:126)
        at io.reactivex.internal.schedulers.ScheduledDirectTask.call(ScheduledDirectTask.java:38)
        at io.reactivex.internal.schedulers.ScheduledDirectTask.call(ScheduledDirectTask.java:26)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:301)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:784)
D/RxBle#ClientOperationQueue: QUEUED   DisconnectOperation(243638813)
V/RxBle#Executors: Terminated (MAC='XX:XX:XX:XX:XX:XX')
D/RxBle#ClientOperationQueue: FINISHED ConnectOperation(185132267) in 35171 ms
D/RxBle#ClientOperationQueue: STARTED  DisconnectOperation(243638813)
W/libEGL: EGLNativeWindowType 0x75ffe5a010 disconnect failed
I/RxBle#ClientOperationQueue: RUNNING  DisconnectOperation{MAC='XX:XX:XX:XX:XX:XX'}
D/BluetoothManager: getConnectionState()
    getConnectedDevices
D/BluetoothGatt: close()
    unregisterApp() - mClientIf=8
D/RxBle#ClientOperationQueue: FINISHED DisconnectOperation(243638813) in 40 ms
```

dthib, 2019-06-26:
Thanks a lot for your help!

dariuszseweryn, 2019-06-27:
Hi,
Nothing comes to my mind apart from the OS not doing it's work well. Probably it keeps the connection between the app restarts

closed 2019-07-03 by dariuszseweryn