dariuszseweryn/RxAndroidBle #597: GATT_ERROR (133)

aamnapm, 2019-07-06:
hi i have ble device
when i connect ble device i have three authentication.
my first and second authentication work correctely but in thrice auth in response i have problem 
i read below link but i could not find by action 
https://github.com/Polidea/RxAndroidBle/issues/149#issue-214464408
what am i doing?

dariuszseweryn, 2019-07-08:
> what am i doing?

I know that you try to connect. After that I can only guess you have a three-step authentication that fails on the third step.

Please adhere to the issue template that you have completely ignored. This template tries to guide you through all steps needed for solving issues.

aamnapm, 2019-07-09:
@dariuszseweryn  i found error with sniffing  .in thrice step when i send byte[] my connection disconnect after 3 seconds and i can not get device response .how i can keep ble connection open ?

aamnapm, 2019-07-09:
i ask a ios developer (he work with this ble)  he says we opening connection with reading rssi .
its true?

dariuszseweryn, 2019-07-09:
> how i can keep ble connection open ?

I have no clue. Ask your peripheral's firmware developer why the peripheral closes the connection.
> i ask a ios developer (he work with this ble) he says we opening connection with reading rssi .
> its true?

I have no clue what you are doing but checking RSSI is a passive action and is not related to establishing a connection.

Either way this does not seem to be a problem with the library. I do not have information at this point that allow me to help you further and you do not seem to care about providing it.

What can you do further? Ask the peripheral's firmware developer on what is expected to happen to get a stable connection. Alternatively read the documentation of your peripheral and act accordingly.

Closing.

closed 2019-07-09 by dariuszseweryn